chipsalliance/riscv-dv #4: VCS "Incompatible complex type"

jerryz123, 2019-02-14:
Hi, I'm seeing errors when running `./run -test riscv_instr_base_test`. I see the following 2 errors:
```
Error-[ICTA] Incompatible complex type
./src/riscv_instr_gen_config.sv, 31
  Incompatible complex type assignment
  Incompatible complex types cannot be used in assignments, initializations 
  and instantiations.
  
Error-[ICTA] Incompatible complex type
./src/riscv_loop_instr.sv, 156
  Incompatible complex type assignment
  Incompatible complex types cannot be used in assignments, initializations 
  and instantiations.  
```
Do you have any suggestions? I am using vcs 2016.6.

taoliug, 2019-02-16:
I have changed the implementation to be complied to standard SV, please check if it solves your problem.

closed 2019-02-16 by taoliug

jerryz123, 2019-02-16:
Thanks, except now I get a VCS segmentation fault. What VCS version did you test with?

```
UVM_INFO ./src/riscv_instr_sequence.sv(72) @ 0: reporter@@sub_5 [sub_5] Start generating 30 instruction
UVM_INFO ./src/riscv_instr_sequence.sv(81) @ 0: reporter@@sub_5 [sub_5] Finishing instruction generation
UVM_INFO ./src/riscv_instr_sequence.sv(72) @ 0: reporter@@main_program [main_program] Start generating 6 instruction
UVM_INFO ./src/riscv_instr_sequence.sv(81) @ 0: reporter@@main_program [main_program] Finishing instruction generation
UVM_INFO ./src/riscv_asm_program_gen.sv(104) @ 0: reporter [asm_gen] Randomizing call stack

An unexpected termination has occurred in ./out_2019-02-16/vcs_simv due to a signal: Segmentation fault
Hostname a5

Command line: ./out_2019-02-16/vcs_simv +UVM_TESTNAME=riscv_instr_base_test +asm_file_name=./out_2019-02-16/asm_tests/riscv_instr_base_test +ntb_random_seed=1550307060 -l ./out_2019-02-16/sim_riscv_instr_base_test.log +num_of_tests=1

--- Stack trace follows:

Dumping VCS Annotated Stack:
No context available
===========================================================
                Generated RISC-V assembly tests
 ----------------------------------------------------------
```

taoliug, 2019-02-16:
I am testing with VCS/2017-12, the segmentation fault doesn't seem like a code issue, maybe try to use a better machine to run it again?

raulbehl, 2019-05-06:
Seeing similar fails in my runs too (using VCS 2014.10): 

```
UVM_INFO @ 0: reporter [RNTST] Running test riscv_instr_base_test...
UVM_INFO riscv_instr_gen_config.sv(311) @ 0: reporter [cfg] supported_privileged_mode = 3
UVM_INFO riscv_instr_gen_config.sv(311) @ 0: reporter [cfg] supported_privileged_mode = 3
UVM_INFO riscv_instr_base_test.sv(107) @ 0: uvm_test_top [uvm_test_top] riscv_instr_gen_config is randomized:
-------------------------------------------------------------
Name                      Type                    Size  Value
-------------------------------------------------------------
cfg                       riscv_instr_gen_config  -     @364 
  main_program_instr_cnt  integral                32    'h4  
  sub_program_instr_cnt   sa(integral)            5     -    
    [0]                   integral                32    'h58 
    [1]                   integral                32    'h4  
    [2]                   integral                32    'h1b 
    [3]                   integral                32    'h16 
    [4]                   integral                32    'h37 
-------------------------------------------------------------

UVM_INFO riscv_instr_base_test.sv(99) @ 0: uvm_test_top [uvm_test_top] All directed instruction is applied
UVM_INFO riscv_asm_program_gen.sv(602) @ 0: reporter [asm_gen] Randomizing page tables, totally 7 page tables, mode = USER_MODE
UVM_INFO riscv_page_table_list.sv(123) @ 0: reporter [page_table_list] Randomizing page table 0, num of PTE: 512
UVM_INFO riscv_page_table_list.sv(123) @ 0: reporter [page_table_list] Randomizing page table 1, num of PTE: 512
UVM_INFO riscv_page_table_list.sv(123) @ 0: reporter [page_table_list] Randomizing page table 2, num of PTE: 512
UVM_INFO riscv_page_table_list.sv(123) @ 0: reporter [page_table_list] Randomizing page table 3, num of PTE: 512
UVM_INFO riscv_page_table_list.sv(123) @ 0: reporter [page_table_list] Randomizing page table 4, num of PTE: 512
UVM_INFO riscv_page_table_list.sv(123) @ 0: reporter [page_table_list] Randomizing page table 5, num of PTE: 512
UVM_INFO riscv_page_table_list.sv(123) @ 0: reporter [page_table_list] Randomizing page table 6, num of PTE: 512
UVM_INFO riscv_asm_program_gen.sv(607) @ 0: reporter [asm_gen] Finished creating page tables
UVM_INFO riscv_page_table_list.sv(443) @ 0: reporter [page_table_list] Processing PT_0_PTE_0, v = 1, level = 2
UVM_INFO riscv_page_table_list.sv(443) @ 0: reporter [page_table_list] Processing PT_0_PTE_1, v = 1, level = 2
UVM_INFO riscv_page_table_list.sv(443) @ 0: reporter [page_table_list] Processing PT_1_PTE_0, v = 1, level = 1
UVM_INFO riscv_page_table_list.sv(443) @ 0: reporter [page_table_list] Processing PT_1_PTE_1, v = 1, level = 1
UVM_INFO riscv_page_table_list.sv(443) @ 0: reporter [page_table_list] Processing PT_2_PTE_0, v = 1, level = 1
UVM_INFO riscv_page_table_list.sv(443) @ 0: reporter [page_table_list] Processing PT_2_PTE_1, v = 1, level = 1
UVM_INFO riscv_instr_sequence.sv(72) @ 0: reporter@@sub_1 [sub_1] Start generating 88 instruction
UVM_INFO riscv_instr_sequence.sv(81) @ 0: reporter@@sub_1 [sub_1] Finishing instruction generation
UVM_INFO riscv_instr_sequence.sv(72) @ 0: reporter@@sub_2 [sub_2] Start generating 4 instruction
UVM_INFO riscv_instr_sequence.sv(81) @ 0: reporter@@sub_2 [sub_2] Finishing instruction generation
UVM_INFO riscv_instr_sequence.sv(72) @ 0: reporter@@sub_3 [sub_3] Start generating 27 instruction
UVM_INFO riscv_instr_sequence.sv(81) @ 0: reporter@@sub_3 [sub_3] Finishing instruction generation
UVM_INFO riscv_instr_sequence.sv(72) @ 0: reporter@@sub_4 [sub_4] Start generating 22 instruction
UVM_INFO riscv_instr_sequence.sv(81) @ 0: reporter@@sub_4 [sub_4] Finishing instruction generation
UVM_INFO riscv_instr_sequence.sv(72) @ 0: reporter@@sub_5 [sub_5] Start generating 55 instruction
UVM_INFO riscv_instr_sequence.sv(81) @ 0: reporter@@sub_5 [sub_5] Finishing instruction generation
UVM_INFO riscv_instr_sequence.sv(72) @ 0: reporter@@main_program [main_program] Start generating 4 instruction
UVM_INFO riscv_instr_sequence.sv(81) @ 0: reporter@@main_program [main_program] Finishing instruction generation
UVM_INFO riscv_asm_program_gen.sv(106) @ 0: reporter [asm_gen] Randomizing call stack

Error-[NOA] Null object access
riscv_instr_sequence.sv, 136
  The object at dereference depth 2 is being used before it was 
  constructed/allocated.
  Please make sure that the object is allocated before using it. 
  
  #0 in \riscv_instr_sequence::post_process_instr  at 
  riscv_instr_sequence.sv:136
  #1 in \riscv_asm_program_gen::gen_program  at riscv_asm_program_gen.sv:127
  #2 in \riscv_instr_base_test::run_phase  at riscv_instr_base_test.sv:100
  #3 in \uvm_run_phase::exec_task  at 
  /apps/vcsmx/etc/uvm-1.2/base/uvm_common_phases.svh:269
  #4 in \uvm_task_phase::execute  at 
  /apps/vcsmx/etc/uvm-1.2/base/uvm_task_phase.svh:152
  #5 in \uvm_phase::execute_phase  at 
  /apps/vcsmx/etc/uvm-1.2/base/uvm_phase.svh:1408
  #6 in \uvm_phase::m_run_phases  at 
  /apps/vcsmx/etc/uvm-1.2/base/uvm_phase.svh:2213
  #7 in \uvm_root::run_test  at /apps/vcsmx/etc/uvm-1.2/base/uvm_root.svh:513
  #8 in run_test at /apps/vcsmx/etc/uvm-1.2/base/uvm_globals.svh:43
```

yinhanquan, 2020-09-27:
> Thanks, except now I get a VCS segmentation fault. What VCS version did you test with?
>
> ```
> UVM_INFO ./src/riscv_instr_sequence.sv(72) @ 0: reporter@@sub_5 [sub_5] Start generating 30 instruction [...]

did you have solved this?

quswar-abid, 2022-05-17:
Having similar issue here (VCS2016)

`Error-[ICTA] Incompatible complex type
/root/Desktop/riscvdv/target/rv32imc/riscv_core_setting.sv, 118
  Incompatible complex type assignment
  Incompatible complex types cannot be used in assignments, initializations 
  and instantiations.
`

quswar-abid, 2022-05-17:
> 

But ommitting the assignments may help avoid the issue. Will it create new issues? I am not sure. Try doing this:
`// Implementation-specific custom CSRs
bit [11:0] custom_csr[] ;//= {};`