chipsalliance/riscv-dv #14: Way to disable some ISA groups

btran1969, 2019-03-28:
If one core supports only imc ( -march=rv32imc), what is the best way to disable other ISA groups?

taoliug, 2019-04-02:
You can configure the selected the ISA group here:
https://github.com/google/riscv-dv/blob/master/src/riscv_instr_gen_config.sv#L34
riscv_instr_group_t supported_isa[] = {RV32I, RV32M, RV64I, RV64M, RV32C, RV64C};

btran1969, 2019-04-04:
I did the following change

 // ISA supported by the processor
  riscv_instr_group_t supported_isa[] = {RV32I, RV32M, RV32C};

But then got these errors
run.int ./run -tool irun
Your DISPLAY variable is set to :3.0
Your job has been submitted to NC - Interactive Queue
Fairshare= /class/gui
Resources= guiHost CORES/0 Limit:gui_riscv_@USER@
Env      = SNAPPROP(@JOBID@)+D(DISPLAY=uls-ep-vhpcetx09:3.0)
Command  = vw ./run -tool irun
Logfile  = /dev/null
JobURL   = http://uls-ep-vhpcnc02:8271/cgi/node.cgi?id=001652034
JobId    = 001652034
    <<<<STARTING ON uls-ep-hpcbe037>>>>
          instr_name == SD; rs2 == saved_regs[i]; rs1 == SP; imm == 8 * (i+1);)
                                                                              |
ncvlog: *E,UNDIDN (./src/riscv_directed_instr_lib.sv,216|78): 'SD': undeclared identifier [12.5(IEEE)].
          instr_name == LD; rd == saved_regs[i]; rs1 == SP; imm == 8 * (i+1);)
                                                                             |
ncvlog: *E,UNDIDN (./src/riscv_directed_instr_lib.sv,282|77): 'LD': undeclared identifier [12.5(IEEE)].
        allowed_instr = {LW, SW, LWU, allowed_instr};
                                   |
ncvlog: *E,UNDIDN (./src/riscv_load_store_instr_lib.sv,110|35): 'LWU': undeclared identifier [12.5(IEEE)].
        allowed_instr = {LD, SD, allowed_instr};
                          |
ncvlog: *E,UNDIDN (./src/riscv_load_store_instr_lib.sv,116|26): 'LD': undeclared identifier [12.5(IEEE)].
        allowed_instr = {LD, SD, allowed_instr};
                              |
ncvlog: *E,UNDIDN (./src/riscv_load_store_instr_lib.sv,116|30): 'SD': undeclared identifier [12.5(IEEE)].
          allowed_instr = {C_LD, C_SD, allowed_instr};
                              |
ncvlog: *E,UNDIDN (./src/riscv_load_store_instr_lib.sv,118|30): 'C_LD': undeclared identifier [12.5(IEEE)].

taoliug, 2019-04-05:
It seems to be unrelated to the supported_isa change, do you have this compile failure before the change?

taoliug, 2019-04-07:
BTW, please set XLEN = 32
https://github.com/google/riscv-dv/blob/master/src/riscv_instr_pkg.sv#L42

btran1969, 2019-04-08:
I can generate a test now but it seems still generate 64-bit instructions since I got error when I ran the test

ascom: Running as on 0.cpp.s (0)..
vascom: Executing /wdc/apps/riscv/bin/riscv64-unknown-elf-as   -march=rv32imc 0.cpp.s -o 0.o
0.s: Assembler messages:
0.s:8139: Error: unrecognized opcode `ld x22,0(x7)'
0.s:8148: Error: unrecognized opcode `sd x22,0(x7)'
0.s:8160: Error: unrecognized opcode `sd x22,0(x7)'
0.s:8171: Error: unrecognized opcode `sd x22,0(x7)'
vascom: ERROR > assembler failed (ret=256)

btran1969, 2019-04-08:
pt_fault_handler: 
                  sfence.vma x0, x0
                  li x27, 2
                  li x17, 0x0
                  csrr x15, 0x343 # mtval
                  srli x15, x15, 12
                  slli x15, x15, 37
                  la x7, page_table_0
                  fix_pte:
                  srli x23, x15, 23
                  slli x23, x23, 2
                  add x7, x7, x23
                  **ld x22, 0(x7)**
                  slli x23, x22, 28
                  srli x23, x23, 29
                  bne zero, x23, fix_leaf_pte
                  beq zero, x27, fix_leaf_pte
                  srli x22, x22, 10
                  slli x22, x22, 10
                  li x23, 0xf1
                  or x22, x22, x23
                  sd x22, 0(x7)
                  srli x7, x22, 10
                  slli x7, x7, 12

taoliug, 2019-04-08:
Have you changed XLEN to 32?

btran1969, 2019-04-08:
Yes, I did. Just did a grep to make sure and got this

riscv_instr_pkg.sv:  parameter int XLEN = 32;

taoliug, 2019-04-09:
Can you check your SATP setting in riscv_instr_pkg? It should be SV32 or BARE for RV32IMC

btran1969, 2019-04-09:
It works now. Thanks. BTW, do you have a way to specify number of instructions we want to genererate  per test?

Thanks,

taoliug, 2019-04-09:
You can check the option here:
https://github.com/google/riscv-dv/blob/master/src/riscv_instr_gen_config.sv#L287

closed 2019-04-09 by taoliug