chipsalliance/riscv-dv #22: moving code around

btran1969, 2019-06-03:
How can I change the location of the program? I saw it starts from 0x0 now but want to change it to our ICCM(instruction closely coupled memory)? Do you have linkerscript or something similar?

taoliug, 2019-06-03:
Here's the reference link script
https://github.com/google/riscv-dv/blob/master/scripts/link.ld

You can modify it based on your real memory mapping.  BTW, the current physical address start from address 0x80000000 to match spike requirement. The virtual address starts from 0x0 which maps to physical address 0x80000000.

simon5656, 2019-06-04:
not really good to assume that all hardware has memory where the spike thinks it is?
surely we should be able to target where the program is located (or maybe use a reference that is not so restrictive.. :-) )
Simon

https://github.com/google/riscv-dv/blob/master/scripts/link.ld

taoliug, 2019-06-07:
Agree with what Simon said, it's just a constraint of spike, you don't have issue if you switch to other ISS (like ovpsim which is supported by riscv-dv now). I am going to close this issue, please let me know if you have further question on this.

closed 2019-06-07 by taoliug