chipsalliance/riscv-dv #25: Instruction weights and custom sequence

btran1969, 2019-06-11:
How can I specify weights for some instructions? 
How can I add new custom sequences?

I saw you reference to test/riscv_instr_test_lib.sv but there is not test dir

taoliug, 2019-06-13:
For the weights, you can add distribution constraints in below class (right now there's no knob to control the weights)
https://github.com/google/riscv-dv/blob/master/src/riscv_rand_instr.sv

For custom sequence, you can find some examples here:
https://github.com/google/riscv-dv/blob/master/src/riscv_directed_instr_lib.sv
https://github.com/google/riscv-dv/blob/master/src/riscv_load_store_instr_lib.sv

Here's how the custom sequence is mixed with the random sequence
https://github.com/google/riscv-dv/blob/master/test/riscv_instr_test_lib.sv#L139

taoliug, 2019-06-17:
I am closing this issue for now, please let me know if you have additional question on this.

closed 2019-06-17 by taoliug