chipsalliance/riscv-dv #31: How to get RTL for this code?

haykp, 2019-07-05:
First of all thanks for the nice code!
I am trying to use your code to setup and run big simulation on Emulator. If succeed I will share my results to community.
I am really new in RISC open-core, please help guys to get good DUT-RTL for your code.
Looking here: https://riscv.org/risc-v-cores/, but so many cores do not know which one to choose.

would you please provide some help with DUT selection? Preferable FPGA confirmed.
Also I checked the TB is self-checking right?

taoliug, 2019-07-09:
Please take a look this core:
https://github.com/lowRISC/ibex/tree/master/dv/uvm

This includes a complete TB with self-checking

taoliug, 2019-07-11:
Closing this bug now, please let me know if you any other questions.

closed 2019-07-11 by taoliug