chipsalliance/riscv-dv #37: Xcelium compile failure

scottj97, 2019-07-17:
After commit 23ad2e95333bad5f41476e89c557552003728d39 I am seeing an issue with the include_file macro during Xcelium (19.03.001) compile.

```
  `include_file(`RISCV_CORE_SETTING)
                                   |
xmvlog: *E,COFILX (./src/riscv_instr_pkg.sv,651|35): cannot open include file '``f``'.
(`define macro: include_file [./src/riscv_instr_pkg.sv line 25], file: ./src/riscv_instr_pkg.sv line 651)                                                                        
```

The syntax in this include_file macro does not seem to be LRM compliant. I believe it should be:
```
  `define include_file(f) `include `"f`"
```

scottj97, 2019-07-18:
You could also do something like this, instead of the `include_file` macro:
```
 `define quoted(f) `"f`"
 `include `quoted(`RISCV_CORE_SETTING)
```

taoliug, 2019-07-19:
Thanks for reporting, fixed

closed 2019-07-19 by taoliug