chipsalliance/riscv-dv #54: README.md corrections for the new flow

vandanaprabhu, 2019-08-01:
With reference to changes in New YAML based regression flow #46, these are my observations:

1. README.md still uses -tool irun instead of ius for the new flow

2. Also suggest adding  "run.py --help" as a first step to show the user all the new options and what the arguments may be since the README.md doesn't quite cover the newer options

3.--help says you expect  SO and CO as arguments for -so and -co but that doesn't work for me so I ended up passing an int after looking at the code. Not passing an argument doesn't work either so the intent of the arguments to these options is not clear in the --help or README.md

4. --verbose VERBOSE doesn't work either so had to pass an int to get it going similar to 3 above

scottj97, 2019-08-01:
For boolean options like these, python's argparse offers `action='store_true'` (instead of `type=int`).

taoliug, 2019-08-01:
Thanks for reporting, I will take a look and get back to you.

danghai, 2019-08-01:
I got the same issue relating to `RISCV_GCC`:
```
danghai@ubuntu:~/google/riscv-dv$ python3 run.py --test=riscv_page_table_exception_test --iss=spike
Processing regression test list : /home/danghai/google/riscv-dv/yaml/testlist.yaml, test: riscv_page_table_exception_test
Found matched tests: riscv_page_table_exception_test, iterations:2
Processing simulator setup file : /home/danghai/google/riscv-dv/yaml/simulator.yaml
Found matching simulator: vcs
Building RISC-V instruction generator
Running RISC-V instruction generator
Generating 2 riscv_page_table_exception_test
Please set the environment variable RISCV_GCC
```

taoliug, 2019-08-01:
Just submitted PR #56 to address the issues, please let me know if there's anything else not working for you.

taoliug, 2019-08-01:
I haven't tested with questa though, please let me know if you have issue running with questa.

danghai, 2019-08-01:
Hi @taoliug , I create a PR to make it work for questa.

vandanaprabhu, 2019-08-01:
The questa flow appears to be broken (hence the empty asm file) because of the spaces preceding the multiline vlog and opt options simulator.yaml file. 
When I changed these to tabs as shown below and the compile and sim worked
Also the correct command is sv_seed <seed>  (not svsejd <seed> )value is not being passed correctly it appears.

- tool: questa
  compile_cmd:
    - "vmap mtiUvm $QUESTA_HOME/questasim/uvm-1.2"
    - "vlog -64
    -access=rwc
    -f <cwd>/files.f
    -sv
    -mfcu -cuname design_cuname
    +define+UVM_REGEX_NO_DPI
    -writetoplevels <out>/top.list
    -l <out>/compile.log <cmp_opts>"
    - "vopt -64 -debug
    +designfile -f <out>/top.list
    -o design_opt"
  sim_cmd: >
    vsim -64 -c -do <cwd>/questa_sim.tcl design_opt <sim_opts>  -sv_seed <seed>

Let me know if you want me to create a pull request but I believe this should fix Hai’s issue.

Vandana

taoliug, 2019-08-01:
Please feel free to create a PR if it fixes the questa sim issue on your side. Thanks.

commit b65e9de7 (2019-08-01): Merge pull request #56 from google/flow

Update README and run.py [fix #54]
closed 2019-08-02 by taoliug