chipsalliance/riscv-dv #67: Log Compare in Co-Simulation Flow

berkkis, 2019-08-08:
Hi,
Regarding the log comparison could you please elaborate on how you verify rtl instruction trace ?
Thank you.

taoliug, 2019-08-08:
In case your processor has a tracer, you can refer to ibex TB for example
Compare script: https://github.com/lowRISC/ibex/blob/master/dv/uvm/sim.py#L193
Convert RTL trace to standard trace CSV: https://github.com/lowRISC/ibex/blob/master/dv/uvm/riscv_dv_extension/ibex_log_to_trace_csv.py
Doc: https://github.com/lowRISC/ibex/blob/master/doc/verification.rst

If there's no tracer,  the recommended approach is to monitor the register file update and compare with ISS trace. We will have an example for this published later.

berkkis, 2019-08-08:
Thank you for the answer.

closed 2019-08-08 by berkkis