chipsalliance/riscv-dv #73: Log Compare in Co-Simulation Flow - spike_log_to_trace_csv.py does not produce any output

sdexter-numa, 2019-08-14:
I set up the ibex example and I'm using the latest master of riscv-dv, and noticed that the comparison at the end doesn't work because the script "spike_log_to_trace_csv.py" does not produce any output in it's resulting csv file. Any clues on what might be going wrong?

taoliug, 2019-08-14:
Can you post the command you are using and the screen log?

sdexter-numa, 2019-08-15:
**Commands:**

`git clone https://github.com/lowRISC/ibex.git`
`cd ibex/dv/uvm`
`make TEST=riscv_arithmetic_basic_test ITERATIONS=1 WAVES=0 SEED=1`

**Screen log:**

Thu, 15 Aug 2019 10:16:30 INFO     Processing regression test list : <ibex_git_repo>/dv/uvm/riscv_dv_extension/testlist.yaml, test: riscv_arithmetic_basic_test
Thu, 15 Aug 2019 10:16:30 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:1
Thu, 15 Aug 2019 10:16:30 INFO     Processing simulator setup file : <ibex_git_repo>/ibex/vendor/google_riscv-dv/yaml/simulator.yaml
Thu, 15 Aug 2019 10:16:30 INFO     Found matching simulator: vcs
Thu, 15 Aug 2019 10:16:30 INFO     Building RISC-V instruction generator
Thu, 15 Aug 2019 10:16:42 INFO     Running RISC-V instruction generator
Thu, 15 Aug 2019 10:16:42 INFO     Generating 1 riscv_arithmetic_basic_test
Thu, 15 Aug 2019 10:17:20 INFO     Processing regression test list : <ibex_git_repo>/dv/uvm/riscv_dv_extension/testlist.yaml, test: riscv_arithmetic_basic_test
Thu, 15 Aug 2019 10:17:20 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:1
Thu, 15 Aug 2019 10:17:20 INFO     Compiling <ibex_git_repo>/dv/uvm/out/instr_gen/asm_tests/riscv_arithmetic_basic_test.0.S
Thu, 15 Aug 2019 10:17:20 INFO     Converting to <ibex_git_repo>/dv/uvm/out/instr_gen/asm_tests/riscv_arithmetic_basic_test.0.bin
Thu, 15 Aug 2019 10:17:20 INFO     Processing regression test list : <ibex_git_repo>/dv/uvm/riscv_dv_extension/testlist.yaml, test: riscv_arithmetic_basic_test
Thu, 15 Aug 2019 10:17:21 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:1
Thu, 15 Aug 2019 10:17:21 INFO     Processing ISS setup file : <ibex_git_repo>/vendor/google_riscv-dv/yaml/iss.yaml
Thu, 15 Aug 2019 10:17:21 INFO     Found matching ISS: spike
Thu, 15 Aug 2019 10:17:21 INFO     spike sim log dir: <ibex_git_repo>/dv/uvm/out/instr_gen/spike_sim
Thu, 15 Aug 2019 10:17:21 INFO     Running ISS simulation: <ibex_git_repo>/dv/uvm/out/instr_gen/asm_tests/riscv_arithmetic_basic_test.0.o
Processing simulator setup file : yaml/rtl_simulation.yaml
Found matching simulator: vcs
Compiling TB
Processing simulator setup file : yaml/rtl_simulation.yaml
Found matching simulator: vcs
Running RTL simulation...
Running core_ibex_base_test with <ibex_git_repo>/dv/uvm/out/instr_gen/asm_tests/riscv_arithmetic_basic_test.0.bin
Processing simulator setup file : yaml/rtl_simulation.yaml
Found matching simulator: vcs
Comparing spike/DUT sim result : <ibex_git_repo>/dv/uvm/out/asm_tests/riscv_arithmetic_basic_test.0.o
Processing ibex log : <ibex_git_repo>/dv/uvm/out/rtl_sim/riscv_arithmetic_basic_test.0/trace_core_00_0.log
Processed instruction count : 7508
0 PASSED, 1 FAILED
RTL & ISS regression report is saved to <ibex_git_repo>/dv/uvm/out/regr.log

**Contents of <ibex_git_repo>/dv/uvm/out/regr.log:**

Test binary: <ibex_git_repo>/dv/uvm/out/asm_tests/riscv_arithmetic_basic_test.0.o
ibex : <ibex_git_repo>/dv/uvm/out/rtl_sim/riscv_arithmetic_basic_test.0/trace_core_00_0.csv
spike : <ibex_git_repo>/dv/uvm/out/instr_gen/spike_sim/riscv_arithmetic_basic_test.0.csv
Mismatch[1]:
[2] ibex : lui              x2, 0xde6af000        -> sp(0xde6af000) addr:0x8000009a
7507 instructions left in trace ibex
Compare result[FAILED]: 0 matched, 1 mismatch

0 PASSED, 1 FAILED

**Contents of <ibex_git_repo>/dv/uvm/out/rtl_sim/riscv_arithmetic_basic_test.0/trace_core_00_0.csv**

instr,rd,rd_val,rs1,rs1_val,rs2,rs2_val,imm,str,addr,binary,mode
,ra,00000000,,,,,,"addi             x1, x0, 0            ",80000098,00000093,
,sp,de6af000,,,,,,"lui              x2, 0xde6af000       ",8000009a,de6af137,
,sp,de6af432,,,,,,"addi             x2, x2, 1074         ",8000009e,43210113,
,gp,b2b85000,,,,,,"lui              x3, 0xb2b85000       ",800000a2,b2b851b7,
,gp,b2b849f8,,,,,,"addi             x3, x3, -1544        ",800000a6,9f818193,
,tp,00000000,,,,,,"addi             x4, x0, 0            ",800000aa,00000213,
,t0,92ed0000,,,,,,"lui              x5, 0x92ed0000       ",800000ac,92ed02b7,
,t0,92ed0792,,,,,,"addi             x5, x5, 1938         ",800000b0,79228293,
,t1,00000000,,,,,,"addi             x6, x0, 0            ",800000b4,00000313,
,t2,00000000,,,,,,"addi             x7, x0, 0            ",800000b6,00000393,
...
etc.

** Contents of <ibex_git_repo>/dv/uvm/out/instr_gen/spike_sim/riscv_arithmetic_basic_test.0.csv**

instr,rd,rd_val,rs1,rs1_val,rs2,rs2_val,imm,str,addr,binary,mode

...
This test file is blank. It is supposed to be generated by <ibex_git_repo>/vendor/google_riscv-dv/scripts/spike_log_to_trace_csv.py. Attempting to run this script manually gives the same result, as follows:

**Command**

`cd <ibex_git_repo>/uvm/dv`
`python3 ../../vendor/google_riscv-dv/scripts/spike_log_to_trace_csv.py --log out/instr_gen/spike_sim/riscv_arithmetic_basic_test.0.log --csv out/instr_gen/spike_sim/riscv_arithmetic_basic_test.0.csv`

**Contents of <ibex_git_repo>/uvm/dv/out/instr_gen/spike_sim/riscv_arithmetic_basic_test.0.log**

core   0: 0x0000000000001010 (0x00028067) jr      t0
core   0: 0xffffffff80000080 (0x0100006f) j       pc + 0x10
core   0: 0xffffffff80000090 (0x5f47c037) lui     zero, 0x5f47c
core   0: 0xffffffff80000094 (0x51100013) li      zero, 1297
core   0: 0xffffffff80000098 (0x00004081) c.li    ra, 0
core   0: 0xffffffff8000009a (0xde6af137) lui     sp, 0xde6af
core   0: 0xffffffff8000009e (0x43210113) addi    sp, sp, 1074
core   0: 0xffffffff800000a2 (0xb2b851b7) lui     gp, 0xb2b85
core   0: 0xffffffff800000a6 (0x9f818193) addi    gp, gp, -1544
core   0: 0xffffffff800000aa (0x00004201) c.li    tp, 0
core   0: 0xffffffff800000ac (0x92ed02b7) lui     t0, 0x92ed0
core   0: 0xffffffff800000b0 (0x79228293) addi    t0, t0, 1938
core   0: 0xffffffff800000b4 (0x00004301) c.li    t1, 0
core   0: 0xffffffff800000b6 (0x00004381) c.li    t2, 0
...
etc.

taoliug, 2019-08-15:
Please enable RISCV_ENABLE_COMMITLOG when install spike
https://github.com/riscv/riscv-isa-sim/blob/master/config.h.in#L73

sdexter-numa, 2019-08-15:
I made that change and reinstalled spike, and pulled all the latest changes from both git repositories, it seems like some changes went in since yesterday, now it completely passes.

closed 2019-08-15 by sdexter-numa