chipsalliance/riscv-dv #77: Subset of Instruction Groups & riscv_core_settings.sv

berkkis, 2019-08-15:
Hi,
Right now i am trying to generate tests with a subset of supported instruction group namely "rv64im". I would appreciate if you could point out the where i am wrong.
1)Following the README.md  and manipulating 
  riscv_core_settings.sv:
  ...
  riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV64I, RV64M};
  //supported but not used = {RV32C, RV64C}
  ...
2)After, these comands are run
  ./run -tool vcs -test all
  ./iss_sim -iss spike -isa rv64im -abi lp64
3)The problem is here. 
  "iss_sim" outputs assembler errors regarding "C" extention.
"""
iss start
iss_processed_command_line
gcc compile start

riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_sfence_exception_test.1.S
./out_2019-08-15/asm_tests/riscv_sfence_exception_test.1.S: Assembler messages:
./out_2019-08-15/asm_tests/riscv_sfence_exception_test.1.S:6295: Error: unrecognized opcode `c.beqz a0,sub_3_stack_p'
Convert ./out_2019-08-15/asm_tests/riscv_sfence_exception_test.1.S.o to ./out_2019-08-15/asm_tests/riscv_sfence_exception_test.1.S.bin
/home/riscv-toolchain/riscv-gnu-ins/bin/riscv64-unknown-elf-objcopy: './out_2019-08-15/asm_tests/riscv_sfence_exception_test.1.S.o': No such file
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_no_fence_test.1.S
./out_2019-08-15/asm_tests/riscv_no_fence_test.1.S: Assembler messages:
./out_2019-08-15/asm_tests/riscv_no_fence_test.1.S:10742: Error: unrecognized opcode `c.bnez a1,sub_2_stack_p'
./out_2019-08-15/asm_tests/riscv_no_fence_test.1.S:13064: Error: unrecognized opcode `c.bnez a2,sub_3_stack_p'
./out_2019-08-15/asm_tests/riscv_no_fence_test.1.S:26722: Error: unrecognized opcode `c.bnez a1,smode_ls_umem_program_stack_p'
Convert ./out_2019-08-15/asm_tests/riscv_no_fence_test.1.S.o to ./out_2019-08-15/asm_tests/riscv_no_fence_test.1.S.bin
/home/riscv-toolchain/riscv-gnu-ins/bin/riscv64-unknown-elf-objcopy: './out_2019-08-15/asm_tests/riscv_no_fence_test.1.S.o': No such file
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_mmu_stress_test.1.S
Convert ./out_2019-08-15/asm_tests/riscv_mmu_stress_test.1.S.o to ./out_2019-08-15/asm_tests/riscv_mmu_stress_test.1.S.bin
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S
./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S: Assembler messages:
./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S:487: Error: unrecognized opcode `c.beqz s1,j__main_sub_1_1'
./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S:711: Error: unrecognized opcode `c.bnez s1,sub_4_stack_p'
./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S:7948: Error: unrecognized opcode `c.beqz a4,j_sub_1_sub_2_3'
./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S:18635: Error: unrecognized opcode `c.bnez a2,j_sub_3_sub_4_8'
./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S:19171: Error: unrecognized opcode `c.beqz a2,sub_5_stack_p'
Convert ./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S.o to ./out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S.bin
/home/riscv-toolchain/riscv-gnu-ins/bin/riscv64-unknown-elf-objcopy: './out_2019-08-15/asm_tests/riscv_rand_instr_test.0.S.o': No such file
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_rand_jump_test.0.S
Convert ./out_2019-08-15/asm_tests/riscv_rand_jump_test.0.S.o to ./out_2019-08-15/asm_tests/riscv_rand_jump_test.0.S.bin
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_ebreak_test.1.S
Convert ./out_2019-08-15/asm_tests/riscv_ebreak_test.1.S.o to ./out_2019-08-15/asm_tests/riscv_ebreak_test.1.S.bin
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_page_table_exception_test.0.S
./out_2019-08-15/asm_tests/riscv_page_table_exception_test.0.S: Assembler messages:
./out_2019-08-15/asm_tests/riscv_page_table_exception_test.0.S:4593: Error: unrecognized opcode `c.bnez s0,sub_3_stack_p'
./out_2019-08-15/asm_tests/riscv_page_table_exception_test.0.S:8499: Error: unrecognized opcode `c.beqz a0,sub_5_stack_p'
./out_2019-08-15/asm_tests/riscv_page_table_exception_test.0.S:8695: Error: unrecognized opcode `c.bnez a4,sub_1_stack_p'
Convert ./out_2019-08-15/asm_tests/riscv_page_table_exception_test.0.S.o to ./out_2019-08-15/asm_tests/riscv_page_table_exception_test.0.S.bin
/home/riscv-toolchain/riscv-gnu-ins/bin/riscv64-unknown-elf-objcopy: './out_2019-08-15/asm_tests/riscv_page_table_exception_test.0.S.o': No such file
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_no_fence_test.0.S
./out_2019-08-15/asm_tests/riscv_no_fence_test.0.S: Assembler messages:
./out_2019-08-15/asm_tests/riscv_no_fence_test.0.S:18550: Error: unrecognized opcode `c.beqz a3,j_sub_2_sub_5_6'
Convert ./out_2019-08-15/asm_tests/riscv_no_fence_test.0.S.o to ./out_2019-08-15/asm_tests/riscv_no_fence_test.0.S.bin
/home/riscv-toolchain/riscv-gnu-ins/bin/riscv64-unknown-elf-objcopy: './out_2019-08-15/asm_tests/riscv_no_fence_test.0.S.o': No such file
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S
./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S: Assembler messages:
./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S:551: Error: unrecognized opcode `c.beqz s1,j__main_sub_1_3'
./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S:1324: Error: unrecognized opcode `c.beqz a5,j_sub_3_sub_5_5'
./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S:1686: Error: unrecognized opcode `c.beqz s0,j_sub_3_sub_4_6'
./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S:2897: Error: unrecognized opcode `c.bnez a1,sub_5_stack_p'
./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S:20295: Error: unrecognized opcode `c.beqz a0,j_sub_2_sub_3_4'
Convert ./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S.o to ./out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S.bin
/home/riscv-toolchain/riscv-gnu-ins/bin/riscv64-unknown-elf-objcopy: './out_2019-08-15/asm_tests/riscv_sfence_exception_test.0.S.o': No such file
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_arithmetic_basic_test.1.S
Convert ./out_2019-08-15/asm_tests/riscv_arithmetic_basic_test.1.S.o to ./out_2019-08-15/asm_tests/riscv_arithmetic_basic_test.1.S.bin
riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_rand_jump_test.1.S
./out_2019-08-15/asm_tests/riscv_rand_jump_test.1.S: Assembler messages:
[... 57 lines cut ...]
Running spike: riscv_rand_instr_test.0.S.o
Running spike: riscv_rand_jump_test.0.S.o
Running spike: riscv_ebreak_test.1.S.o
Running spike: riscv_page_table_exception_test.0.S.o
Running spike: riscv_no_fence_test.0.S.o
Running spike: riscv_sfence_exception_test.0.S.o
Running spike: riscv_arithmetic_basic_test.1.S.o
Running spike: riscv_rand_jump_test.1.S.o
Running spike: riscv_machine_mode_rand_test.0.S.o
Running spike: riscv_page_table_exception_test.1.S.o
Running spike: riscv_wfi_test.1.S.o
Running spike: riscv_machine_mode_rand_test.1.S.o
Running spike: riscv_privileged_mode_rand_test.1.S.o
Running spike: riscv_ebreak_test.0.S.o
Running spike: riscv_illegal_instr_test.0.S.o
Running spike: riscv_illegal_instr_test.1.S.o
Running spike: riscv_mmu_stress_test.0.S.o
Running spike: riscv_rand_instr_test.1.S.o
Running spike: riscv_wfi_test.0.S.o
Running spike: riscv_hint_instr_test.0.S.o
Running spike: riscv_arithmetic_basic_test.0.S.o
Running spike: riscv_privileged_mode_rand_test.0.S.o
Running spike: riscv_hint_instr_test.1.S.o

"""

taoliug, 2019-08-15:
Do you want to disable compressed instruction in one test or all tests?

berkkis, 2019-08-15:
I tried to disable for all the tests. But if there is a test-specific solution I can use it as well.

taoliug, 2019-08-16:
Please try run riscv_non_compressed_instr_test test
https://github.com/google/riscv-dv/blob/master/yaml/testlist.yaml#L83

If the processor doesn't support compressed instruction, you can disable it in the processor setting
https://github.com/google/riscv-dv/blob/master/setting/riscv_core_setting.sv#L33

berkkis, 2019-08-16:
For the first link, I will try the riscv_non_compressed instr_test. 
  For the second link, that's what I've tried to do. But at the riscv_gcc compilation step, there were still compressed instructions from the instruction stream resulting in assembler error. My question is this an expected behavior.
> `./iss_sim -iss spike -isa rv64im -abi lp64`

> ```
> riscv_gcc compiling : ./out_2019-08-15/asm_tests/riscv_rand_instr_test.1.S
> ./out_2019-08-15/asm_tests/riscv_rand_instr_test.1.S: Assembler messages: [...]

taoliug, 2019-08-16:
This issue should be fixed by #81 , can you try again?

taoliug, 2019-08-19:
I am closing this issue for now, please let me know if it doesn't work for you.

closed 2019-08-19 by taoliug