chipsalliance/riscv-dv #83: DV_CHECK_FATAL & QuestaSim

berkkis, 2019-08-19:
Hi I am trying use questasim with riscv-dv platform. 
But during test generation i get the following error.
'''
QuestaSim-64 vlog 10.5 Compiler 2016.02 Feb 12 2016
Start time: 12:16:06 on Aug 19,2019
vlog -64 -access=rwc -f ./files.f -sv -mfcu -cuname design_cuname "+define+UVM_REGEX_NO_DPI" -writetoplevels ./out_2019-08-19/top.list -l ./out_2019-08-19/compile.log 
-- Compiling package riscv_instr_pkg
-- Importing package mtiUvm.uvm_pkg (uvm-1.2 Built-in)
** Note: (vlog-2286) ./src/riscv_instr_pkg.sv(21): Using implicit +incdir+/sw/mentor/questa_10_5c/questasim/uvm-1.2/../verilog_src/uvm-1.2/src from import uvm_pkg
** Error: ** while parsing file included at ./src/riscv_instr_pkg.sv(774)
** at ./src/riscv_reg.sv(129): (vlog-2266) Expansion of macro 'DV_CHECK_FATAL' failed: The number of actual arguments (3) are not equal to the number of formal arguments (4).
'''

berkkis, 2019-08-20:
Details:
The following code works with vcs but it is not working for questa.

Under "dv_defines.svh" there are the related macros:
...
`ifndef DV_CHECK_FATAL
  `define DV_CHECK_FATAL(T_, MSG_="", ID_=`gfn, WITH_C_=) \
    `DV_CHECK(T_, MSG_, fatal, ID_, WITH_C_)
`endif

// Shorthand for common foo.randomize() + fatal check
`ifndef DV_CHECK_RANDOMIZE_FATAL
  `define DV_CHECK_RANDOMIZE_FATAL(VAR_, MSG_="Randomization failed!", ID_=`gfn) \
    `DV_CHECK_FATAL(VAR_.randomize(), MSG_, ID_)
`endif
...

It seems the error comes from here. The "DV_CHECK_RANDOMIZE_FATAL" macro has 3 variables passed to "DV_CHECK_FATAL" but "DV_CHECK_FATAL" macro expects 4.

scottj97, 2019-08-20:
This seems related to my change 187796b (part of #41) to fix compile warnings from Questa. I'll investigate.

scottj97, 2019-08-20:
I am unable to recreate with Questa 2019.01. On the latest master (6b126d9) I ran the following, per README:

```
python3 run.py --test riscv_arithmetic_basic_test --simulator questa
```

And it ran successfully. My compile log shows:
```
QuestaSim-64 vlog 2019.1 Compiler 2019.01 Jan  1 2019
Start time: 11:05:55 on Aug 20,2019
vlog -64 -access=rwc -f /home/sjohnson/riscv-dv/files.f -sv -mfcu -cuname design_cuname "+define+UVM_REGEX_NO_DPI" -writetoplevels /home/sjohnson/riscv-dv/out_2019-08-20/top.list -l /home/sjohnson/riscv-dv/out_2019-08-20/compile.log
-- Compiling package riscv_instr_pkg
-- Importing package mtiUvm.uvm_pkg (uvm-1.2 Built-in)
** Note: (vlog-2286) ./src/riscv_instr_pkg.sv(21): Using implicit +incdir+/d1/cad/tools/mentor/questa/2019.1/questasim/uvm-1.2/../verilog_src/uvm-1.2/src from import uvm_pkg
-- Compiling package riscv_instr_test_pkg
-- Importing package riscv_instr_pkg
-- Compiling module riscv_instr_gen_tb_top
-- Importing package riscv_instr_test_pkg

Top level modules:
        riscv_instr_gen_tb_top
End time: 11:05:56 on Aug 20,2019, Elapsed time: 0:00:01
Errors: 0, Warnings: 0
```

I suspect the problem is your old version of Questa.

I don't see any issue with the macros in question. There are 4 formal arguments to DV_CHECK_FATAL, but the 2nd, 3rd, and 4th all have defaults specified. It is therefore legal to call it with 1, 2, 3, or 4 actual arguments.

danghai, 2019-08-20:
It works for me `2019.1` version.

berkkis, 2019-08-21:
Thank you I will look into this.

closed 2019-08-21 by berkkis

tejasmal56, 2020-05-12:
Hi,
Can we reopen this issue, i have a query related to this.
I am trying to generate binary file from the assembly test file.
I am getting the below error(link is attached) when i try to make use of make gen command.
[image].
I tried using the alternate method you had provided that is using the command as below:
python3 run.py --test riscv_arithmetic_basic_test --simulator questa
But even after using this command I am getting the following error .(link is attached
[image]
).
Can you please help me with this.

scottj97, 2020-05-12:
That looks like the exact same error message as @berkkis had with 2016.02. I assume a newer Questa version fixed the issue for him.

Can you try with 2019.1 or later? They're up to 2020.2 now.

tejasmal56, 2020-05-14:
Thanks, will try the latest version for questa as suggested.

danghai, 2020-05-14:
@tejasmal56 . You can try `qrun` from Questa. version (> 2019) has this feature. Let me know if you have any issues relating to it