chipsalliance/riscv-dv #86: Getting AttributeError: module 'yaml' has no attribute 'safe_load'

munish37, 2019-08-21:


munish37, 2019-08-21:
Steps followed 
1. Installed python 
2. Installed bitstring
3, While running the below command inside riscv_dv folder
python3 run.py --test=riscv_arithmetic_basic_test

We are getting AttributeError: module 'yaml' has no attribute 'safe_load'.
......../././riscv-dv/scripts/lib.py", line 40,

Please help us resolve the issue

taoliug, 2019-08-21:
Can you try if "pip install pyyaml" solves your problem?

munish37, 2019-08-21:
Thanks for reply .
I tried with the above command . Still getting the same issue.
I think this is due to the attribute name safe_load is not present in yaml.
Please help to resolve this issue.

taoliug, 2019-08-21:
The command works fine for me. safe_load is part of yaml package
https://pyyaml.org/wiki/PyYAMLDocumentation

taoliug, 2019-08-22:
Please check if this helps you:
https://stackoverflow.com/questions/9286637/pyyaml-error-attribute-error-no-attribute-load

I am closing this issue now as it doesn't seem like a flow issue itself.

closed 2019-08-22 by taoliug

margej, 2019-10-14:
You sill see this error if python can't find the yaml module.
The reason you dont see the normal "ModuleNotFoundError: No module named 'yaml'" error is because the base dir of the repo contains a directory called yaml, which python presumably assumes is the missing module so you get the slightly more cryptic message.

(You can confirm this by renaming the yaml directory)

Like me, you probably have a broken Python install

margej, 2019-10-14:
sudo pip3 install PyYAML worked for me

weedmank, 2019-10-25:
The "sudo pip3 ..." seems to have solved the YAML problem.  Now I have figure out how to get gcc and questasim setup for this RV32i:
[engr@localhost riscv-dv]$ python3 run.py --test=riscv_arithmetic_basic_test

Thu, 24 Oct 2019 17:47:56 INFO     Processing regression test list : /home/engr/riscv-dv/riscv-dv/yaml/testlist.yaml, test: riscv_arithmetic_basic_test
Thu, 24 Oct 2019 17:47:56 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Thu, 24 Oct 2019 17:47:56 INFO     Processing simulator setup file : /home/engr/riscv-dv/riscv-dv/yaml/simulator.yaml
Thu, 24 Oct 2019 17:47:56 INFO     Found matching simulator: vcs
Thu, 24 Oct 2019 17:47:56 INFO     Building RISC-V instruction generator
Thu, 24 Oct 2019 17:47:56 INFO     Running RISC-V instruction generator
Thu, 24 Oct 2019 17:47:56 INFO     Generating 2 riscv_arithmetic_basic_test
Thu, 24 Oct 2019 17:47:56 INFO     Running riscv_arithmetic_basic_test with 1 batches
Thu, 24 Oct 2019 17:47:56 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Thu, 24 Oct 2019 17:47:56 INFO     Compiling out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.S
riscv64-unknown-elf-gcc: error: out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.S: No such file or directory
riscv64-unknown-elf-gcc: error: unrecognized command line option '--with-arch=rv32imac'
riscv64-unknown-elf-gcc: error: unrecognized command line option '--with-abi=ilp32'
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.

weedmank, 2019-10-25:
The "sudo pip3 ..." seems to have solved the YAML problem. Now I have figure out how to get gcc and questasim setup for this RV32i:

[engr@localhost riscv-dv]$ python3 run.py --test riscv_arithmetic_basic_test --simulator questa
Thu, 24 Oct 2019 18:05:51 INFO     Processing regression test list : /home/engr/riscv-dv/riscv-dv/yaml/testlist.yaml, test: riscv_arithmetic_basic_test
Thu, 24 Oct 2019 18:05:51 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Thu, 24 Oct 2019 18:05:51 INFO     Processing simulator setup file : /home/engr/riscv-dv/riscv-dv/yaml/simulator.yaml
Thu, 24 Oct 2019 18:05:51 INFO     Found matching simulator: questa
Thu, 24 Oct 2019 18:05:51 INFO     Building RISC-V instruction generator
Thu, 24 Oct 2019 18:05:54 INFO     Running RISC-V instruction generator
Thu, 24 Oct 2019 18:05:54 INFO     Generating 2 riscv_arithmetic_basic_test
Thu, 24 Oct 2019 18:05:54 INFO     Running riscv_arithmetic_basic_test with 1 batches
Thu, 24 Oct 2019 18:05:54 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Thu, 24 Oct 2019 18:05:57 INFO     Compiling out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.S
riscv64-unknown-elf-gcc: error: out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.S: No such file or directory
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.

ww18735135443, 2021-04-13:
I has this problem"AttributeError: module 'yaml' has no attribute 'safe_load'" too