chipsalliance/riscv-dv #109: setup

agrobman, 2019-09-03:
I'm new with the riscv-dv and never dealt with python before. 

I'm getting following error :

alexander.grobman/lx1/riscv-dv]% python3 run.py --test=riscv_arithmetic_basic_test
Tue, 03 Sep 2019 14:28:43 INFO     Processing regression test list : /wdc/proj/riscv/mustang/users/alexander.grobman/lx1/riscv-dv/yaml/testlist.yaml, test: riscv_arithmetic_basic_test
Traceback (most recent call last):
  File "/wdc/proj/riscv/mustang/users/alexander.grobman/lx1/riscv-dv/scripts/lib.py", line 38, in read_yaml
    yaml_data = yaml.safe_load(f)
AttributeError: module 'yaml' has no attribute 'safe_load'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "run.py", line 433, in <module>
    main()
  File "run.py", line 407, in main
    process_regression_list(args.testlist, args.test, args.iterations, matched_list)
  File "/wdc/proj/riscv/mustang/users/alexander.grobman/lx1/riscv-dv/scripts/lib.py", line 147, in process_regression_list
    yaml_data = read_yaml(testlist)
  File "/wdc/proj/riscv/mustang/users/alexander.grobman/lx1/riscv-dv/scripts/lib.py", line 39, in read_yaml
    except yaml.YAMLError as exc:
AttributeError: module 'yaml' has no attribute 'YAMLError'

what's wrong?

udinator, 2019-09-03:
Can you try "pip install pyyaml" to see if this solves the issue?

agrobman, 2019-09-04:
thanks it did work.  However 
1) I don't have system administrator privileges, so I had to use --user switch
2) why didn't I get the email about your advice?

agrobman, 2019-09-04:
Now I'm getting constraint solving problem:
VM_INFO ./src/riscv_instr_gen_config.sv(326) @ 0: reporter [cfg] riscv_instr_pkg::supported_privileged_mode = 3
    `DV_CHECK_RANDOMIZE_FATAL(cfg);
                                 |
xmsim: *W,SVRNDF (./test/riscv_instr_base_test.sv,109|33): The randomize method call failed. The unique id of the failed randomize call is 0.
Observed simulation time : 0 FS + 44
xmsim: *W,RNDOCS: These constraints contribute to the set of conflicting constraints:

    !(signature_addr_reg inside {ZERO, default_reserved_regs, loop_regs}); (./src/riscv_instr_gen_config.sv,259)
xmsim: *W,RNDOCS: These variables contribute to the set of conflicting constraints:

state variables:
       default_reserved_regs[4] [./src/riscv_instr_gen_config.sv, 145] <ZERO=0, RA, SP, GP, TP, T0, T1, T2, S0, S1 ..., to change enum value display threshold, use -xceligen display_enums_val_count=<count>, 0 means all> 
rand variables:
       signature_addr_reg [./src/riscv_instr_gen_config.sv, 125] <ZERO=0, RA, SP, GP, TP, T0, T1, T2, S0, S1 ..., to change enum value display threshold, use -xceligen display_enums_val_count=<count>, 0 means all> 

xmsim: *W,RNDOCS: These constraints contribute to the set of conflicting constraints:

    !(signature_data_reg inside {ZERO, default_reserved_regs, loop_regs}); (./src/riscv_instr_gen_config.sv,260)
xmsim: *W,RNDOCS: These variables contribute to the set of conflicting constraints:

state variables:
       default_reserved_regs[4] [./src/riscv_instr_gen_config.sv, 145] <ZERO=0, RA, SP, GP, TP, T0, T1, T2, S0, S1 ..., to change enum value display threshold, use -xceligen display_enums_val_count=<count>, 0 means all> 
rand variables:
       signature_data_reg [./src/riscv_instr_gen_config.sv, 126] <ZERO=0, RA, SP, GP, TP, T0, T1, T2, S0, S1 ..., to change enum value display threshold, use -xceligen display_enums_val_count=<count>, 0 means all> 

UVM_FATAL ./test/riscv_instr_base_test.sv(109) @ 0: uvm_test_top [uvm_test_top] Check failed (cfg.randomize()) Randomization failed! 
UVM_INFO /wdc/apps/cadence/xcelium/19.03.002/tools/methodology/UVM/CDNS-1.2/sv/src/base/uvm_report_catcher.svh(705) @ 0: reporter [UVM/REPORT/CATCHER] 
--- UVM Report catcher Summary ---

I'm running :
xrun(64): 19.03-s002: (c) Copyright 1995-2019 Cadence Design Systems, Inc.
T

taoliug, 2019-09-04:
Have you made any local change?

udinator, 2019-09-04:
I've been unable to reproduce this on my end, what test are you running, and what is the seed of the failing simulation?

agrobman, 2019-09-04:
no, but when I commented out these constraints it generated the test

agrobman, 2019-09-04:
my command:

 python3 run.py --test=riscv_arithmetic_basic_test --simulator ius

agrobman, 2019-09-04:
ls there way to not generate exception handlers and startup/exit code?

agrobman, 2019-09-04:
Does the riscv-dv generates backward branches?

taoliug, 2019-09-04:
The constraint solver issue is solved. Can you file separate issue for new questions? To quickly address your question:
Q: ls there way to not generate exception handlers and startup/exit code?
A: It's already supported, please refer to riscv_asm_program_gen::gen_all_trap_handler

Q: Does the riscv-dv generates backward branches?
A: Yes. It's implemented by a dedicated loop instruction sequence.
     https://github.com/google/riscv-dv/blob/master/src/riscv_loop_instr.sv

closed 2019-09-04 by taoliug

agrobman, 2019-09-04:
thanks a lot, It works now

commit df27d84f (2019-09-04): Fix #109 ius constraint solver failure
commit be226226 (2019-09-04): Fix #109 ius constraint solver failure
commit f64cc2f6 (2019-09-04): Merge pull request #111 from google/flow

Fix #109 ius constraint solver failure