chipsalliance/riscv-dv #112: How to generate a test with backward branches?

agrobman, 2019-09-04:


taoliug, 2019-09-04:
Please refer to the riscv_rand_instr_test
https://github.com/google/riscv-dv/blob/master/yaml/testlist.yaml#L76
You can increase the ratio of loop instructions to generate more backward branch instructions.

The loop instruction is generated by below class, you can enhance it if you want to cover other backward branch scenarios.
https://github.com/google/riscv-dv/blob/master/src/riscv_loop_instr.sv

agrobman, 2019-09-04:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.

Thanks, I got this one generated, how do I increase probability of the loops?

https://github.com/google/riscv-dv/blob/master/yaml/testlist.yaml#L76

https://github.com/google/riscv-dv/blob/master/src/riscv_loop_instr.sv

taoliug, 2019-09-04:
Just increase the number 4 (this is the knob for the possibilities)
+directed_instr_1=riscv_loop_instr,4
You can also remove the other directed tests from the command line options if you don't want to include them in your test.

agrobman, 2019-09-04:
Thanks a lot again.

taoliug, 2019-09-04:
BTW, are you using it for swerv core? I have done some work locally, the ISS/RTL sim comparison part as the tracer from swerv core doesn't have the rd value information. Let me know if you need help on this.

agrobman, 2019-09-04:
Yes, we want to use riscv_dv  as  a 2nd source test generator for next generation of the swerv cores. I’m not sure what did you mean by “rd value information”. Internally (in house) we are comparing RTL CPU state with ISS for every executed instruction ..

taoliug, 2019-09-04:
I mean the external trace interface doesn't provide the result of the instruction. Maybe there'e some other way to get the complete trace info?
https://github.com/westerndigitalcorporation/swerv_eh1/blob/master/design/swerv.sv#L34

agrobman, 2019-09-04:
I need to talk to management. To get full CPU state you need to integrate our UVM tracer/whisper(ISS).

https://github.com/westerndigitalcorporation/swerv_eh1/blob/master/design/swerv.sv#L34

taoliug, 2019-09-04:
Sure, let me close this issue for now, we can discuss with a separate thread.
I actually have a plan to open-source an integrated TB with swerv-eh1.  Adding whisper(ISS) should be straightforward, but I failed to install in my local (got core dump). If you get it work, it will be great to add this support in the riscv-dv repository. Let me if you question with ISS integration.

closed 2019-09-04 by taoliug

agrobman, 2019-09-04:
Our ISS guy is going to vacation soon, I’ll ask if he can help you with whisper .. Still whisper need to be integrated to TB …