chipsalliance/riscv-dv #113: How to generate a test without exception handlers, CPU init and exit code?

agrobman, 2019-09-04:
we want to use our exception handlers init and exit code

taoliug, 2019-09-04:
How do you plan to integrate your own exception handlers, CPU init and exit code? Do you want to them to stay in a separate file or you want to integrate them into the instruction generator?

agrobman, 2019-09-04:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.

We have them written already, and would prefer be included via include statements. Also we probably need to select “reserved” registers randomly.

taoliug, 2019-09-04:
Let me get to it one by one.

CPU init:
You can extend riscv_asm_program_gen and override gen_init_section function
https://github.com/google/riscv-dv/blob/master/src/riscv_asm_program_gen.sv#L354

You can even make this function empty, and include your own init program in the gcc compilation. Make sure to name the program with _init label.

taoliug, 2019-09-04:
Exception handler:
This part is a bit tricky. The easiest way I can think of is overrding trap_vector_init function
https://github.com/google/riscv-dv/blob/master/src/riscv_asm_program_gen.sv#L571
You can re-write this function, program MTVEC to the label of your own exception handler.  BTW, we have built-in a lot of exception detection and recovery logic in the trap handling routine. You might want to try if it's adding value to your verification. 

For the exit of the program:
Right now the program is finished by executing an ecall instruction. If you want to have your own way to do it, just design an ecall exception handler to exit the program in your own way.

agrobman, 2019-09-06:
can you, please, add a configuration flag to generate "bare" random test w/o cpu init/exit code and exception handlers and make sure that last executed instruction is the last line in the file?
this way one can include the generated code in our template with custom init/exit and exception handlers. 

we want to exercise variable reset vector, nmi vector and etc. Our TB extracts certain labels automatically to feed the RTL model with inputs defined by the testcase

taoliug, 2019-09-06:
I added a +bare_program_mode option, you can give a try.  In this mode, all init/exit/page table setup/exception handling/privileged mode setup stuff. I can see the program run to ecall and fail because there's no exception handling for ecall. Please try it matches your requirement.

taoliug, 2019-09-06:
BTW, I don't have a good environment to fully test this mode. Please report back if you have issue with integrating with your environment.

agrobman, 2019-09-06:
thanks for doing this,

I would like to have:

.text
_start:
init_GPS
TEST_BODY
LONG_JUMP_TO_EOF
SUBS
.data
DATA 
.text
EOF:

so I could include this file in our test wrapper :
...
"
.include "riscv_test_env.h"
.include "kuku_test.0.S"
TEST_END(0)
"

BTW, can we have "_" instead of "." between test name and the test number?
( or make the separator programmable via setting file?)

taoliug, 2019-09-06:
for now the program structure is like this

.text
_start:
  main program
  ecall // This should be handled by ecall_handler to terminate the test
  sub programs
.data
DATA

Will this work for you? I prefer to keep all the instruction sections together so that it's easy to setup the page tables.

agrobman, 2019-09-06:
No,

I don't need ecall we stop when CPU writes tohost and  reaches "_finish:"  if the write data is 0 the TB considers that the set passes, otherwise failed

I don't want you write anything for CSRs , we have different CSRs to program,

initialize GPRs to random values before main is good idea too ..

I don't need your handlers - they won't work in our environment - we inject various errors, which need to be handled certain way to match ISS ..

agrobman, 2019-09-06:
I dont like this:
trap_vec_init:    
                  la a0, mtvec_handler
                  ori a0, a0, 0
                  csrw 0x305, a0 # MTVEC

mepc_setup:       
                  la x10, _init
                  csrw mepc, x10
                  j init_machine_mode

and
test_done:        
                  li gp, 1
                  ecall

for test_done: I need:
la t5,tohost
sw 0,0(t5)
_finish: 
 j _finish

taoliug, 2019-09-06:
trap_vec_init and mepc_setup are both gone with bare program mode.  I can replace ecall with jump to test_done in bare program mode

taoliug, 2019-09-06:
BTW, is _finish a special keyword for your ISS? I am using _exit now as this is the way riscv-ovpsim terminate simulation.

agrobman, 2019-09-06:
this is our TB "special" symbol . It extracts the value and provides to verilog simulator to finish.

BTW, I got run away simulations with spike - 12 TB logfiles  looping in the write_tohost loop! 
Any idea why it didn't stop?

agrobman, 2019-09-06:
maybe just lable last instruction with _exit and _finish?

taoliug, 2019-09-06:
you can use "ps" to find out the run-away spike thread and kill it. I need to figure out a way to completely kill the subprocess when the iss sim timeout.

agrobman, 2019-09-06:
Actually I was wondering why didn’t spike stop? What is the end of test condition for it?

agrobman, 2019-09-06:
Did you push the changes?

taoliug, 2019-09-06:
Please try the latest commit, here's the latest program structure for bare program mode

.text
_start:
init_gpr
main program
j test_end
sub programs
test_end
.data
DATA

You can override below function in riscv_asm_program_gen.sv to the way that works for your TB.

  virtual function void gen_program_end();
    // Use write_tohost to terminate spike simulation
    gen_section("write_tohost", {"sw gp, tohost, t5"});
    gen_section("_exit", {"j write_tohost"});
  endfunction

agrobman, 2019-09-06:
Do I miss something? I use  following command to generate  the test:
run.int xrun -R -svseed 123456 +UVM_TESTNAME=riscv_instr_base_test +num_of_tests=1 +asm_file_name=ttt +instr_cnt=1500 +num_of_sub_program=20 +directed_instr_0=riscv_load_store_rand_instr_stream,8 +bare_program_mode

I still got:

                  li x29, 0x0
                  li x30, 0xf11707a2
                  li x31, 0xe
                  li x15, 0x20001105
                  csrw misa, x15
kernel_sp:
                  la tp, _kernel_stack_end <------------------------------------ what is this for?

trap_vec_init:    <================================= I thought it should go?
                  la a0, mtvec_handler
                  ori a0, a0, 0
                  csrw 0x305, a0 # MTVEC

mepc_setup:
                  la x10, _init
                  csrw mepc, x10
                  j init_machine_mode

_init:
                  la sp, _user_stack_end
_main:            la         s7, data_page_15+2854 #start riscv_load_store_rand_instr_stream_75
                  mul        a6, a3, a5
                  sb         t5, -865(s7)
                  csrrwi     t6, 0x340, 18
                  srl        s1, s1, t1
                  c.nop
…
1035:             mulhu      s5, s7, s9
1036:             mulhu      a4, t3, s9
1037:             c.and      s0, s0
                  bgeu       t0, zero, 1039f
1039:             nop
test_done:
                  li gp, 1
                 ecall <<<<<<<<<<<<<<<<<<<<<<<<<<<<<< still ecall ???
sub_17:    addi       sp, sp, -44
                  sw         ra, 4(sp)
                  c.slli     a0, 7
                  sltiu      a0, s10, -716
                  sw         t0, 8(sp)
                  sw         t1, 12(sp)
                  sw         t3, 16(sp)
                  srl        t6, s6, t6
                  sw         s1, 20(sp)

….

And later all these handlers …

taoliug, 2019-09-06:
Please use "+bare_program_mode =1" rather than "+bare_program_mode"

agrobman, 2019-09-06:
Isn’t because of “=” in the + string?

src/riscv_instr_gen_config.sv:    get_bool_arg_value("+bare_program_mode=", bare_program_mode);

agrobman, 2019-09-06:
Got it,

Now we a missing .text – everything is .section text.init …

taoliug, 2019-09-07:
.text.init is changed to .text, mark this issue as fixed.

PR #134 (2019-09-07): Re-organize text and data section
Fix #113, update #129
closed 2019-09-07 by taoliug