chipsalliance/riscv-dv #116: How to generate subroutine/nested subroutine tests?

agrobman, 2019-09-04:
Need to verify branch predictor

taoliug, 2019-09-04:
Can you elaborate a bit more? Do you want a mix of backward to forward branch instructions? or you mean nested jump instructions?

agrobman, 2019-09-04:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.

I meant emulation of C like functions which call another functions or call itself recursively . Our CPU has return stack logic in branch predictor.  This logic needs nested functions calls/returns pairs to be tested.

taoliug, 2019-09-04:
Try this test
https://github.com/google/riscv-dv/blob/master/yaml/testlist.yaml#L95

Increase the number of sub-programs to generate a deeper stack. 
For more details of the call stack generation, you can refer to this file:
https://github.com/google/riscv-dv/blob/master/src/riscv_callstack_gen.sv

taoliug, 2019-09-04:
BTW, recursive call to itself is not supported, it can be added as a directed instruction stream.

agrobman, 2019-09-04:
thanks for support

closed 2019-09-05 by taoliug