chipsalliance/riscv-dv #118: generator performance

agrobman, 2019-09-05:
it looks like the test generation is quite slow ( at least with Cadence xrun )- it seems we simulate the tests much faster than they are generated. Are there any plans to run verilog profiler to improve generation speed?

It took ~19min to generate 20 riscv_rand_jump_test tests..

taoliug, 2019-09-05:
Are you running with default riscv_rand_jump_test setting? or with increased sub-programs count?

agrobman, 2019-09-05:
the same as defined in tests.yaml

taoliug, 2019-09-05:
It takes me 1 minute to generate 10 tests with VCS. but irun seems to be much slower. Let me do a profiling with irun and see what can be improved.

Thu, 05 Sep 2019 10:58:27 INFO     Running RISC-V instruction generator
Thu, 05 Sep 2019 10:58:27 INFO     Generating 10 riscv_rand_jump_test
Thu, 05 Sep 2019 10:59:27 INFO     Compiling vcs_sim/asm_tests/riscv_rand_jump_test.0.S
....

agrobman, 2019-09-05:
I ran the same command, compiling w/o "-access"  - got 15 min 20 sec

taoliug, 2019-09-06:
With #120 , I can see performance improvement around 60%. It's now taking around 4 mins to generate 10 tests, down from 10 mins.  It's still 4x slower than VCS, need to explore more coding style changes to further improve performance.

Thu, 05 Sep 2019 20:53:33 INFO     Found matching simulator: ius
Thu, 05 Sep 2019 20:53:33 INFO     Building RISC-V instruction generator
Thu, 05 Sep 2019 20:53:52 INFO     Running RISC-V instruction generator
Thu, 05 Sep 2019 20:53:52 INFO     Generating 10 riscv_rand_jump_test
Thu, 05 Sep 2019 20:57:47 INFO     Compiling out_2019-09-05/asm_tests/riscv_rand_jump_test.0.S

taoliug, 2019-09-06:
Down to 2 minutes now... (I also reduced the instruction count for this test by 1/3 because it could generate a very long test when there's a lot of jumps between sub programs)

Thu, 05 Sep 2019 23:22:11 INFO     Generating 10 riscv_rand_jump_test
Thu, 05 Sep 2019 23:23:14 INFO     Compiling out_2019-09-05/asm_tests/riscv_rand_jump_test.0.S

taoliug, 2019-09-06:
Can you verify if this solves your problem? There are still some tests (riscv_mmu_stress_test) takes longer time (30s) to generate with ius, I will further look into it.

agrobman, 2019-09-06:
Oh,  it runs much faster now!  I got 20 tests for 1m 21s, thanks a lot, again

closed 2019-09-06 by taoliug