chipsalliance/riscv-dv #124: solver issue for load/stores

simon5656, 2019-09-06:
trying to use: 
Linux 32bit Questa Sim Version 10.7c_3 linux Dec 21 2018

python3 run.py --test riscv_rand_instr_test --simulator questa32 --iss spike,ovpsim --iterations 1 --seed 1234 --isa rv32i --mabi ilp32 --testlist yaml/testlist_work.yaml --iss_yaml yaml/iss_ovpsim.yaml
I get: 
....
# UVM_FATAL ./src/riscv_asm_program_gen.sv(1079) @ 0: reporter [asm_gen] Check failed (new_instr_stream.randomize()) Randomization failed! 

if I edit yaml/testlist_work.yaml and remove one of the lines:
    +directed_instr_*
it does not matter which one, then the generator creates a legal .S file and all works.

thanks
Simon

taoliug, 2019-09-06:
Thanks for reporting, I don't see this failure with vcs/irun. Is there any more information about the failure? Like the conflict constraint cannot be solved?

taoliug, 2019-09-06:
BTW, looks like you are simulating with your own testlist_work.yaml. Can you share the setting of the failed test?

simon5656, 2019-09-06:
my own testlist_yaml is just a copy so I dont touch the original when I make changes. the error appears when I use the original testlist.yaml with no changes.

in a good run I see (with the     +directed_instr_0=riscv_load_store_rand_instr_stream,4 line in testlist.yaml removed which somehow makes things run ok)
...
UVM_INFO ./test/riscv_instr_base_test.sv(110) @ 0: uvm_test_top [uvm_test_top] riscv_instr_gen_config is randomized:
# -------------------------------------------------------------
# Name                      Type                    Size  Value
# -------------------------------------------------------------
# cfg                       riscv_instr_gen_config  -     @384 
#   main_program_instr_cnt  integral                32    'h11 
#   sub_program_instr_cnt   sa(integral)            5     -    
#     [0]                   integral                32    'ha  
#     [1]                   integral                32    'hf  
#     [2]                   integral                32    'hb  
#     [3]                   integral                32    'h13 
#     [4]                   integral                32    'h1c 
# -------------------------------------------------------------
# 
# UVM_INFO ./test/riscv_instr_base_test.sv(101) @ 0: uvm_test_top [uvm_test_top] All directed instruction is applied
# UVM_INFO ./src/riscv_asm_program_gen.sv(227) @ 0: reporter [asm_gen] sub program name: sub
# UVM_INFO ./src/riscv_instr_sequence.sv(77) @ 0: reporter@@sub_1 [sub_1] Start generating 10 instruction
# UVM_INFO ./src/riscv_instr_sequence.sv(86) @ 0: reporter@@sub_1 [sub_1] Finishing instruction generation
# UVM_INFO ./src/riscv_asm_program_gen.sv(227) @ 0: reporter [asm_gen] sub program name: sub
# UVM_INFO ./src/riscv_instr_sequence.sv(77) @ 0: reporter@@sub_2 [sub_2] Start generating 15 instruction
# UVM_INFO ./src/riscv_instr_sequence.sv(86) @ 0: reporter@@sub_2 [sub_2] Finishing instruction generation
# UVM_INFO ./src/riscv_asm_program_gen.sv(227) @ 0: reporter [asm_gen] sub program name: sub
# UVM_INFO ./src/riscv_instr_sequence.sv(77) @ 0: reporter@@sub_3 [sub_3] Start generating 11 instruction
# UVM_INFO ./src/riscv_instr_sequence.sv(86) @ 0: reporter@@sub_3 [sub_3] Finishing instruction generation
# UVM_INFO ./src/riscv_asm_program_gen.sv(227) @ 0: reporter [asm_gen] sub program name: sub
# UVM_INFO ./src/riscv_instr_sequence.sv(77) @ 0: reporter@@sub_4 [sub_4] Start generating 19 instruction
# UVM_INFO ./src/riscv_instr_sequence.sv(86) @ 0: reporter@@sub_4 [sub_4] Finishing instruction generation
# UVM_INFO ./src/riscv_asm_program_gen.sv(227) @ 0: reporter [asm_gen] sub program name: sub
# UVM_INFO ./src/riscv_instr_sequence.sv(77) @ 0: reporter@@sub_5 [sub_5] Start generating 28 instruction
# UVM_INFO ./src/riscv_instr_sequence.sv(86) @ 0: reporter@@sub_5 [sub_5] Finishing instruction generation
...

but in the bad run I see: (with the yaml section being:
- test: riscv_rand_instr_test
  description: >
    Random instruction stress test
  iterations: 2
  gen_test: riscv_instr_base_test
  gen_opts: >
    +instr_cnt=100
    +num_of_sub_program=5
    +directed_instr_0=riscv_load_store_rand_instr_stream,4
    +directed_instr_1=riscv_loop_instr,4
    +directed_instr_2=riscv_hazard_instr_stream,4
    +directed_instr_3=riscv_load_store_hazard_instr_stream,4
    +directed_instr_4=riscv_cache_line_stress_instr_stream,4
    +directed_instr_5=riscv_multi_page_load_store_instr_stream,4
  rtl_test: core_base_test

I see:
...
# UVM_INFO ./test/riscv_instr_base_test.sv(110) @ 0: uvm_test_top [uvm_test_top] riscv_instr_gen_config is randomized:
# -------------------------------------------------------------
# Name                      Type                    Size  Value
# -------------------------------------------------------------
# cfg                       riscv_instr_gen_config  -     @384 
#   main_program_instr_cnt  integral                32    'h11 
#   sub_program_instr_cnt   sa(integral)            5     -    
#     [0]                   integral                32    'ha  
#     [1]                   integral                32    'hf  
#     [2]                   integral                32    'hb  
#     [3]                   integral                32    'h13 
#     [4]                   integral                32    'h1c 
# -------------------------------------------------------------
# 
# UVM_INFO ./test/riscv_instr_base_test.sv(80) @ 0: uvm_test_top [uvm_test_top] Got directed instr[0] riscv_load_store_rand_instr_stream, ratio = 4/1000
[... 18 lines cut ...]
# UVM_INFO ./src/riscv_asm_program_gen.sv(1067) @ 0: reporter [asm_gen] Insert directed instr stream riscv_loop_instr 0/19 times
# UVM_INFO ./src/riscv_asm_program_gen.sv(1067) @ 0: reporter [asm_gen] Insert directed instr stream riscv_multi_page_load_store_instr_stream 0/19 times
# UVM_INFO ./src/riscv_instr_sequence.sv(77) @ 0: reporter@@sub_4 [sub_4] Start generating 19 instruction
# UVM_INFO ./src/riscv_instr_sequence.sv(86) @ 0: reporter@@sub_4 [sub_4] Finishing instruction generation
# UVM_INFO ./src/riscv_asm_program_gen.sv(227) @ 0: reporter [asm_gen] sub program name: sub
# UVM_INFO ./src/riscv_asm_program_gen.sv(1067) @ 0: reporter [asm_gen] Insert directed instr stream riscv_cache_line_stress_instr_stream 0/28 times
# UVM_INFO ./src/riscv_asm_program_gen.sv(1067) @ 0: reporter [asm_gen] Insert directed instr stream riscv_hazard_instr_stream 0/28 times
[... cut ...]
# UVM_INFO ./src/riscv_asm_program_gen.sv(1067) @ 0: reporter [asm_gen] Insert directed instr stream riscv_loop_instr 1/17 times
# UVM_INFO ./src/riscv_asm_program_gen.sv(1067) @ 0: reporter [asm_gen] Insert directed instr stream riscv_multi_page_load_store_instr_stream 1/17 times
# UVM_FATAL ./src/riscv_asm_program_gen.sv(1079) @ 0: reporter [asm_gen] Check failed (new_instr_stream.randomize()) Randomization failed! 
# UVM_INFO verilog_src/uvm-1.2/src/base/uvm_report_server.svh(847) @ 0: reporter [UVM/REPORT/SERVER] 

and so the output is different

thanks.

taoliug, 2019-09-06:
Seem like the instr_cnt is set to 100, which is pretty small. This could cause some failure as each sub-program has some minimum size constraint. Can you change it to 500 to see if the problem is solved?

taoliug, 2019-09-06:
Actually I simulated with your setting to generate 100 tests, and no constraint failure is observed. Might be something not working well is modelsim.

taoliug, 2019-09-09:
Simon, does the simulator give more information beyond below message? The current information is not enough to debug what goes wrong.

UVM_FATAL ./src/riscv_asm_program_gen.sv(1079) @ 0: reporter [asm_gen] Check failed (new_instr_stream.randomize()) Randomization failed!

taoliug, 2019-09-24:
I am closing this bug for inactivity, feel free to reopen it or file a new bug if you still see this issue.

closed 2019-09-24 by taoliug