chipsalliance/riscv-dv #125: run.py testlist option name consistency

simon5656, 2019-09-06:
the name in the run.py to select different yaml files tends to be the file start with _yaml, eg 
iss_yaml, csr_yaml, simulator_yaml
and then it is --testlist
pls make it consistent
suggest --testlist_yaml
thanks

taoliug, 2019-09-06:
I think the reason is that we have --iss and --simulator options as well, therefore the yaml file options have to be different. We don't have this issue for testlist, so intend to keep this name short.

taoliug, 2019-09-06:
I prefer to keep this as it is to avoid breaking the existing use model. Thanks for reporting.

closed 2019-09-06 by taoliug

simon5656, 2019-09-06:
I disagree - you need to have consistency in the naming of command arguments and --testlist is really --testlist_yaml
short should not come in to it.
and it is better to fix things now - at the beginning of the 10 year life of this program and not later...