chipsalliance/riscv-dv #127: Generator source localization

agrobman, 2019-09-06:
Is there way to generate tests in any other directory than riscv-dv?

We usually don't run in repository. We run randoms in separate working area which is cleaned periodically.

the generator file list contains "./' references, pointing to the current directory ...

taoliug, 2019-09-06:
Do you mean redirect the output to other directory? this can be done with --out <output dir> option.
Here's an example of running riscv-dv from other directory. If your flow is make file based, you can do the same thing. I can look into changing the relative path to absolute path in the filelist.
https://github.com/lowRISC/ibex/blob/master/dv/uvm/Makefile#L70

agrobman, 2019-09-06:
we have repository of the swerv in one directory, but compile and run simulations in completely different directory - one thing helps to do so is a env var pointing to the DB directory ... ($RV_ROOT)

all the files in the file list are referenced with $RV_ROOT/...

taoliug, 2019-09-06:
This should be solved now, please set RISCV_DV_ROOT to point to the source directory of riscv-dv. If it's not specified, it will be set to "." by default

closed 2019-09-06 by taoliug

agrobman, 2019-09-06:
thank a lot for taking care!

PR #128 (2019-09-06): Allow running riscv-dv from other directories
fix #127