chipsalliance/riscv-dv #132: bit manipulation instructions

agrobman, 2019-09-06:
are you going to support these?

taoliug, 2019-09-06:
This is not in the near term plan. It won't be too hard to support it as the instructions are mostly arithmetic types.  My main concern is that the spec is still work in progress, and the toolchain/ISS support might not be mature enough.

agrobman, 2019-09-06:
We have them in RTL already ...

taoliug, 2019-09-06:
I understand that, I can take a look at the effort and the toolchain support. I may need to talk to the working group to understand their plan as well.

taoliug, 2019-09-09:
Get some feedback from the folks working on bit manipulation support. The spec is still constantly changing and not ratified yet. There are some support on the toolchain and ISS side. However, since the spec is still evolving, these toolchain and ISS are subject to change as well. We'd like to add the support once it's ratified. We need to get other stuff like floating point, coverage model ready first before implement bit manipulation extension. 

BTW, if you have interest to add bit manipulation extension to riscv-dv shortly, we can discuss how we can collaborate on this effort.

taoliug, 2019-09-13:
Closing this issue as I don't think there's any immediate action in the near term. Will file a new bug once we start working on this extension.

closed 2019-09-13 by taoliug