chipsalliance/riscv-dv #142: test doesn't compile

agrobman, 2019-09-11:
strange if I generate a test with 100 instructions our script doesn't compile it -linker fails. For 1000 instructions it links.

the failure looks like:

(.text+0x544): relocation truncated to fit: R_RISCV_PCREL_LO12_S against `.L0 '
vascom: ERROR > linker/loader failed (ret=256)

objdump from test.o 

...
00000544 <write_tohost>:
 544:   00000f17                auipc   t5,0x0
 548:   003f2023                sw      gp,0(t5) # 544 <write_tohost>

0000054c <_exit>:
 54c:   bfe5                    j       544 <write_tohost>

I'm generating test as: 

 xrun -R -svseed 699553742 +UVM_TESTNAME=riscv_instr_base_test +num_of_tests=1 +asm_file_name=test +instr_cnt=100 +bare_program_mode=1  +num_of_sub_program=5 +directed_instr_0=riscv_lr_sc_instr_stream,10 +directed_instr_1=riscv_amo_instr_stream,10

taoliug, 2019-09-11:
It works for me, can you try to set num_of_sub_program to a small number when you run with small number of instructions? 0-2?

agrobman, 2019-09-11:
why does this matter?

taoliug, 2019-09-11:
I am just suspecting if there's any problem with handling subprogram generation when the instruction count is small. Does the program compile with the default link script?

agrobman, 2019-09-11:
Ok, I'm not sure what is going on - I'm not familiar with our compilation script. it generates  linker script automatically and the script almost empty for both cases - long or short test:
OUTPUT_ARCH( "riscv" )
ENTRY(_start)

SECTIONS
{
   _end = .;
}

I'm bringing our handlers / startup and exit code now by removing offending  pieces by perl script ..

taoliug, 2019-09-11:
OK, this seems to be problem of your custom link script generation flow? Let me know if this is something related to the section organization in the generated ELF. Please note that there's no .data section in the ELF anymore. All the data section is named specifically by the memory region name.