chipsalliance/riscv-dv #143: ncvlog building Error

zhangkunta, 2019-09-11:
We tried the current version of riscv-dv and found a lot of building error when running the command:
python3 run.py --test riscv_arithmetic_basic_test --simulator ius

The first line of the compile.log is:
    function new(string name="");
ncvlog: *E,NEWOIC (./src/dv_defines.svh,19|15): The class constructor method 'new' is only allowed within a class declaration.
(`include file: ./src/dv_defines.svh line 19, file: ./src/riscv_instr_pkg.sv line 23)

Thanks for any advice!

taoliug, 2019-09-11:
Thanks for reporting. It doesn't look like a real issue as the code it's pointing to is a macro define.
I pushed a PR to move the macro defines out anyway. If you are still seeing compilation issue, please paste the all the failures, thanks.

zhangkunta, 2019-09-11:
We tried FIX144 and the Error remains the same.
The first line of the compile.log REMAINS the SAME except the line number changed:
    function new(string name="");
ncvlog: *E,NEWOIC (./src/dv_defines.svh,19|15): The class constructor method 'new' is only allowed within a class declaration.
(`include file: ./src/dv_defines.svh line 19, file: ./src/riscv_instr_pkg.sv line **17**)
and the following error seems results from the 'new' function error:
    function new (string name=""); \
ncvlog: *E, NULLEI (./src/dv_defines.svh,19|36): zero length escaped identifier [2.7.1(IEEE)].
(`include file: ./src/dv_defines.svh line 19, file: ./src/riscv_instr_pkg.sv line 17)
    super.new(name); \
ncvlog: *E,MISEXX (./src/dv_defines.svh,20|10): expecting an '=' or '<=' sign in an assignment [9.2(IEEE)].
(`include file: ./src/dv_defines.svh line 20, file: ./src/riscv_instr_pkg.sv line 17)
....
....
AND anther 'new' function in dv_defines.svh results the same ERROR:
    function new(string name="", uvm_component parent=null);
ncvlog: *E,NEWOIC (./src/dv_defines.svh,26|15): The class constructor method 'new' is only allowed within a class declaration.
(`include file: ./src/dv_defines.svh line 26, file: ./src/riscv_instr_pkg.sv line **17**)

taoliug, 2019-09-11:
Looks werid, are you using an old version of incisive? I cannot reproduce this error on my side.

zhangkunta, 2019-09-11:
We are using "irun(64): 15.20-p001" (INCISIVE 15.2)

taoliug, 2019-09-11:
The closet version I have is 15.20-s010, but I cannot reproduce your issue. You might need to contact  the AE for a solution. I don't see there's a problem in the code itself.

taoliug, 2019-09-11:
Close this issue as not reproducible. I believe there are many other users run the flow with ius and dp not see this issue.
Feel free to reopen it if you think this is some issue in the code.

zhangkunta, 2019-09-11:
We solved the problem. 
The reason is that  `./src/dv_defines.svh` and several other files are in Windows format. We changed the format of these files and everything worked fine.
We suggest that the windows format files should be changed in case of future problems.

taoliug, 2019-09-11:
Cool, thanks for resolving this issue, I am not aware of this. Could you create a PR of your changes? I'd like to see what's causing the problem.