chipsalliance/riscv-dv #164: How do I control/redefine illegals

agrobman, 2019-09-17:
Hi,

Since we implemented B instructions I need to make sure that these are not generated as illegals or at least do not target SP or other reserved registers. How do I enforce this?

                  srai       t2, s4, 23
                  .4byte 0x21aed133 # kIllegalFunc7
                  slt        s0, a1, a7
                  fence

80003ea4:	417a5393          	srai	t2,s4,0x17
80003ea8:	21aed133          	0x21aed133
80003eac:	0115a433          	slt	s0,a1,a7

UVM_INFO [tracer] @816500.0ns: =>       x7 = ffffff00
UVM_INFO [tracer] @816500.0ns: => @816500 #3075 C0 80003ea4 [417a5393] srai     x7, x20, 0x17   OK
UVM_INFO [tracer] @816600.0ns: =>       x2 = fffffc00
UVM_INFO [tracer] @816600.0ns: => @816600 #3076 C0 80003ea8 [21aed133] sro      x2, x29, x26    OK
UVM_INFO [tracer] @816700.0ns: =>       x8 = 00000001
UVM_INFO [tracer] @816700.0ns: => @816700 #3077 C0 80003eac [0115a433] slt      x8, x11, x17    OK

this illegal corrupts stack pointer (x2)...

taoliug, 2019-09-17:
I'd recommend you extend below class and add constraint to avoid using opcode of B instructions.
https://github.com/google/riscv-dv/blob/master/src/riscv_illegal_instr.sv

agrobman, 2019-09-17:
I just added
    instr_bin[11:7] != 2; // do not generate target SP
to 
instr_bit_assignment_c constraint

taoliug, 2019-09-17:
I think it's better to make sure it's actually generating an illegal instruction. Otherwise, it could generate a valid instruction without proper constraint, which may do something crazy and break the program execution. For example, you could potentially generate load/store instruction without proper address, or branch/jump instruction which could lead to a dead loop. 

For the sp constraint, depending on the instruction format, you might need to constraint different bit position, which may make this solution more complicated to implement.

agrobman, 2019-09-17:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.

I think target register is always in the same position in the instruction …

taoliug, 2019-09-17:
It's not always true for the compressed instructions. Besides, bit [11:7] could be use for other purposes if there's no rd for the instruction. This constraint ends up constrains something else which could pose a hole in the randomization.

agrobman, 2019-09-17:
You are correct – but since we got B – there is even bigger hole. I’m looking for better, systematic way to generate illegals, than manually written constraints …  The RTL instruction decoder is written as pretty simple PLA table , much more clear/reviewable , than these constraints  - would be nice to have something like this to generate illegals:

.definition

clz       =  [011000000000.....001.....0010011]
ctz       =  [011000000001.....001.....0010011]
pcnt      =  [011000000010.....001.....0010011]
slo       =  [0010000..........001.....0110011]
…

.decode
rv32i[clz]       =  { alu zbb rs1     rd                       clz   }
rv32i[ctz]       =  { alu zbb rs1     rd                       ctz   }
rv32i[pcnt]      =  { alu zbb rs1     rd                       pcnt  }
rv32i[slo]       =  { alu zbb rs1 rs2 rd                       slo   }
….

So illegal is something that is  not defined in .definition
.decode defines flags/decoder outputs which used in the CPU pipe and execution units

Using in .decode a class type like B, I, M, F .. etc,  additional constraints for illegal would be for ex “F”  if floating is not implemented …

taoliug, 2019-09-18:
I think these are just two different ways to implement it. Either have a full legal binary encoding table and generate something not belongs to this table, or directly constraint the binary to target the illegal encoding. I will consider to refactor this class a bit to make it more readable. 
I am closing this issue for now as I believe the original issue is solved. Thanks for reporting.

closed 2019-09-18 by taoliug