chipsalliance/riscv-dv #190: "+disable_comparessed_instr=1" in run.py

abdullahyildiz, 2019-09-27:
When I run

`python3 run.py --test riscv_arithmetic_basic_test --iss spike --verbose --isa rv64g`

I see the following line at the terminal:

> /bin/sh: line 1: +disable_comparessed_instr=1: command not found

When I grep the repository I couldn't find a file which includes `disable_comparessed_instr` word (except run.py).

There is no `disable_compressed_instr` word in the repository either.

For reference:
https://github.com/google/riscv-dv/blob/2370dc84cf14d46715466960d55ddd93ab830874/run.py#L222

abdullahyildiz, 2019-09-27:
By the way,

I compiled spike with

`--with-isa=RV64IMAFD`

option and

removed RV32C and RV64C values in setting/riscv_core_setting.sv:

https://github.com/google/riscv-dv/blob/cfd95c9e9649ef110032b4ee4a6016abc69d08e4/setting/riscv_core_setting.sv#L33

taoliug, 2019-09-30:
Thanks for reporting, this is now fixed, please let me know if there's any issue.

closed 2019-09-30 by taoliug

abdullahyildiz, 2019-10-04:
@taoliug,

I get another problem (commit id: 8726466e6b566802fbaa2a945559b24dc3321218). I modified two lines in two files (run.py and testlist.yaml) to fix it:

```
diff --git a/run.py b/run.py
index 354a136..ae2ab05 100644
--- a/run.py
+++ b/run.py
@@ -219,7 +219,7 @@ def gen(test_list, csr_file, end_signature_addr, isa, simulator,
             if "gen_opts" in test:
               cmd += test['gen_opts']
             if not re.search("c", isa):
-              cmd += "+disable_comparessed_instr=1";
+              cmd += "+disable_compressed_instr=1";
             if lsf_cmd:
               cmd_list.append(cmd)
             else:
diff --git a/yaml/testlist.yaml b/yaml/testlist.yaml
index da4e331..a11eb62 100644
--- a/yaml/testlist.yaml
+++ b/yaml/testlist.yaml
@@ -38,7 +38,7 @@
     +no_fence=1
     +no_data_page=1
     +no_branch_jump=1
-    +boot_mode=m
+    +boot_mode=m \
   iterations: 2
   gen_test: riscv_instr_base_test
   rtl_test: core_base_test
```
Could you please check?

taoliug, 2019-10-04:
I fixed the first one you mentioned, but I don't know what the problem with the 2nd change you made.

abdullahyildiz, 2019-10-05:
I made the second change because the string

"+disable_compressed_instr=1"

was not being appended as an option to vcs_simv executable without adding a line continuation character (\).

PR #194 (2019-09-30): Add support for disable_compressed_instr
Fix #190