chipsalliance/riscv-dv #202: testcase fails with  RISCV_OBJCOPY environment variable not found

amitgargmentor, 2019-10-04:
Hi,

i have instaled the  RISCV installation and ISS Simulator (OVPSIM) and set the enviornment variable RISCV_GCC and OVPSIM_PATH as mentioned in README file.  
but when i ru the below test i see the error: 

// Run ISS with riscv-ovpsim
python3 run.py --test riscv_rand_instr_test --iss ovpsim --simulator questa

The Error is :
Fri, 04 Oct 2019 08:07:13 WARNING  Please set the environment variable RISCV_OBJCOPY

Am i missing some step which sets the RISCV_OBJCOPY enviornment variable ? 

Please help me out here. 

Thanks,
Amit

taoliug, 2019-10-04:
Yes, please set RISCV_OBJCOPY to point to you objcopy binary path, example:
export RISCV_OBJCOPY="$RISCV_TOOLCHAIN/bin/riscv64-unknown-linux-gnu-objcopy"

taoliug, 2019-10-04:
updated README for this.

closed 2019-10-04 by taoliug

amitgargmentor, 2019-10-04:
Thanks, It worked!!