chipsalliance/riscv-dv #204: coverage is not getting collected

amitgargmentor, 2019-10-04:
Hi, The README file mentions the step to generate the instruction and collect the coverage without running the simulation. 
I ran the below command as mentioned :
// Randomly generate 100000 instructions, split to 20000 instructions per batch
**python3 cov.py -d -i 100000 -bz 20000**

**The command give below output on  terminal:**
[amgarg@soc-pe-001 riscv-dv-master]$ python3 cov.py -d -i 100000 --iss ovpsim
Fri, 04 Oct 2019 08:16:21 INFO     Building the coverage collection framework
Fri, 04 Oct 2019 08:16:22 INFO     Randomizing 100000 instructions in 1 batches
Fri, 04 Oct 2019 08:16:22 INFO     Running batch 1/1
Fri, 04 Oct 2019 08:16:22 INFO     Collecting functional coverage from 100000 random instructions...done

It seems like the coverage has been collected but i dont find any file /logfile with the information.  i see directory out_2019_10_04 is created but it doesnt have any thing. 
Can you please  tell me what we need to look here? 

Thanks,

taoliug, 2019-10-04:
Right now the coverage flow only support VCS + spike, will need some work to add support for ovpsim.

amitgargmentor, 2019-10-04:
Does it support Questa+ SPIKE as well ? 

Also, while installing SPIKE, the first steps is apt-get install device-tree-compiler which gives me the issue. 

since i have RED HAT OS, i use the below command:
sudo tum install device-tree-compiler

but it gives me below error:

Loaded plugins: langpacks, product-id, search-disabled-repos, subscription-manager
This system is not registered with an entitlement server. You can use subscription-manager to register.
No package device-tree-compiler available.
Error: Nothing to do

what are the steps followed by you to install spike ?

taoliug, 2019-10-06:
To enable coverage collection for questa, you need to add coverage related compile/sim options here
https://github.com/google/riscv-dv/blob/master/yaml/simulator.yaml#L47

taoliug, 2019-10-06:
For spike setup, I am just following the standard steps here
https://github.com/riscv/riscv-isa-sim#build-steps

amitgargmentor, 2019-10-07:
Thanks, I am able to set up Spike.  
Now looking into coverage aspects through Questa. 

ti run with VCS, you pass --cov option while running run.py and then run the cov.py command as mentioned in README file?

taoliug, 2019-10-08:
--cov is not needed in your command line.

amitgargmentor, 2019-10-08:
Thanks, I am able to setup coverage flow with Questa now.  
(I also actually made Questa as default simulator as well)

taoliug, 2019-10-08:
Cool, this is good to know.
I am closing this issue now. If you want to share you cov setting for Questa, please create a PR for it, thanks.

closed 2019-10-08 by taoliug

amitgargmentor, 2019-10-08:
Hi Tao,

Below are the changes in simulator.yaml file to enable coverage for Questa:
- tool: questa
  compile:
    cmd:
      - "vmap mtiUvm $QUESTA_HOME/uvm-1.2"
      - "vlog -64
        +incdir+<setting>
        +incdir+<user_extension>
        -access=rwc
        -f <cwd>/files.f
        -sv
        -mfcu -cuname design_cuname
        +define+UVM_REGEX_NO_DPI
        -writetoplevels <out>/top.list
        -l <out>/compile.log <cmp_opts>"
      - "vopt -64 -debug
        -designfile design.bin -f <out>/top.list
        -l <out>/optimize.log <cmp_opts>
        -o design_opt"
  sim:
    cmd: >
      vsim -64 -c  <cov_opts> -do <cwd>/questa_sim.tcl design_opt <sim_opts>  -qwavedb=+signal+memory+transaction+class+uvm_schematic -sv_seed <seed>
    cov_opts: >
      -do "coverage save -onexit cov.ucdb;"