chipsalliance/riscv-dv #206: Question: retaining seeds to run generator with again

sdexter-numa, 2019-10-07:
I was wondering if it's possible to re-use the same seeds after running the generator with random seeds once.

For example, if I ran the generator, but noticed something wrong with one piece of generated assembly code, and want to fix and verify that it's working. I understand it's possible to extract the seed for a particular generator run, and use it to rerun that particular test. Is it possible to do this all at once for all tests in one go? To run again with the exact same seeds that were just used?

taoliug, 2019-10-08:
yes, this is supported, you can find out the seed in the generator log, and rerun the generator with "--seed xxxx"

taoliug, 2019-10-08:
I got your point, I can add a rerun feature for this.

taoliug, 2019-10-08:
This is now supported, every regression will generate a seed.yaml in the output directory. You can rerun the same regression with --seed_yaml <out>/seed.yaml to rerun the regression with the same seed.

sdexter-numa, 2019-10-09:
That's perfect! Works well as far as I can see.

closed 2019-10-09 by sdexter-numa