chipsalliance/riscv-dv #212: run.py script does not filter directory name inputs for special characters

sdexter-numa, 2019-10-14:
Minor issue, but it's coming up in my setup.

The run.py script accepts arguments for file/folder names, but cannot accept something like the following:

python3 run.py -o example/folder:riscv:build0/riscv_dv_out

Even if the backslashes are used, when the simulator runs the UVM testbench, it won't compile because it thinks the colons are input errors for a simulation name:

Error-[INVALID_SIMULATION_NAME] Invalid simulation name.
 The given simulation name
 /example/folder:riscv:build0/riscv_dv_out/vcs_simv
 is invalid.
 Please do not use the characters ':' in the executable's name.

taoliug, 2019-10-15:
Can you avoid using ":" in the directory name? This could cause issue in a few places. I am leaning to not fix this to avoid adding complexity to the script.

taoliug, 2019-10-16:
Closing this issue as there's no plan to fix this behavior.

closed 2019-10-16 by taoliug