chipsalliance/riscv-dv #213: VCS: Error-[CNST-CIF] Constraints inconsistency failure

margej, 2019-10-14:
Occassionally seeing no output assembly for some tests. Running with --verbose reveals the following:

Error-[CNST-CIF] Constraints inconsistency failure
./src/riscv_amo_instr_lib.sv, 133
  Constraints are inconsistent and cannot be solved.
  Please check the inconsistent constraints being printed above and rewrite 
  them.

UVM_FATAL ./src/riscv_amo_instr_lib.sv(133) @ 0: reporter [riscv_amo_instr_stream_0] Check failed (amo_instr[i].randomize()) Randomization failed! 

This is with vcs_mx.P-2019.06-SP1, and I would not rule out a VCS issue at this point. 
Is there a known good version of VCS to test against?

taoliug, 2019-10-15:
I ran the test 100 times but cannot reproduce the issue. Can you provide more information about the failure. I believe VCS could extract the conflict constraints to a separate file.

margej, 2019-10-15:
Thank you. 

I failed to mention I was not using the standard riscv_core_setting.sv configuration, so this is probably a function of my settings. I have reverted the setting file to original and the test passes (or at least the script runs to completion without reporting any errors)

My failing setup has the following changes to settings:

parameter int XLEN = 32;
parameter satp_mode_t SATP_MODE = BARE;
privileged_mode_t supported_privileged_mode[] = {USER_MODE};
riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV32C};
bit support_sfence = 0;

Which I now recognise is invalid since MACHINE_MODE is the only mandatory mode, not USER_MODE.

taoliug, 2019-10-15:
yes, if you add MACHINE_MODE to the list, do you still see this issue?

margej, 2019-10-15:
No, that resolves this issue, but then most of the tests timeout running spike sim.

Could this be because I have not removed User and Supervisor variants from implemented_csr[], implemented_interrupt[] and implemented_exception[]? 

I'll try this next

taoliug, 2019-10-15:
Just search for "exception" in the spike log to see if there's any exception raised.

taoliug, 2019-10-15:
implemented_csr[], implemented_interrupt[] and implemented_exception[] shouldn't matter, they are mostly used for coverage purposes.

margej, 2019-10-15:
Yes, about 1 million instances of this:
core   0: exception trap_load_access_fault, epc 0x0000000080012338

taoliug, 2019-10-15:
Can you let me know which test are you running, and paste a few lines of spike log before the first failure?

margej, 2019-10-15:
Sure. I'm running the full regression (I.e. simply python3 run.py). The first test to fail is:
Tue, 15 Oct 2019 18:31:07 ERROR    Timeout[25s]: /usr/julien/riscv-gnu-toolchain/install/bin/spike --isa=rv64gc -l out_2019-10-15/asm_tests/riscv_machine_mode_rand_test_0.o &> out_2019-10-15/spike_sim/riscv_machine_mode_rand_test.0.log

Here is the leadup to the first 2 exceptions:

```
core   0: 0x0000000080005768 (0x02588a33) mul     s4, a7, t0
core   0: 0x000000008000576c (0x01caa533) slt     a0, s5, t3
core   0: 0x0000000080005770 (0x00000661) c.addi  a2, 24
core   0: 0x0000000080005772 (0x0049a083) lw      ra, 4(s3)
core   0: 0x0000000080005776 (0x340378f3) csrrci  a7, mscratch, 6
core   0: 0x000000008000577a (0x0089a283) lw      t0, 8(s3)
core   0: 0x000000008000577e (0x000082a9) c.srli  a3, 10
core   0: 0x0000000080005780 (0x03898993) addi    s3, s3, 56
core   0: 0x0000000080005784 (0x340bb8f3) csrrc   a7, mscratch, s7
core   0: 0x0000000080005788 (0x03a6b8b3) mulhu   a7, a3, s10
core   0: 0x000000008000578c (0x00008d85) c.sub   a1, s1
core   0: 0x000000008000578e (0x00000001) c.nop
core   0: 0x0000000080005790 (0x00008082) ret
core   0: exception trap_instruction_access_fault, epc 0xffffffff800043b6
core   0:           tval 0xffffffff800043b6
core   0: 0x0000000080009000 (0x340999f3) csrrw   s3, mscratch, s3
core   0: 0x0000000080009004 (0x000289b3) add     s3, t0, zero
core   0: 0x0000000080009008 (0xf8498993) addi    s3, s3, -124
core   0: 0x000000008000900c (0x0019a223) sw      ra, 4(s3)
core   0: exception trap_store_access_fault, epc 0x000000008000900c
core   0:           tval 0xffffffff8001de04
core   0: 0x0000000080009000 (0x340999f3) csrrw   s3, mscratch, s3
core   0: 0x0000000080009004 (0x000289b3) add     s3, t0, zero
core   0: 0x0000000080009008 (0xf8498993) addi    s3, s3, -124
core   0: 0x000000008000900c (0x0019a223) sw      ra, 4(s3)
```

margej, 2019-10-15:
Is --isa=rv64gc expected given my setting file:
riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV32C};

taoliug, 2019-10-15:
Can you run with "--isa rv32imc"?

margej, 2019-10-15:
That results in:
cc1: error: ABI requires -march=rv64
So trying with --mabi ilp32 also ...

margej, 2019-10-15:
GCC is still being called with -march=rv64im so we now get:
cc1: error: ABI requires -march=rv32

Also adding --gcc_opts='-march=rv32imc' is looking promising on a single test

margej, 2019-10-15:
Mmm,  then this one fails "riscv_non_compressed_instr_test", presumably because we forced the 'C' extension...

taoliug, 2019-10-15:
What's the failure for riscv_non_compressed_instr_test?

taoliug, 2019-10-16:
I am closing issue as the original problem is solved. Please file a new issue if needed.

closed 2019-10-16 by taoliug

margej, 2019-10-16:
>python3 run.py -tn riscv_non_compressed_instr_test --isa rv32imc -m ilp32 --gcc_opts='-march=rv32imc'
:
cc1: error: ABI requires -march=rv32