chipsalliance/riscv-dv #223: riscv_non_compressed_instr_test forces rv64 using 'gcc_opts'

margej, 2019-10-16:
This makes it incompatible with --isa rv32* -m ilp32, which is required to make an XLEN=32 config work.

Failing invocation:
`python3 run.py --isa rv32imc -m ilp32 -tn riscv_non_compressed_instr_test`

Failure:
```
Wed, 16 Oct 2019 12:04:04 INFO     Compiling out_2019-10-16/asm_tests/riscv_non_compressed_instr_test_0.S
cc1: error: ABI requires -march=rv32
```
The following resolves the issue for rv32, but clearly breaks the default config, so this should ideally be done dynamically based on the runtime args to run.py:

`--- a/yaml/testlist.yaml
+++ b/yaml/testlist.yaml
@@ -99,9 +100,11 @@
   gen_opts: >
     +disable_compressed_instr=1
   gcc_opts: >
-    -march=rv64im
+    -march=rv32im
`

taoliug, 2019-10-16:
Just want to make sure i understand you setup
XLEN=32, and run riscv_non_compressed_instr_test?

You can probably refer to ibex DV setup
https://github.com/lowRISC/ibex/blob/master/dv/uvm/Makefile#L85
https://github.com/lowRISC/ibex/pull/404

margej, 2019-10-16:
Yes, if i set XLEN to 32 in settings and run:
python3 run.py --isa rv32imc -m ilp32 -tn riscv_non_compressed_instr_test
I get the error:
cc1: error: ABI requires -march=rv32
Which I understand the reason for, and I can work around by editing testlist.yaml as above.

My suggestion is that rather than force the GCC -march for this test to a hardcoded value, you could have the script look at the 'isa' argument and remove the "c" from it if it is present (which is all that is really needed here) E.g. isa_to_gcc = re.sub("c", "", isa)

taoliug, 2019-10-16:
This replacement should only be done in the test you want to disable compressed instruction, which you need to tell the compiler through some form of gcc options. 
If your processor doesn't support compressed instruction at all, you should run with rv32im for all tests. This test is only for processors that support C extension, but want to have a dedicated test to run without compressed instructions.

margej, 2019-10-17:
I understand that yes. My core _does_ support compressed instructions, but is 32-bit. I get that we only want to disable "c" for this test, but as it stands this test can only be run out of the box on a 64-bit core. 

The following alternative solution also works well:
```
--- a/yaml/testlist.yaml
+++ b/yaml/testlist.yaml
@@ -99,7 +100,7 @@
   gen_opts: >
     +disable_compressed_instr=1
   gcc_opts: >
-    -march=rv64im
+    -march=rvXXim
   rtl_test: core_base_test

--- a/run.py
+++ b/run.py
@@ -269,8 +269,18 @@ def gcc_compile(test_list, output_dir, isa, mabi, opts):
              -T%s/scripts/link.ld %s -o %s " % \
              (get_env_var("RISCV_GCC"), asm, get_env_var("RISCV_DV_ROOT"),
               get_env_var("RISCV_DV_ROOT"), opts, elf))
       if 'gcc_opts' in test:
         cmd += test['gcc_opts']
+      # If 'wildcard' architecture width is specified, default to that of the 'isa' argument
+      if re.search("-march=rvXX", cmd):
+        if re.search("rv32", isa):
+          cmd = re.sub("rvXX", "rv32", cmd)
+        else:
+          cmd = re.sub("rvXX", "rv64", cmd)
       # If march/mabi is not defined in the test gcc_opts, use the default
       # setting from the command line.
       if not re.search('march', cmd):
```

taoliug, 2019-10-25:
This is now fixed, no gcc_opts is needed anymore

PR #263 (2019-10-25): Fix compressed instruction test setup
Fix #223
closed 2019-10-25 by taoliug