chipsalliance/riscv-dv #224: ISS process not killed in the event of an ISS timeout

margej, 2019-10-16:
Most easily provoked by changing XLEN to 32 in setting/riscv_core_setting.sv and running run.py with no arguments.

This results in most tests timing out in the ISS, and multiple instances of the spike process still running (and worse still actively logging) after run.py exits

While this may not be a valid run (adding --isa rv32imc -m ilp32 should be specified too) there will be genuine instances where ISS timeouts occur (which I guess is why the --iss_timeout arg exists).

Killing all spawned processes on SIGINT would also be a welcome addition.

taoliug, 2019-10-16:
Yes, this is a known issue. Will fix it.

taoliug, 2019-10-16:
This should be fixed by https://github.com/google/riscv-dv/commit/2707df9e41fac5f80ee481e7f8223e07086d7a66

closed 2019-10-16 by taoliug