chipsalliance/riscv-dv #250: Spike sim gets timeout

amitgargmentor, 2019-10-22:
Hi, 

I generated the instruction for RV32I with XLEN=32 and while runing the spike sim, i see that it gets timeout most of the time. 

**Terminal log:**
Tue, 22 Oct 2019 15:28:42 run.py:266   INFO     Compiling out_2019-10-22/asm_tests/riscv_arithmetic_basic_test_0.S
Tue, 22 Oct 2019 15:28:42 run.py:267   DEBUG    /home/amgarg/project/Infact/customer_evaluation/google/riscv-dv-master/riscv_install/bin/riscv64-unknown-elf-gcc -static -mcmodel=medany              -fvisibility=hidden -nostdlib              -nostartfiles out_2019-10-22/asm_tests/riscv_arithmetic_basic_test_0.S              -I./user_extension              -T./scripts/link.ld  -o out_2019-10-22/asm_tests/riscv_arithmetic_basic_test_0.o  -march=rv64gc -mabi=lp64
Tue, 22 Oct 2019 15:28:43 run.py:269   DEBUG    b''
Tue, 22 Oct 2019 15:28:43 run.py:271   INFO     Converting to out_2019-10-22/asm_tests/riscv_arithmetic_basic_test_0.bin
Tue, 22 Oct 2019 15:28:43 run.py:274   DEBUG    b''
Tue, 22 Oct 2019 15:28:43 run.py:86    INFO     Processing ISS setup file : /home/amgarg/project/Infact/customer_evaluation/google/riscv-dv-master/riscv-dv-master/yaml/iss.yaml
Tue, 22 Oct 2019 15:28:43 run.py:91    INFO     Found matching ISS: spike
Tue, 22 Oct 2019 15:28:43 run.py:330   INFO     spike sim log dir: out_2019-10-22/spike_sim
Tue, 22 Oct 2019 15:28:43 run.py:341   INFO     Running spike sim: out_2019-10-22/asm_tests/riscv_arithmetic_basic_test_0.o
Tue, 22 Oct 2019 15:28:43 lib.py:111   DEBUG    /home/amgarg/project/Infact/customer_evaluation/google/riscv-dv-master/riscv_install_spike/bin/spike --isa=rv64gc --log-commits -l out_2019-10-22/asm_tests/riscv_arithmetic_basic_test_0.o &> out_2019-10-22/spike_sim/riscv_arithmetic_basic_test.0.log
Tue, 22 Oct 2019 15:29:08 lib.py:124   ERROR    Timeout[25s]: /home/amgarg/project/Infact/customer_evaluation/google/riscv-dv-master/riscv_install_spike/bin/spike --isa=rv64gc --log-commits -l out_2019-10-22/asm_tests/riscv_arithmetic_basic_test_0.o &> out_2019-10-22/spike_sim/riscv_arithmetic_basic_test.0.log
Tue, 22 Oct 2019 15:29:08 lib.py:127   DEBUG    
Tue, 22 Oct 2019 15:29:08 run.py:343   DEBUG    /home/amgarg/project/Infact/customer_evaluation/google/riscv-dv-master/riscv_install_spike/bin/spike --isa=rv64gc --log-commits -l out_2019-10-22/asm_tests/riscv_arithmetic_basic_test_0.o &> out_2019-10-22/spike_sim/riscv_arithmetic_basic_test.0.log

**I also observed the exception but not sure if this is causing the issue.  Snapshot of the log file:**

3 0x00000000800002e8 (0x01e602b3) x 5 0x000000001eb2d4f7
core   0: 0x00000000800002ec (0x01670633) add     a2, a4, s6
3 0x00000000800002ec (0x01670633) x12 0x0000000221b3b477
core   0: 0x00000000800002f0 (0x00c607b3) add     a5, a2, a2
3 0x00000000800002f0 (0x00c607b3) x15 0x00000004436768ee
core   0: 0x00000000800002f4 (0x01de0bb3) add     s7, t3, t4
3 0x00000000800002f4 (0x01de0bb3) x23 0xfffffe03504be4eb
core   0: 0x00000000800002f8 (0x006a8633) add     a2, s5, t1
3 0x00000000800002f8 (0x006a8633) x12 0x00000003a0e0cab4
core   0: 0x00000000800002fc (0x00f482b3) add     t0, s1, a5
3 0x00000000800002fc (0x00f482b3) x 5 0x0000000343c48aae
core   0: 0x0000000080000300 (0x01c58633) add     a2, a1, t3
3 0x0000000080000300 (0x01c58633) x12 0x00000001fe6d35c1
core   0: 0x0000000080000304 (0x010b09b3) add     s3, s6, a6
3 0x0000000080000304 (0x010b09b3) x19 0x0000000299e3f82a
core   0: 0x0000000080000308 (0x7b200073) dret
core   0: exception trap_illegal_instruction, epc 0x0000000080000308
core   0:           tval 0x0000000000000000
core   0: 0x0000000080001000 (0x0400006f) j       pc + 0x40
3 0x0000000080001000 (0x0400006f)
core   0: 0x0000000080001040 (0x34011173) csrrw   sp, mscratch, sp
3 0x0000000080001040 (0x34011173) x 2 0x000000000000001b
core   0: 0x0000000080001044 (0x00020133) add     sp, tp, zero
3 0x0000000080001044 (0x00020133) x 2 0x000000025de29656
core   0: 0x0000000080001048 (0xf8410113) addi    sp, sp, -124
3 0x0000000080001048 (0xf8410113) x 2 0x000000025de295da
core   0: 0x000000008000104c (0x0000c206) c.swsp  ra, 4(sp)
core   0: exception trap_store_address_misaligned, epc 0x000000008000104c
core   0:           tval 0x000000025de295de
core   0: 0x0000000080001000 (0x0400006f) j       pc + 0x40
3 0x0000000080001000 (0x0400006f)

Can someone please help ?

taoliug, 2019-10-23:
Looks like it's a problem of running RV32I with debug mode enabled. Can you try to run with debug mode disabled?
https://github.com/google/riscv-dv/blob/master/setting/riscv_core_setting.sv#L44

amitgargmentor, 2019-10-23:
Hi Tao,

Thanks for quick reply but i see that suppport_debug_mode = 0 in riscv_core_setting.sv file. 

below is my riscv_core_setting file:

//-----------------------------------------------------------------------------
// Processor feature configuration
//-----------------------------------------------------------------------------
// XLEN
//amit parameter int XLEN = 64;
parameter int XLEN = 32;

// Parameter for SATP mode, set to BARE if address translation is not supported
//amit parameter satp_mode_t SATP_MODE = SV39;
parameter satp_mode_t SATP_MODE = BARE;

// Supported Privileged mode
privileged_mode_t supported_privileged_mode[] = {USER_MODE, SUPERVISOR_MODE, MACHINE_MODE};

// Unsupported instructions
riscv_instr_name_t unsupported_instr[];

// ISA supported by the processor
//riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV64I, RV64M, RV32C, RV64C, RV32A, RV64A};
riscv_instr_group_t supported_isa[$] = {RV32I};

// Interrupt mode support
mtvec_mode_t supported_interrupt_mode[$] = {DIRECT, VECTORED};

// The number of interrupt vectors to be generated, only used if VECTORED interrupt mode is
// supported
int max_interrupt_vector_num = 16;

// Debug mode support
**bit support_debug_mode = 0;**

// Support delegate trap to user mode
bit support_umode_trap = 0;

// Support sfence.vma instruction
bit support_sfence = 1;

// ----------------------------------------------------------------------------
// Previleged CSR implementation
// ----------------------------------------------------------------------------

// Implemented previlieged CSR list
parameter privileged_reg_t implemented_csr[] = {
    // User mode CSR
    USTATUS,    // User status
    UIE,        // User interrupt-enable register
    UTVEC,      // User trap-handler base address
    USCRATCH,   // Scratch register for user trap handlers
    UEPC,       // User exception program counter
    UCAUSE,     // User trap cause
    UTVAL,      // User bad address or instruction
    UIP,        // User interrupt pending
    // Supervisor mode CSR
    SSTATUS,    // Supervisor status
    SEDELEG,    // Supervisor exception delegation register
    SIDELEG,    // Supervisor interrupt delegation register
    SIE,        // Supervisor interrupt-enable register
    STVEC,      // Supervisor trap-handler base address
    SCOUNTEREN, // Supervisor counter enable
    SSCRATCH,   // Scratch register for supervisor trap handlers
    SEPC,       // Supervisor exception program counter
    SCAUSE,     // Supervisor trap cause
    STVAL,      // Supervisor bad address or instruction
    SIP,        // Supervisor interrupt pending
    SATP,       // Supervisor address translation and protection
    // Machine mode mode CSR
    MVENDORID,  // Vendor ID
    MARCHID,    // Architecture ID
    MIMPID,     // Implementation ID
    MHARTID,    // Hardware thread ID
    MSTATUS,    // Machine status
    MISA,       // ISA and extensions
    MEDELEG,    // Machine exception delegation register
    MIDELEG,    // Machine interrupt delegation register
    MIE,        // Machine interrupt-enable register
    MTVEC,      // Machine trap-handler base address
    MCOUNTEREN, // Machine counter enable
    MSCRATCH,   // Scratch register for machine trap handlers
    MEPC,       // Machine exception program counter
    MCAUSE,     // Machine trap cause
    MTVAL,      // Machine bad address or instruction
    MIP         // Machine interrupt pending
};

// ----------------------------------------------------------------------------
// Supported interrupt/exception setting, used for functional coverage
// ----------------------------------------------------------------------------

parameter interrupt_cause_t implemented_interrupt[] = {
    U_SOFTWARE_INTR,
    S_SOFTWARE_INTR,
    M_SOFTWARE_INTR,
    U_TIMER_INTR,
    S_TIMER_INTR,
    M_TIMER_INTR,
    U_EXTERNAL_INTR,
    S_EXTERNAL_INTR,
    M_EXTERNAL_INTR
};

parameter exception_cause_t implemented_exception[] = {
    INSTRUCTION_ADDRESS_MISALIGNED,
    INSTRUCTION_ACCESS_FAULT,
    ILLEGAL_INSTRUCTION,
    BREAKPOINT,
    LOAD_ADDRESS_MISALIGNED,
    LOAD_ACCESS_FAULT,
    STORE_AMO_ADDRESS_MISALIGNED,
    STORE_AMO_ACCESS_FAULT,
    ECALL_UMODE,
    ECALL_SMODE,
    ECALL_MMODE,
    INSTRUCTION_PAGE_FAULT,
    LOAD_PAGE_FAULT,
    STORE_AMO_PAGE_FAULT
};

Thanks,
Amit

amitgargmentor, 2019-10-23:
i actually ran again with both ovpsim and spike sim and i see the trap exception on c.j instruction . Is there some setting needs to be done  ? 

**instruction:**
mepc_setup:       
                  la x10, _init
                  csrw mepc, x10
                  **j init_machine_mode**

**SPIKE SIM:**
core   0: 0x00000000800000f4 (0x30551073) csrw    mtvec, a0
3 0x00000000800000f4 (0x30551073)
core   0: 0x00000000800000f8 (0x00000517) auipc   a0, 0x0
3 0x00000000800000f8 (0x00000517) x10 0x00000000800000f8
core   0: 0x00000000800000fc (0x00e50513) addi    a0, a0, 14
3 0x00000000800000fc (0x00e50513) x10 0x0000000080000106
core   0: 0x0000000080000100 (0x34151073) csrw    mepc, a0
3 0x0000000080000100 (0x34151073)
**core   0: 0x0000000080000104 (0x0000a625) c.j     pc + 808
core   0: exception trap_illegal_instruction, epc 0x0000000080000104
core   0:           tval 0x0000000000000000
core   0: 0x0000000080001000 (0x0400006f) j       pc + 0x40
3 0x0000000080001000 (0x0400006f)**
core   0: 0x0000000080001040 (0x34011173) csrrw   sp, mscratch, sp
3 0x0000000080001040 (0x34011173) x 2 0x0000000000000000
core   0: 0x0000000080001044 (0x00020133) add     sp, tp, zero
3 0x0000000080001044 (0x00020133) x 2 0x000000008000ce7c

OVPSIM:

Info 70: 'riscvOVPsim/cpu', 0x00000000800000f8(mepc_setup): 00000517 auipc   a0,0x0
Info   a0 0000000080001001 -> 00000000800000f8
Info 71: 'riscvOVPsim/cpu', 0x00000000800000fc(mepc_setup+4): 00e50513 addi    a0,a0,14
Info   a0 00000000800000f8 -> 0000000080000106
Info 72: 'riscvOVPsim/cpu', 0x0000000080000100(mepc_setup+8): 34151073 csrw    mepc,a0
Info   mepc 0000000000000000 -> 0000000080000106
**Info 73: 'riscvOVPsim/cpu', 0x0000000080000104(mepc_setup+c): a625     j       8000042c
Warning (RISCV_ADF) CPU 'riscvOVPsim/cpu' 0x80000104 a625     j       8000042c: Illegal instruction - extension C (compressed instructions) absent or inactive
Info   mstatus 0000000a00000000 -> 0000000a00001800**
Info   mepc 0000000080000106 -> 0000000080000104
Info   mcause 0000000000000000 -> 0000000000000002
Info   mtval 0000000000000000 -> 000000000000a625
Info 74: 'riscvOVPsim/cpu', 0x0000000080001000(mtvec_

Thank,s
Amit

taoliug, 2019-10-23:
It seems you are still running with -march=rv64gc -mabi=lp64
Can you run with "--isa rv32i --mabi ilp32"?

amitgargmentor, 2019-10-23:
i ran with --isa rv32i --mabi ilp32 , one thing which i observed that run.py script add option "+disable_compressed_instr=1" which is coming in new line and as well there was typo in passing define which i fixed and below command is run as vsim:

**vsim -c -do "/home/amgarg/project/Infact/customer_evaluation/google/riscv-dv-master/riscv-dv-master/questa_sim.tcl" design_opt "+UVM_VERBOSITY=UVM_HIGH" -qwavedb=+signal+memory+transaction+class+uvm_schematic -sv_seed 1125189848 "+UVM_TESTNAME=riscv_instr_base_test" "+num_of_tests=1" "+start_idx=0" "+asm_file_name=out_2019-10-23/asm_tests/riscv_arithmetic_basic_test" -l out_2019-10-23/sim_riscv_arithmetic_basic_test_0.log "+disable_compressed_instr=1" "+instr_cnt=200" "+num_of_sub_program=0" "+no_fence=1" "+no_data_page=1b1 +no_branch_jump=1b1" "+boot_mode=m"**

After instruction has been generated, below commands are run for compiling the assembly file and running the SPIKE sim:

**Wed, 23 Oct 2019 11:58:53 run.py:266   INFO     Compiling out_2019-10-23/asm_tests/riscv_arithmetic_basic_test_0.S
Wed, 23 Oct 2019 11:58:53 run.py:267   DEBUG    riscv64-unknown-elf-gcc -static -mcmodel=medany              -fvisibility=hidden -nostdlib              -nostartfiles out_2019-10-23/asm_tests/riscv_arithmetic_basic_test_0.S              -I./user_extension              -T./scripts/link.ld  -o out_2019-10-23/asm_tests/riscv_arithmetic_basic_test_0.o  -march=rv32i -mabi=ilp32
Wed, 23 Oct 2019 11:58:53 run.py:269   DEBUG    b''
Wed, 23 Oct 2019 11:58:53 run.py:271   INFO     Converting to out_2019-10-23/asm_tests/riscv_arithmetic_basic_test_0.bin
Wed, 23 Oct 2019 11:58:53 run.py:274   DEBUG    b''
Wed, 23 Oct 2019 11:58:53 run.py:86    INFO     Processing ISS setup file : ./riscv-dv-master/riscv-dv-master/yaml/iss.yaml
Wed, 23 Oct 2019 11:58:53 run.py:91    INFO     Found matching ISS: spike
Wed, 23 Oct 2019 11:58:53 run.py:330   INFO     spike sim log dir: out_2019-10-23/spike_sim
Wed, 23 Oct 2019 11:58:53 run.py:341   INFO     Running spike sim: out_2019-10-23/asm_tests/riscv_arithmetic_basic_test_0.o
Wed, 23 Oct 2019 11:58:53 lib.py:111   DEBUG    spike --isa=rv32i --log-commits -l out_2019-10-23/asm_tests/riscv_arithmetic_basic_test_0.o &> out_2019-10-23/spike_sim/riscv_arithmetic_basic_test.0.log**

Below is the top of assembly file generated, the error is getting generated while calling j init_machine_mode instruction which cause exception due to misalign address. 

.include "user_define.h"
.globl _start
.section .text
.option norvc;
_start:
                  li x0, 0x80000000
                  li x1, 0xf
                  li x2, 0xf74ada2d
                  li x3, 0x9203f518
                  li x4, 0xf7a592c4
                  li x5, 0xf81f01be
                  li x6, 0xfe049fd2
                  li x7, 0xfe2c15b7
                  li x8, 0xd
                  li x9, 0x0
                  li x10, 0x0
                  li x11, 0x80000000
                  li x12, 0x0
                  li x13, 0x80000000
                  li x14, 0xf2df95da
                  li x15, 0x0
                  li x16, 0x7
                  li x17, 0x97ca5bcd
                  li x18, 0x0
                  li x19, 0x80000000
                  li x20, 0x0
                  li x21, 0x0
                  li x22, 0xfad5526e
                  li x23, 0xf3d5daae
                  li x24, 0x8912bc2
                  li x25, 0xe03da511
                  li x26, 0x80000000
                  li x27, 0xfe0337d2
                  li x28, 0x0
                  li x29, 0x1
                  li x30, 0xd2aa4e6e
                  li x31, 0x80000000
                  li x15, 0x20040100
                  csrw misa, x15
kernel_sp:        
                  la tp, _kernel_stack_end

trap_vec_init:    
                  la a0, mtvec_handler
                  ori a0, a0, 1
                  csrw 0x305, a0 # MTVEC

mepc_setup:       
                  la x10, _init
                  csrw mepc, x10
                  **j init_machine_mode**

_init:            
                  la sp, _user_stack_end
_main:            sub        s6, s6, a7
                  srli       s3, s3, 0

Below is the exception occured:
core   0: 0xffffffff80000430 (0x30051073) csrw    mstatus, a0
3 0x80000430 (0x30051073)
**core   0: 0xffffffff80000434 (0x00004501) c.li    a0, 0
core   0: exception trap_illegal_instruction, epc 0xffffffff80000434**
core   0:           tval 0x0000000000000000
core   0: 0x0000000080001000 (0x0400006f) j       pc + 0x40
3 0x80001000 (0x0400006f)
core   0: 0xffffffff80001040 (0x34011173) csrrw   sp, mscratch, sp
3 0x80001040 (0x34011173) x 2 0x00000000

I have very limited knowledge of risc compiler and setup, hence puuting all the details which may be required.  

Thanks
,Amit

amitgargmentor, 2019-10-23:
The main issue i see is mis alignment of address during jump or sw instruction which is added as part of set up instruction. so we do anything specific to make sure the instruction are word aligned so that address is word aligned ? 

mepc_setup:       
                  la x10, _init
                  csrw mepc, x10
                  **j init_machine_mode**

write_tohost:     
                  **sw gp, tohost, t5**

mmode_intr_vector_1:
                  csrrw sp, 0x340, sp
                  add sp, tp, zero
                  1: addi sp, sp, -124
                  **sw  x1, 4(sp)**
                  sw  x2, 8(sp)
                  sw  x3, 12(sp)
                  sw  x4, 16(sp)
                  sw  x5, 20(sp)

taoliug, 2019-10-25:
There is a lot of information in this bug. Can you let me know what's your run command, and your local modifications?

taoliug, 2019-10-31:
Closing this bug for inactivity, please file a new bug if you still see this issue.

closed 2019-10-31 by taoliug