chipsalliance/riscv-dv #255: can't generate csv from generated spike log

syibin, 2019-10-24:
Hi,

I'm trying to setup the whole End-to-end RTL and ISS co-simulation flow for IBEX RISC-V core verification these days.

I'm close to run through the whole flow, but stuck by generating the csv from generated spike log.
(the csv generated by RTL sim log is ready)

The spike log looks like:
core   0: 0x0000000000001010 (0x00028067) jr      t0
core   0: 0xffffffff80000080 (0xf15df037) lui     zero, 0xf15df
core   0: 0xffffffff80000084 (0x47500013) li      zero, 1141
core   0: 0xffffffff80000088 (0x800000b7) lui     ra, 0x80000
core   0: 0xffffffff8000008c (0xf207b137) lui     sp, 0xf207b
core   0: 0xffffffff80000090 (0x61a10113) addi    sp, sp, 1562
core   0: 0xffffffff80000094 (0xfb3931b7) lui     gp, 0xfb393
core   0: 0xffffffff80000098 (0x5cd18193) addi    gp, gp, 1485
core   0: 0xffffffff8000009c (0x80000237) lui     tp, 0x80000
core   0: 0xffffffff800000a0 (0x00004281) c.li    t0, 0
core   0: 0xffffffff800000a2 (0xfaff6337) lui     t1, 0xfaff6
core   0: 0xffffffff800000a6 (0x54030313) addi    t1, t1, 1344
core   0: 0xffffffff800000aa (0x00004381) c.li    t2, 0

I checked the spike_log_to_trace_csv.py, based on my understanding, the workflow is: it will try to extract instruction information using CORE_RE, then read nextline, and check whether nextline matches RD_RE or not. If matches, RD information will be extracted and then written to csv.
But my question is according to spike.log, it will never match RD_RE as shown in below (I also tried to run this script standalone, and no match). Where does this regular expression comes from?
RD_RE    = re.compile(r"(?P<pri>\d) 0x(?P<addr>[a-f0-9]+?) " \
                      "\((?P<bin>.*?)\) x\s*(?P<reg>\d*?) 0x(?P<val>[a-f0-9]+)")

Some env info:
1. spike version: Spike RISC-V ISA Simulator 1.0.0
2. os: ubuntu 18.04.3
3.Python: 3.6.8

Look forward to your reply! :)

syibin, 2019-10-24:
One More Question: I check the csv generated from RTL sim log which matches my expectation that every instruction has been recorded.
While spike_log_to_trace_csv.py will read next line inside a loop, so that even the instruction which matches RD_RE is recorded, it will be 2/4/6, etc. and 1/3/5 which matches CORE_RE will be skipped /not written to csv? do I misunderstand the workflow?

Thanks
Yibin

taoliug, 2019-10-24:
Have you installed spike with "--enable-commitlog" ?

taoliug, 2019-10-24:
It seems to me there's no commit information generated by spike. Please make sure you are using the latest version of spike, and build it with --enable-commitlog.

syibin, 2019-10-25:
Hi, taoliug

Thanks for your reply. I did use latest version of spike, but built it without enable-commitlog option. I will have a try with this option as soon as I can and keep you updated.

Thanks
Yibin

syibin, 2019-10-25:
Hi, taoliug

--enable-commitlog works for me, now I can see the expected csv with all the contents! Thanks again for your feedback! :)

Thanks
Yibin

taoliug, 2019-10-25:
Good to know it works

closed 2019-10-25 by taoliug