chipsalliance/riscv-dv #256: Spike sim hanged and timeout but log file size grows

amitgargmentor, 2019-10-24:
Hi,

if i run the spike simulation with command as shown below and if there is any bad instruction whcih cause the exception, the simulation gets timeout which is good but log file is still getting updated and grows as GB size.  

spike --isa=rv32i --log-commits -l out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.o &> out_2019-10-24/spike_sim/riscv_arithmetic_basic_test.0.log

Have anyone faced this issue? Some time, there is stale file .nfs* created in spike_sim directory which grows in size. 

Has anyone seen this issue?

taoliug, 2019-10-24:
if you use ctrl+c to kill the spike simulation while it's running, I think the spike subprocess won't be killed based on the current implementation. However, if the spike timeout after 25s, I'd expect run.py kill the spike subprocess correctly and the log should not grow afterwards. You can use --iss_timeout=xx to set a shorter timeout period. Normally spike should finish within 5s.

amitgargmentor, 2019-10-24:
Hi ,

I didn’t use the ctrl+c and I wait for 25s for timeout ( sure, I can reduce the timeout) but the log file is still getting dumped and grow in size.

Thanks,
Amit

amitgargmentor, 2019-10-24:
I cloned the latest  repository and the spike sim doesn't grow in size after timeout . 

Thanks!!

closed 2019-10-24 by amitgargmentor