chipsalliance/riscv-dv #257: no python3-bitstring available for CentOS7

weedmank, 2019-10-24:


weedmank, 2019-10-24:
Directions say I need this. 
   1) sudo apt-get install python3-bitstring (or your OS-specific package manager)

I have python3 installed - installed it yesterday:
[engr@localhost riscv-dv]$ which python3
/usr/bin/python3

I tried this:
[engr@localhost riscv-dv]$ sudo yum install python3-bitstring
Loaded plugins: fastestmirror, langpacks
Loading mirror speeds from cached hostfile
 * base: mirror.web-ster.com
 * centos-sclo-rh: mirror.web-ster.com
 * centos-sclo-sclo: mirror.web-ster.com
 * epel: mirrors.syringanetworks.net
 * extras: mirror.web-ster.com
 * updates: repo1.sea.innoscale.net
No package python3-bitstring available.
Error: Nothing to do

taoliug, 2019-10-24:
Can you run the instruction generator without python3-bitstring? This package is only needed for one CSR test, I may change the description to make this requirement optional.

weedmank, 2019-10-25:
I tried this:
[engr@localhost riscv-dv]$ python3 run.py --test=riscv_arithmetic_basic_test

Thu, 24 Oct 2019 17:07:11 INFO     Processing regression test list : /home/engr/riscv-dv/riscv-dv/yaml/testlist.yaml, test: riscv_arithmetic_basic_test
Traceback (most recent call last):
  File "/home/engr/riscv-dv/riscv-dv/scripts/lib.py", line 54, in read_yaml
    yaml_data = yaml.safe_load(f)
AttributeError: module 'yaml' has no attribute 'safe_load'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "run.py", line 563, in <module>
    main()
  File "run.py", line 535, in main
    process_regression_list(args.testlist, args.test, args.iterations, matched_list)
  File "/home/engr/riscv-dv/riscv-dv/scripts/lib.py", line 173, in process_regression_list
    yaml_data = read_yaml(testlist)
  File "/home/engr/riscv-dv/riscv-dv/scripts/lib.py", line 55, in read_yaml
    except yaml.YAMLError as exc:
AttributeError: module 'yaml' has no attribute 'YAMLError'

taoliug, 2019-10-25:
Can you check if this comment helps?
https://github.com/google/riscv-dv/issues/86#issuecomment-541604432

weedmank, 2019-10-25:
The "sudo pip3 ..." seems to have solved the YAML problem. Now I have figure out how to get gcc and questasim setup for my RV32i. Should I start another thread for this?

[engr@localhost riscv-dv]$ python3 run.py --test riscv_arithmetic_basic_test --simulator questa
Thu, 24 Oct 2019 18:05:51 INFO Processing regression test list : /home/engr/riscv-dv/riscv-dv/yaml/testlist.yaml, test: riscv_arithmetic_basic_test
Thu, 24 Oct 2019 18:05:51 INFO Found matched tests: riscv_arithmetic_basic_test, iterations:2
Thu, 24 Oct 2019 18:05:51 INFO Processing simulator setup file : /home/engr/riscv-dv/riscv-dv/yaml/simulator.yaml
Thu, 24 Oct 2019 18:05:51 INFO Found matching simulator: questa
Thu, 24 Oct 2019 18:05:51 INFO Building RISC-V instruction generator
Thu, 24 Oct 2019 18:05:54 INFO Running RISC-V instruction generator
Thu, 24 Oct 2019 18:05:54 INFO Generating 2 riscv_arithmetic_basic_test
Thu, 24 Oct 2019 18:05:54 INFO Running riscv_arithmetic_basic_test with 1 batches
Thu, 24 Oct 2019 18:05:54 INFO Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Thu, 24 Oct 2019 18:05:57 INFO Compiling out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.S
riscv64-unknown-elf-gcc: error: out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.S: No such file or directory
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.

taoliug, 2019-10-25:
Yes, please file a separate bug for this.
Here's the instruction for gcc setup
https://github.com/google/riscv-dv#setup-riscv-gcc-compiler-toolchain

taoliug, 2019-10-25:
use "-v" to turn on verbose logging if you see any failure

closed 2019-10-25 by taoliug