chipsalliance/riscv-dv #259: ovpsim gets hanged and reruning the same instruction again

amitgargmentor, 2019-10-24:
Hi, 

i clone to latest repostiry today and i ran the ovpsim simulation with below command:
python3 run.py --test riscv_arithmetic_basic_test --v --iss ovpsim --isa rv32i --mabi ilp32

OVPSIM command is as below:
**/home/amgarg/project/Infact/customer_evaluation/google/riscv-dv-master/riscv-ovpsim/bin/Linux64/riscvOVPsim.exe --controlfile ../riscv-dv/target/rv32i/riscvOVPsim.ic --objfilenoentry out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.o --override riscvOVPsim/cpu/PMP_registers=0 --override riscvOVPsim/cpu/simulateexceptions=T --trace --tracechange --traceshowicount --tracemode --traceregs --finishafter 1000000 &> out_2019-10-24/ovpsim_sim/riscv_arithmetic_basic_test.0.log**

i did few changes in riscv_core_setting file as attached to run rv32i simulation. 

the issue i see that after the test is done, it is restarting again which was not the issue earlier.  
please find the attached log file. 
[riscv_arithmetic_basic_test.0.log](https://github.com/google/riscv-dv/files/3769647/riscv_arithmetic_basic_test.0.log)

[image]

taoliug, 2019-10-24:
Does it work with spike?
Can you also try python3 run.py --test riscv_arithmetic_basic_test --v --iss ovpsim --target rv32i

amitgargmentor, 2019-10-24:
Yes, it works with spike. please find the log for both spike and ovpsim. 

Spike:
[riscv_arithmetic_basic_test.0.log](https://github.com/google/riscv-dv/files/3769742/riscv_arithmetic_basic_test.0.log)

OVPSIM:

[riscv_arithmetic_basic_test.0.log](https://github.com/google/riscv-dv/files/3769750/riscv_arithmetic_basic_test.0.log)

taoliug, 2019-10-24:
There are some recent submit breaks the ovpsim sim, will address it shortly.

amitgargmentor, 2019-10-24:
Also, running this : python3 run.py --test riscv_arithmetic_basic_test --v --iss ovpsim --target rv32i doesn't help.

taoliug, 2019-10-25:
I think this is now fixed, please sync and retry.

taoliug, 2019-10-25:
Closing this issue for now, let me know if you still see this problem

closed 2019-10-25 by taoliug

amitgargmentor, 2019-10-25:
Hi Tao,

I picked up riscvopsim.ic file in setting directory and copy the latest target directory and i see that ovpsim is not working fine now.  There is <variant> passed as it is which is causing issue. 

 DEBUG    /riscvOVPsim.exe --controlfile ../riscv-dv/target/<variant>/riscvOVPsim.ic --objfilenoentry out_2019-10-25/asm_tests/riscv_arithmetic_basic_test_0.o --override riscvOVPsim/cpu/PMP_registers=0 --override riscvOVPsim/cpu/simulateexceptions=T --trace --tracechange --traceshowicount --tracemode --traceregs --finishafter 1000000 &> out_2019-10-25/ovpsim_sim/riscv_arithmetic_basic_test.0.log
**Fri, 25 Oct 2019 10:15:37 lib.py:127   DEBUG    /bin/sh: variant: No such file or directory**

Fri, 25 Oct 2019 10:15:37 run.py:362   DEBUG    riscvOVPsim.exe --controlfile ../riscv-dv/target**/<variant>/**riscvOVPsim.ic --objfilenoentry out_2019-10-25/asm_tests/riscv_arithmetic_basic_test_0.o --override riscvOVPsim/cpu/PMP_registers=0 --override riscvOVPsim/cpu/simulateexceptions=T --trace --tracechange --traceshowicount --tracemode --traceregs --finishafter 1000000 &> out_2019-10-25/ovpsim_sim/riscv_arithmetic_basic_test.0.log

amitgargmentor, 2019-10-25:
Please ignore the above request. i merged the iss.yaml change and issue is resolved. 
Thanks for the help Tao.