chipsalliance/riscv-dv #262: GCC problem

weedmank, 2019-10-25:
I installed the riscv-gnu-toolchain using this config because I want to be able to do compiles for an RV32i cpu I'm designing in RTL System Verilog:

./configure --prefix=/opt/riscv --enable-multilib
make

I set my .bashrc with the following:

export RISCV_TOOLCHAIN=/opt/riscv
export RISCV_GCC="$RISCV_TOOLCHAIN/bin/riscv64-unknown-elf-gcc"
export RISCV_OBJCOPY="$RISCV_TOOLCHAIN/bin/riscv64-unknown-elf-objcopy"
export SPIKE_PATH=$RISCV_TOOLCHAIN/bin

Then opened a new terminal window, changed to the riscv-dv folder and I tried this:

[engr@localhost riscv-dv]$ python3 run.py --test riscv_arithmetic_basic_test --simulator questa
Thu, 24 Oct 2019 19:28:37 INFO     Processing regression test list : /home/engr/riscv-dv/riscv-dv/yaml/testlist.yaml, test: riscv_arithmetic_basic_test
Thu, 24 Oct 2019 19:28:37 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Thu, 24 Oct 2019 19:28:37 INFO     Processing simulator setup file : /home/engr/riscv-dv/riscv-dv/yaml/simulator.yaml
Thu, 24 Oct 2019 19:28:37 INFO     Found matching simulator: questa
Thu, 24 Oct 2019 19:28:37 INFO     Building RISC-V instruction generator
Thu, 24 Oct 2019 19:28:37 INFO     Running RISC-V instruction generator
Thu, 24 Oct 2019 19:28:37 INFO     Generating 2 riscv_arithmetic_basic_test
Thu, 24 Oct 2019 19:28:37 INFO     Running riscv_arithmetic_basic_test with 1 batches
Thu, 24 Oct 2019 19:28:37 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Thu, 24 Oct 2019 19:28:38 INFO     Compiling out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.S
riscv64-unknown-elf-gcc: error: out_2019-10-24/asm_tests/riscv_arithmetic_basic_test_0.S: No such file or directory
riscv64-unknown-elf-gcc: fatal error: no input files
compilation terminated.

How should I set this for compiling for RV32i?

taoliug, 2019-10-25:
I think the problem is the program is not generated properly. Please check compile.log under output directory. You can also run with -v for verbose logging.

weedmank, 2019-10-25:
I get this in my compile.log:

QuestaSim-64 vlog 10.7-QCF Compiler 2018.03
Start time: 19:28:37 on Oct 24,2019
vlog -64 "+incdir+/home/engr/riscv-dv/riscv-dv/setting" "+incdir+/home/engr/riscv-dv/riscv-dv/user_extension" -access=rwc -f /ho
me/engr/riscv-dv/riscv-dv/files.f -sv -mfcu -cuname design_cuname "+define+UVM_REGEX_NO_DPI" -writetoplevels /home/engr/riscv-dv
/riscv-dv/out_2019-10-24/top.list -l /home/engr/riscv-dv/riscv-dv/out_2019-10-24/compile.log 
-- Compiling package riscv_signature_pkg
-- Compiling package riscv_instr_pkg
** Error: ./src/riscv_instr_pkg.sv(22): Cannot find `include file "uvm_macros.svh" in directories:
    /home/engr/riscv-dv/riscv-dv/setting, /home/engr/riscv-dv/riscv-dv/user_extension, ./src, ./test, /home/engr/Mentor/Questa_C
DC_Formal_10.7a/share/modeltech/ovm-2.1.2/../verilog_src/ovm-2.1.2/src
** Error: (vlib-35) Failed to create directory "/questasim/uvm-1.2".
No such file or directory. (errno = ENOENT)
** Error (suppressible): (vlib-19) Failed to access library '/questasim/uvm-1.2' at "/questasim/uvm-1.2".
No such file or directory. (errno = ENOENT)
** Error: (vlib-35) Failed to create directory "/questasim/uvm-1.2".
No such file or directory. (errno = ENOENT)
** Error (suppressible): (vlib-19) Failed to access library '/questasim/uvm-1.2' at "/questasim/uvm-1.2".
No such file or directory. (errno = ENOENT)

There is no /home/engr/Mentor/Questa_C -> its /home/engr/Mentor/Questa_CDC_Formal_10.7a  There's also no subfolder called /questasim/uvm-1.2.  They are in a different folder for this Questa.  How do I specify that?

I find this :

[engr@localhost Questa_CDC_Formal_10.7a]$ ls linux_x86_64/modeltech/
avm          ieee_env  linuxpe         mpich2     registerassistantuvm  synopsys    uvm_reg-1.1   vital1995  win32pe_edu
bin          ieeepure  linux_x86_64    osver      RELEASE_NOTES         tcl         vco           vital2000  win32pe_starter

[engr@localhost Questa_CDC_Formal_10.7a]$ ls linux_x86_64/modeltech/uvm-1.1d/
_dpi  _info  _lib1_0.qdb  _lib1_0.qpg  _lib1_0.qtl  _lib.qdb  linux  linux_x86_64  _vmake  win32  win64
[engr@localhost Questa_CDC_Formal_10.7a]$ ls linux_x86_64/modeltech/uvm-1.1d/linux
uvm_dpi.so

[engr@localhost Questa_CDC_Formal_10.7a]$ ls linux_x86_64/modeltech/uvmc-2.3.1/
_dpi  _info  _lib1_0.qdb  _lib1_0.qpg  _lib1_0.qtl  _lib.qdb  linux  linux_x86_64
[engr@localhost Questa_CDC_Formal_10.7a]$ 

cov_src      include   linux_x86_64pe  osvvm      RELEASE_NOTES.html    tcl.fs      verilog       vital2.2b  win64
docs         infact    mc2_lib         ovm-2.1.2  rnm                   upf_lib     verilog_src   vm_src     win64pe
drill_src    keyring   mgc_ams         pa_lib     std                   upf_src     vhdlopt_lib   vovl_src
floatfixlib  lib       modelsim.ini    perl_src   std_developerskit     uvm-1.1d    vhdl_src      win32
ieee         LICENSE   modelsim_lib    plat       sv_std                uvmc-2.3.1  vh_ux01v_lib  win32pe

taoliug, 2019-10-25:
Please make sure UVM_HOME is setup properly. BTW, uvm 1.2 is needed to run this generator.

weedmank, 2019-10-25:
I installed uvm-1.2 folder and added UVM_HOME to my .bashrc and also added it to PATH:

export RISCV_TOOLCHAIN=/opt/riscv
#export RISCV_GCC="$RISCV_TOOLCHAIN/bin/riscv64-unknown-elf-gcc --with-arch=rv32imac --with-abi=ilp32"
export RISCV_GCC="$RISCV_TOOLCHAIN/bin/riscv64-unknown-elf-gcc"
export RISCV_OBJCOPY="$RISCV_TOOLCHAIN/bin/riscv64-unknown-elf-objcopy"
export SPIKE_PATH=$RISCV_TOOLCHAIN/bin
export UVM_HOME=/home/engr/uvm/uvm-1.2
export QHOME=/home/engr/Mentor/Questa_CDC_Formal_10.7a/linux_x86_64
export QUESTAHOME=$QHOME/modeltech
export LMTOOL=$QHOME/mgls/bin
export PATH=$UVM_HOME:$QHOME/bin:$QUESTAHOME/bin:$LMTOOL:$PATH

But it still doesn't appear to search the UVM_HOME path as shown below. It did search some of the PATHs. I can see the missing "uvm_macros.svh" in directory /home/engr/uvm/uvm-1.2

QuestaSim-64 vlog 10.7-QCF Compiler 2018.03 Mar 14 2018
Start time: 09:24:10 on Oct 25,2019
vlog -64 "+incdir+/home/engr/riscv-dv/setting" "+incdir+/home/engr/riscv-dv/user_extension" -access=rwc -f /home/eng
r/riscv-dv/files.f -sv -mfcu -cuname design_cuname "+define+UVM_REGEX_NO_DPI" -writetoplevels /home/engr/riscv-dv/ou
t_2019-10-25/top.list -l /home/engr/riscv-dv/out_2019-10-25/compile.log 
-- Compiling package riscv_signature_pkg
-- Compiling package riscv_instr_pkg
** Error: ./src/riscv_instr_pkg.sv(22): Cannot find `include file "uvm_macros.svh" in directories:
    /home/engr/riscv-dv/setting, /home/engr/riscv-dv/user_extension, ./src, ./test, /home/engr/Mentor/Questa_CDC_For
mal_10.7a/share/modeltech/ovm-2.1.2/../verilog_src/ovm-2.1.2/src
** Error: (vlib-35) Failed to create directory "/questasim/uvm-1.2".
No such file or directory. (errno = ENOENT)
** Error (suppressible): (vlib-19) Failed to access library '/questasim/uvm-1.2' at "/questasim/uvm-1.2".
No such file or directory. (errno = ENOENT)
** Error: (vlib-35) Failed to create directory "/questasim/uvm-1.2".

amitgargmentor, 2019-10-25:
hi,

Please look at file yaml/simulator for vmap command ,  you need to set QUESTA_HOME env varibale  instead of QUESTAHOME.

weedmank, 2019-10-25:
OK. I changed QUESTAHOME to QUESTA_HOME which helped for that issue, but it's still not seeing the uvm1.2 folder info:

[engr@localhost ~]$ echo $UVM_HOME
/home/engr/uvm/uvm-1.2
[engr@localhost ~]$ ls $UVM_HOME
bin   examples  LICENSE.txt  README.md   release-notes.txt  UVM_Reference.html
docs  LICENSE   NOTICE.txt   README.txt  src
[engr@localhost ~]$ ls $UVM_HOME/src
base   dap         dpi     reg  tags  tlm2            uvm_pkg.sv
comps  deprecated  macros  seq  tlm1  uvm_macros.svh  uvm.sv
[engr@localhost ~]$ 
[engr@localhost riscv-dv]$ echo $PATH
/Mentor/Questa_inFact_10.7a/infact/bin:/home/engr/uvm/uvm-1.2:/home/engr/Mentor/Questa_CDC_Formal_10.7a/linux_x86_64/bin:/home/engr/Mentor/Questa_CDC_Formal_10.7a/linux_x86_64/modeltech/bin:/home/engr/Mentor/Questa_CDC_Formal_10.7a/linux_x86_64/mgls/bin:/Mentor/Questa_inFact_10.7a/infact/bin:/home/engr/Mentor/Questa_CDC_Formal_10.7a/linux_x86_64/bin:/home/engr/Mentor/Questa_CDC_Formal_10.7a/linux_x86_64/modeltech/bin:/home/engr/Mentor/Questa_CDC_Formal_10.7a/linux_x86_64/mgls/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin:/opt/riscv/bin:/home/engr/.local/bin:/home/engr/bin:/opt/riscv/bin

It seems the main problem now is it can't find uvm_macros.sv which is 2nd item in my PATH above.

QuestaSim-64 vlog 10.7-QCF Compiler 2018.03 Mar 14 2018
Start time: 10:17:10 on Oct 25,2019
vlog -64 "+incdir+/home/engr/riscv-dv/setting" "+incdir+/home/engr/riscv-dv/user
_extension" -access=rwc -f /home/engr/riscv-dv/files.f -sv -mfcu -cuname design_
cuname "+define+UVM_REGEX_NO_DPI" -writetoplevels /home/engr/riscv-dv/out_2019-1
0-25/top.list -l /home/engr/riscv-dv/out_2019-10-25/compile.log 
-- Compiling package riscv_signature_pkg
-- Compiling package riscv_instr_pkg
** Error: ./src/riscv_instr_pkg.sv(22): Cannot find `include file "uvm_macros.sv
h" in directories:
    /home/engr/riscv-dv/setting, /home/engr/riscv-dv/user_extension, ./src, ./te
st, /home/engr/Mentor/Questa_CDC_Formal_10.7a/share/modeltech/ovm-2.1.2/../veril
og_src/ovm-2.1.2/src
** Error: (vlib-35) Failed to create directory "/home/engr/Mentor/Questa_CDC_For
mal_10.7a/linux_x86_64/modeltech/questasim/uvm-1.2".
No such file or directory. (errno = ENOENT)

I tried changing UVM_HOME to /home/engr/uvm/uvm-1.2/src  but it just doesn't seem to be looking in the specified folder

amitgargmentor, 2019-10-25:
which location your QUESTA_HOME points to ?

weedmank, 2019-10-25:
Here's everything in my .bashrc:

# GNU Toolchain i.e. riscv-gnu-toolchain
export RISCV=/opt/riscv
export PATH=$PATH:$RISCV/bin

# riscv-dv
export RISCV_DV_ROOT=/home/engr/riscv-dv
export RISCV_TOOLCHAIN=/opt/riscv
export RISCV_GCC="$RISCV_TOOLCHAIN/bin/riscv64-unknown-elf-gcc"
export RISCV_OBJCOPY="$RISCV_TOOLCHAIN/bin/riscv64-unknown-elf-objcopy"
export SPIKE_PATH=$RISCV_TOOLCHAIN/bin
export UVM_HOME=/home/engr/uvm/uvm-1.2/src

export QHOME=/home/engr/Mentor/Questa_CDC_Formal_10.7a/linux_x86_64
export QUESTA_HOME=$QHOME/modeltech

export LMTOOL=$QHOME/mgls/bin

export PATH=$UVM_HOME:$QHOME/bin:$QUESTA_HOME/bin:$LMTOOL:$PATH

# other definitions 
export LM_LICENSE_FILE=/home/engr/Mentor/Mentor_license.txt
export LD_LIBRARY_PATH=$QHOME/lib:$LD_LIBRARY_PATH

export INFACT_HOME=/Mentor/Questa_inFact_10.7a/infact
export INFACT_PLATFORM=linux_x86_64
export INFACT_CXX=gcc_5.3.0

export PATH=$INFACT_HOME/bin:$PATH

#LD_LIBRARY_PATH — Specifies the path to the inFact directory containing
#dynamic shared libraries necessary for compiling/simulating the inFact test component.
export LD_LIBRARY_PATH=$INFACT_HOME/$INFACT_PLATFORM/$INFACT_CXX/lib:$LD_LIBRARY_PATH

#INFACT_LIBDIR — Specifies the location of the inFact library files and is typically
#used in custom scripts that compile and run simulations. For example:
export INFACT_LIBDIR=$INFACT_HOME/$INFACT_PLATFORM/$INFACT_CXX/lib

#INFACT_BINDIR — Specifies the location of the inFact executable files and is
#typically used in custom scripts that compile and run simulations. For example:
export INFACT_BINDIR=$INFACT_HOME/$INFACT_PLATFORM/$INFACT_CXX/bin

#INFACT_SDM_TIMEOUT — Specifies the time period in hours that a Simulation
#Distribution Manager (SDM) process runs after the last simulation client disconnects.
export INFACT_SDM_TIMEOUT=4

taoliug, 2019-10-25:
Can you find UVM 1.2 in this directory?
$QUESTA_HOME/questasim/uvm-1.2

weedmank, 2019-10-25:
No.  In the compile.log it said it was trying to CREATE this folder ????

weedmank, 2019-10-25:
What is UVM_HOME for if not to specify where the uvm info is located

taoliug, 2019-10-25:
I think this command is used to map UVM library
https://github.com/google/riscv-dv/blob/master/yaml/simulator.yaml#L50

I don't have Questasim environment though, maybe other user can help with this.

weedmank, 2019-10-25:
Even if I put the uvm-1.2 folder in $QUESTA_HOME/questasim/uvm-1.2, it still has the same problem

weedmank, 2019-10-25:
Why is it looking in a $QUESTA_HOME ovm folder?:

** Error: /home/engr/riscv-dv/src/riscv_instr_pkg.sv(22): Cannot find `include file "uvm_macros.svh" in directories:
    /home/engr/riscv-dv/setting, /home/engr/riscv-dv/user_extension, /home/engr/riscv-dv/src, /home/engr/riscv-dv/test, /home/engr/Mentor/Ques
ta_CDC_Formal_10.7a/share/modeltech/ovm-2.1.2/../verilog_src/ovm-2.1.2/src

danghai, 2019-10-25:
Remove `export UVM_HOME=/home/engr/uvm/uvm-1.2/src`? 
""vmap mtiUvm $QUESTA_HOME/questasim/uvm-1.2" will map uvm for you

amitgargmentor, 2019-10-25:
I am also using QUESTA_SIM, and for me,  UVM-1.2 is available at QUESTA_HOME directory (parallel to bin directory)

$ echo $QUESTA_HOME
/u/release/2019.3/questasim

$ echo $QUESTA_HOME/uvm-1.2/
/u/release/2019.3/questasim/uvm-1.2/

I changed the command vmap commad in yaml/simulator.yaml to:
- "vmap mtiUvm $QUESTA_HOME/uvm-1.2"

Thanks,
Amit

danghai, 2019-10-25:
I think he uses Questa inside CDC-Formal tool

weedmank, 2019-10-25:
Tried suggestions and variations but still get this:
QuestaSim-64 vlog 10.7-QCF Compiler 2018.03 Mar 14 2018
Start time: 15:27:02 on Oct 25,2019
vlog -64 "+incdir+/home/engr/riscv-dv/setting" "+incdir+/home/engr/riscv-dv/user_extension" -access=rwc -f /home/engr/riscv-dv/files.f -sv -mf
cu -cuname design_cuname "+define+UVM_REGEX_NO_DPI" -writetoplevels /home/engr/riscv-dv/out_2019-10-25/top.list -l /home/engr/riscv-dv/out_201
9-10-25/compile.log 
-- Compiling package riscv_signature_pkg
-- Compiling package riscv_instr_pkg
** Error: /home/engr/riscv-dv/src/riscv_instr_pkg.sv(22): Cannot find `include file "uvm_macros.svh" in directories:
    /home/engr/riscv-dv/setting, /home/engr/riscv-dv/user_extension, /home/engr/riscv-dv/src, /home/engr/riscv-dv/test, /home/engr/Mentor/Ques

It just wont look in the folder where I have the uvm-1.2.  It must be expecting an absolute location...
In the above I no longer have UVM_HOME defined and simulator.yaml has this:
"vmap mtiUvm /home/engr/uvm/uvm-1.2/src"

Also tried "vmap mtiUvm /home/engr/uvm/uvm-1.2"

taoliug, 2019-10-25:
Does UVM work for your other TB?
If you know the command that works for you, you can modify the command setting in this file
https://github.com/google/riscv-dv/blob/master/yaml/simulator.yaml#L47

weedmank, 2019-10-25:
I can get Questasim to run a SV simulation and use Questa Verify (CDC stuff) but haven't tried UVM.

taoliug, 2019-10-31:
Closing this issue for now as it seems to be a general Questa setup issue, not specific to riscv-dv. Feel free to reopen it if you think there's an issue with riscv-dv Questa flow.

closed 2019-10-31 by taoliug