chipsalliance/riscv-dv #266: Duplicates

eroom1966, 2019-10-25:
@taoliug 
Did you report you had this working under dsim, I see some valid issues

I am seeing a number of Duplicate warning messages by dsim, and they seem correct
first one is here 
riscv-dv/src/riscv_instr_cov_item.sv:184:7

>       BGEU   : is_branch_hit = (rs1_value > rs2_value);
>       BGEU   : is_branch_hit = (rs1_value > rs2_value);

Other duplicates appear at these lines

riscv-dv/src/riscv_instr_base.sv:968:7
riscv-dv/src/riscv_instr_base.sv:945:7

riscv-dv/src/riscv_instr_base.sv:969:7
riscv-dv/src/riscv_instr_base.sv:946:7                  

Also I am seeing the following
Incompatible types for second/third operand of the following ternary operators:    
riscv-dv/src/riscv_directed_instr_lib.sv:210:13         <unknown>$[$]    <unknown>$[$]    

Do you not reproduce these issues ?

Thx
Lee

taoliug, 2019-10-25:
Thanks for reporting, I will fix the duplicate
I can reproduce the dsim compile issue as you pointed out. However, I think this should have been fixed now.
https://github.com/google/riscv-dv/blob/master/src/riscv_directed_instr_lib.sv#L210

taoliug, 2019-10-25:
This is now fixed (changes to lib.py are also merged)

dsim compilation is clean for me now:
Elaborating...
  Top-level modules:
    $unit
    riscv_instr_gen_tb_top
  Found 3 unique specialization(s) of 3 design element(s).
Optimizing...
Building models...
Simulation time precision is 1ns.
Linking image.so...
  [1/9] module $root: 1682 functions, 3732 basic blocks
  [2/9] module $pkgs: 1 functions, 2 basic blocks
  [3/9] package std: 31 functions, 80 basic blocks
  [4/9] package uvm_pkg: 4 submodules, 3868 functions, 27868 basic blocks
  [5/9] package riscv_signature_pkg: 22 functions, 88 basic blocks
  [6/9] package riscv_instr_pkg: 3 submodules, 1569 functions, 16283 basic blocks
  [7/9] package riscv_instr_test_pkg: 41 functions, 388 basic blocks
  [8/9] CU scope $unit: 1 functions, 2 basic blocks
  [9/9] module riscv_instr_gen_tb_top: 2 functions, 6 basic blocks

closed 2019-10-25 by taoliug