chipsalliance/riscv-dv #283: Cross-comparing outputs of spike and riscv-ovpsim fails

abdullahyildiz, 2019-11-05:
I work on commit id 44bec7695fd2399166e181fa84b66a608b5f745f.

When I run the following command, outputs of Spike and OVPSim differ.

`python3.5 run.py --target rv64imc --test=riscv_ebreak_test --iss=ovpsim,spike -v`

This is the output of iss_regr.log:

```
Test binary: out_2019-11-05/asm_tests/riscv_ebreak_test_0.o
ovpsim : out_2019-11-05/ovpsim_sim/riscv_ebreak_test.0.csv
spike : out_2019-11-05/spike_sim/riscv_ebreak_test.0.csv
24495 instructions left in trace spike
Compare result[FAILED]: 93 matched, 24495 mismatch

Test binary: out_2019-11-05/asm_tests/riscv_ebreak_test_1.o
ovpsim : out_2019-11-05/ovpsim_sim/riscv_ebreak_test.1.csv
spike : out_2019-11-05/spike_sim/riscv_ebreak_test.1.csv
30000 instructions left in trace spike
Compare result[FAILED]: 254 matched, 30000 mismatch

0 PASSED, 2 FAILED
```

Spike build commit id: 5652b9d.
OVPSim commit id: 1f6f5b7.

simon5656, 2019-11-05:
Please can you put the .S files somewhere and we will have a look
thanks
Simon

taoliug, 2019-11-05:
I just run this test 10 times, all passed. Did you run with any local changes?
....
Tue, 05 Nov 2019 09:13:35 INFO     CSV saved to : out_2019-11-05/spike_sim/riscv_ebreak_test.9.csv
Tue, 05 Nov 2019 09:13:35 INFO     Processing ovpsim log [0]: out_2019-11-05/ovpsim_sim/riscv_ebreak_test.9.log
Tue, 05 Nov 2019 09:13:37 INFO     Processed instruction count : 38821
Tue, 05 Nov 2019 09:13:37 INFO     CSV saved to : out_2019-11-05/ovpsim_sim/riscv_ebreak_test.9.csv
Tue, 05 Nov 2019 09:13:37 INFO     10 PASSED, 0 FAILED
Tue, 05 Nov 2019 09:13:37 INFO     ISS regression report is saved to out_2019-11-05/iss_regr.log

abdullahyildiz, 2019-11-06:
> I just run this test 10 times, all passed. Did you run with any local changes?

No, I use a clean copy of the repository.

abdullahyildiz, 2019-11-06:
@simon5656 ,

The files you asked for are attached.

[run.log](https://github.com/google/riscv-dv/files/3812637/run.log)
[seed.yaml.txt](https://github.com/google/riscv-dv/files/3812642/seed.yaml.txt)
[riscv_ebreak_test_0.S.txt](https://github.com/google/riscv-dv/files/3812644/riscv_ebreak_test_0.S.txt)
[riscv_ebreak_test_1.S.txt](https://github.com/google/riscv-dv/files/3812643/riscv_ebreak_test_1.S.txt)

taoliug, 2019-11-06:
Can you share the spike/ovpsim iss sim log, and the iss_regr.log?

abdullahyildiz, 2019-11-06:
@taoliug,

The files you asked for are attached.

[riscv_ebreak_test.1.spike.log](https://github.com/google/riscv-dv/files/3812784/riscv_ebreak_test.1.spike.log)
[riscv_ebreak_test.1.ovpsim.log](https://github.com/google/riscv-dv/files/3812785/riscv_ebreak_test.1.ovpsim.log)
[riscv_ebreak_test.0.spike.log](https://github.com/google/riscv-dv/files/3812786/riscv_ebreak_test.0.spike.log)
[riscv_ebreak_test.0.ovpsim.log](https://github.com/google/riscv-dv/files/3812787/riscv_ebreak_test.0.ovpsim.log)
[iss_regr.log](https://github.com/google/riscv-dv/files/3812788/iss_regr.log)

taoliug, 2019-11-06:
Seems like ovpsim stops execution upon "ebreak"

Info 134: 'riscvOVPsim/cpu', 0x00000000800001ba(main_4_0_t+32): Machine 9002     ebreak
Fatal (RISCV/PK) SYSCALL(unhandled(nr=-2147483648))
Info Exiting

abdullahyildiz, 2019-11-06:
Which commit id of OVPSim do you use?
I used the latest commit (commit id: 56ca927) and cross-comparing still fails.

taoliug, 2019-11-06:
Will leave it to @simon5656  to comment on ovpsim behavior.

abdullahyildiz, 2019-11-06:
By the way, there is no such problem for another test:

`python3.5 run.py --target rv64imc --test=riscv_arithmetic_basic_test --iss=ovpsim,spike -v`

[riscv_arithmetic_basic_test_0.S.txt](https://github.com/google/riscv-dv/files/3812976/riscv_arithmetic_basic_test_0.S.txt)
[riscv_arithmetic_basic_test_1.S.txt](https://github.com/google/riscv-dv/files/3812977/riscv_arithmetic_basic_test_1.S.txt)
[iss_regr.log](https://github.com/google/riscv-dv/files/3812978/iss_regr.log)
[riscv_arithmetic_basic_test.0.log.ovpsim.log](https://github.com/google/riscv-dv/files/3812980/riscv_arithmetic_basic_test.0.log.ovpsim.log)
[riscv_arithmetic_basic_test.1.log.ovpsim.log](https://github.com/google/riscv-dv/files/3812981/riscv_arithmetic_basic_test.1.log.ovpsim.log)
[riscv_arithmetic_basic_test.0.log.spike.log](https://github.com/google/riscv-dv/files/3812984/riscv_arithmetic_basic_test.0.log.spike.log)
[Uploading riscv_arithmetic_basic_test.1.log.spike.log…]()

eroom1966, 2019-11-06:
@taoliug 

Hi Tao,
I think I need to understand your expected exit mechanism ?
As far as I can see, you expect your generated assembler to exit when an 'ecall' instruction is executed.
In riscvOVPsim, we are utilizing our semihost library, whereby when an ecall instruction is executed the A7 register is inspected to see what the intended SYSCALL was being invoked.

From your generated assembler, it appears you do not generate a specific value in register A7, which usually results in the simulation terminating due to an unknown System Call. Firstly I would recommend either using the syscall(kill), or syscall(exit) codes , which are as follows
syscall(kill) : A7=129
syscall(exit) : A7=93

This would cause the simulator to terminate.
The question is then what do we do about ebreak, which effectively is the same as ecall, but returning to a debugger, rather than an executing environment.
If the intention is that the ebreak should trap, and be handled by the traphandler, then should the same not be true of the ecall, which is it should trap - and be handled by a traphandler

It sounds like it is a mistake for riscvOVPsim to be using the semihoting routines for the ecall/ebreak instructions - we can easily disable this, but equally you need to define the exit sequence so that code does not continue executing/trapping

Thx
Lee

taoliug, 2019-11-06:
I expect ebreak to cause a trap and handled by the trap handler. It works for me with older ovpsim version, is this something updated recently?

eroom1966, 2019-11-06:
@taoliug 
in riscvOVPsim, ebreak and ecall can both be configured to either
1. Perform a trap
2. Perform a semihosted operation

> I expect ebreak to cause a trap and handled by the trap handler. 

OK, I understand you expect ebreak to trap, as per (1) above
What would you expect an ecall instruction  to do, (1) or (2), or something else ?

Thx
Lee

taoliug, 2019-11-06:
(1)

abdullahyildiz, 2019-11-08:
> I expect ebreak to cause a trap and handled by the trap handler. It works for me with older ovpsim version, is this something updated recently?

@taoliug,

Which commit id of OVPSim does not cause a problem for you?

eroom1966, 2019-11-08:
@taoliug 

Hi Tao
I am still unclear on how your generated stream is intended to terminate the program execution on whatever the target is (iss sim, rtl sim, ...) ?

If my understanding is correct, executing a trap handler for ecall would cause execution on any target to run infinitely.
So how would you expect a verilog simulator or iss to stop executing the generated ISG code, you must be expecting some bailout condition ?

Thx
Lee

taoliug, 2019-11-08:
The trap handler generated by ISG will handle the trap properly. For ovpsim, eventually it will execute the code with _exit label which will terminate ISS. In RTL simulation, we will terminate the test if ecall is detected by by the processor.

taoliug, 2019-11-12:
@eroom1966 , any comment on this? I actually don't know why the same test pass on my side, maybe because of some recent ovpsim changes?

abdullahyildiz, 2019-11-12:
@taoliug,

Which commit id of OVPSim does not cause a problem for you?

taoliug, 2019-11-12:
commit db82ac94ae9d9d2e1182c947424b78e340fd4dae (HEAD -> master, tag: v20190708.0, origin/master, origin/HEAD)

abdullahyildiz, 2019-11-12:
@taoliug,

When I use the commit id you use (commit db82ac9), cross-comparison of log files for riscv_ebreak_test passes.

taoliug, 2019-11-12:
cool, @eroom1966 , can you comment what we need to do make upstream ovpsim work as intended?

eroom1966, 2019-11-12:
@taoliug @abdullahyildiz 
The problem you are seeing is due to the semihosting behavior of ecall and ebreak instructions
We need to disable semihosting for the usage of riscvOVPsim as part of the riscv-dv flow, I will speak with @simond-imperas  in order to facilitate this

Additionally we need to implement the semantic behavior in order to stop execution
@taoliug If I understand correctly from your earlier message, you are saying that execution should stop when executing at an address with the label '_exit'.
Again I need to check with Simon but I think we had scenarios where this was not happening

I will comment further, later today

taoliug, 2019-12-12:
@eroom1966, please take a look at this issue. This could prevent us from using upstream ovpsim in co-simulation. I haven't tried the latest ovpsim myself, not sure if it's solved or not.

abdullahyildiz, 2019-12-13:
@taoliug,

I tried the latest riscvOVPsim and observed that this time the following command fails while running riscvOVPsim.exe:

`python3 run.py --target rv64imc --test=riscv_arithmetic_basic_test --iss=ovpsim,spike -v`

These are the last three lines of riscv_arithmetic_basic_test.0.log file:

```
Info 10097: 'riscvOVPsim/cpu', 0x0000000080008662(test_done+2): Machine 00000073 ecall
Fatal (RISCV/PK) SYSCALL(unhandled(nr=-2147361544))
Info Exiting
```
I have attached the related files.

[riscv_arithmetic_basic_test.0.log](https://github.com/google/riscv-dv/files/3959396/riscv_arithmetic_basic_test.0.log)
[riscv_arithmetic_basic_test_0.S.txt](https://github.com/google/riscv-dv/files/3959399/riscv_arithmetic_basic_test_0.S.txt)

paulzcgu, 2019-12-19:
@taoliug 

The cross-comparing outputs of spike and ovpsim are constantly failed at the following two test cases:

* riscv_unaligned_load_store_test
* riscv_full_interrupt_test

ovpsim build from commit: db82ac94ae9d9d2e1182c947424b78e340fd4dae 
spike build from commit: master
vcs version: 2019.06-SP2
python: 3.6

```
python3 run.py --test=riscv_full_interrupt_test --iss=spike,ovpsim
python3 run.py --test=riscv_unaligned_load_store_test --iss=spike,ovpsim
```

The issue was initially found from the ibex project, where the comparison between the outputs of spike and rtl_sim was always mismatched, but the comparison between the outputs of ovpsim and rtl_sim with exact same .o .S and .bin files was always matched. 

* riscv_unaligned_load_store_test
[riscv_unaligned_load_store_test_0.S.txt](https://github.com/google/riscv-dv/files/3981189/riscv_unaligned_load_store_test_0.S.txt)
[riscv_unaligned_load_store_test.0_ovpsim.log](https://github.com/google/riscv-dv/files/3981220/riscv_unaligned_load_store_test.0_ovpsim.log)
[riscv_unaligned_load_store_test.0_spike.log](https://github.com/google/riscv-dv/files/3981221/riscv_unaligned_load_store_test.0_spike.log)

* riscv_full_interrupt_test
[riscv_full_interrupt_test_0.S.txt](https://github.com/google/riscv-dv/files/3981190/riscv_full_interrupt_test_0.S.txt)
[riscv_full_interrupt_test.0_ovpsim.log](https://github.com/google/riscv-dv/files/3981240/riscv_full_interrupt_test.0_ovpsim.log)
[riscv_full_interrupt_test.0.spike.log](https://github.com/google/riscv-dv/files/3981241/riscv_full_interrupt_test.0.spike.log)

paulzcgu, 2019-12-19:
Tested with the spike version mentioned in previous comment, the issue persists.
spike: 5652b9dd6298c7b61fe80bc2a508bbd3c9e5c5ef

taoliug, 2019-12-20:
riscv_full_interrupt_test is fixed, it's related mie compare mismatch. 
riscv_unaligned_load_store_test mismatch is mostly caused by the setup mismatch between two ISS, ovpsim is configured to support unaligned load/store, but spike by default will raise exception for it. We have locally build spike with unaligned load/store support.

taoliug, 2019-12-20:
You can set unaligned load/store to F to see if it solves the problem.
https://github.com/google/riscv-dv/blob/master/target/rv32imc/riscvOVPsim.ic#L7

taoliug, 2019-12-20:
Made a temp fix to ovpsim, most tests should pass except for the ebreak test.
However, still need to have a way to run ovpsim in baremetal mode rather than proxy kernel mode.

paulzcgu, 2019-12-20:
> riscv_full_interrupt_test is fixed, it's related mie compare mismatch.
> riscv_unaligned_load_store_test mismatch is mostly caused by the setup mismatch between two ISS, ovpsim is configured to [...]

> You can set unaligned load/store to F to see if it solves the problem.
> https://github.com/google/riscv-dv/blob/master/target/rv32imc/riscvOVPsim.ic#L7

riscv_full_interrupt_test fix tested, no longer mismatch.
riscv_unaligned_load_store_test no longer mismatch after set unaligned load/store to F in OVPSIM setup.

Thanks a lot.

taoliug, 2020-01-02:
I just tied the latest version of ovpsim, and it seems all issues are gone. Closing this issue now.

closed 2020-01-02 by taoliug

abdullahyildiz, 2020-01-08:
Have you tried the riscv_ebreak_test? It still fails on my machine.

`python3 run.py --target rv64imc --test=riscv_ebreak_test --iss=ovpsim,spike -v`

The commit id I work on is 92023bb.
Spike build commit id: c3b28ab.
riscvOVPsim commit id: ea0e1f5.

I've attached the output of the command above.
[run.log](https://github.com/google/riscv-dv/files/4034855/run.log)

abdullahyildiz, 2020-01-12:
@taoliug,

Any update on this?