chipsalliance/riscv-dv #284: Pull request 282

eroom1966, 2019-11-05:
https://github.com/google/riscv-dv/pull/282

@taoliug @udinator 

This pull request has reversed the update we put in here
https://github.com/google/riscv-dv/pull/274

taoliug, 2019-11-05:
The change is to make it works for both predefined target and custom target. I don't think anything is broken here. The <cfg_path> is to allow user specifies a custom target directory which have the core_setting, testlist and ovpsim configure file.

Do you see any problem with this PR?

udinator, 2019-11-05:
Hi @eroom1966 - I changed this code, as it was not possible to use custom OVPsim configuration files to run simulations. With the current setup after #282, if you do not pass in an argument for `--custom_target`, and pass some `--target` into the script, the flow will pick up the OVPsim configuration file located at `/target/<variant>` by default.

taoliug, 2019-11-06:
Let me know if anything needs to be done for this issue. I believe the issue can be solved by either using --target or --custom_target option.

closed 2019-11-08 by taoliug