chipsalliance/riscv-dv #287: Inconsistency between target riscvOVPsim.ic and riscv_core_setting.sv

eroom1966, 2019-11-07:
@udinator @taoliug 

Having reviewed a few of the recent commits (which we have commented), I think there is an inconsistency between what you are defining in terms of supported privilege modes, in the following directory (and I am sure others)
riscv-dv/tree/master/target/rv32i
You have defined that the DUT supports MACHINE mode only with this statement

> // Supported Privileged mode
> privileged_mode_t supported_privileged_mode[] = {MACHINE_MODE};

This maps to RVB32I as the --variant option in riscvOVPsim.ic, but the actual variant you have assigned is RV32I - this includes USER and SUPERVISOR modes

So these 2 configurations are now in conflict with each other

Is the intention to have 
MACHINE + USER + SUPERVISOR, or
MACHINE 

Thx
Lee

taoliug, 2019-11-12:
I tried to change it to RVB32I, but got error like below. Do I need a newer ovpsim version?

Error (OP_IU) Setting enumerated parameter 'riscvOVPsim/cpu/variant' to the illegal value 'RVB32I'. Legal values are:
    RV32I
    RV32IM
    RV32IMC
    RV32IMAC
    RV32G
    RV32GC
    RV32GCN
    RV32GCV
    RV32E
    RV32EC
    RV64I
    RV64IM
    RV64IMC
    RV64IMAC
    RV64G
    RV64GC
    RV64GCN
    RV64GCV

simond-imperas, 2019-11-12:
RV32I includes Machine, Supervisor and User
RVB32I, and RVB32E and RV64I were committed to riscv-ovpsim  earlier this year - from June 2019

so yes, please use a later release.
Simon

taoliug, 2019-11-17:
This issue is currently blocked by #283

taoliug, 2020-01-02:
I tried the latest ovpsim version, but still get the same error as the error. RVB32I is not accepted.

eroom1966, 2020-01-11:
Hi Tao
Can you post the command line you are using and the full output log from the simulator
Thx
Lee

closed 2020-02-12 by taoliug