chipsalliance/riscv-dv #294: conditional coverage

simond-imperas, 2019-11-10:
To run the compliance suite through coverage and get approproprate results I need to conditionally enable the coverage groups. Currently if I set supported_isa (riscv_core_etting.sv) then it enables the coverage for that group with the default RV32I cgs and rv32i_misc_cg and opcode_cg.

when running one compliance test suite I only want to record, see, and report the cg for that suite, eg RV32M - so we can see how those specific instructions are covered by those specific tests.

so in riscv_instr_cover_group I need to be able to turn off RV32I cgs  even though supported_isa  includes it, and also turn off the others extra ones.

it would be simple conditionals, like:
>  if (RV32M inside {supported_isa}) begin

but what would it depend on and be set by?

I thought of maybe a simulator compile time +define to set a variable in a class somewhere, or maybe in the cov_testlist.yaml like 'no_iss' again setting a variable in an accessible class.

what would you suggest and can you point me at where to hold the setting.

thanks
Simon

simond-imperas, 2019-11-10:
Maybe it could be like in riscv_core_setting.sv with enum etc:
> riscv_instr_group_t supported_isa[$] = {RV32I};

eg:
> riscv_coverage_options_t supported_coverage[$] = {NO_RV32I, NO_MISC, RV32V, RV64V};

and then it can be just used in the other .sv files

with defaults being RV32I, MISC, and no RV32V, no RV64V

so we can conditionally bring in different coverage groups to get the coverage data to reflect the compliance testing

and also as it is in the settings file - it is by targets
and with good defaults the only edit is needed in the settings files to add [so this is change needed by all], eg:
riscv_coverage_options_t coverage_options[$] = {}

shall I add and submit a PR so you can have a look?

pls let me know.
thanks

simond-imperas, 2019-11-11:
I have made the changes and they can be seen on my branch 

> https://github.com/simond-imperas/riscv-dv/tree/vectors1

pls have a look and comment and if ok I will submit as a PR
thanks

taoliug, 2019-11-12:
I think it's better to have a runtime option to override the setting in riscv_core_setting.sv. For example, in riscv_core_setting.sv, you can enable RV32IMC. However, if you want to run with only RV32M covergroup, you can run like "python3 cov.py --isa rv32m", I can put together a PR to add this support.

taoliug, 2019-11-12:
Actually I think it might be more complicated if you want to just filter out a subset of the covergroup. There are many many cases like branch hit/miss combinations, illegal instructions, interrupt/trap which do not have 1:1 mapping to a specific ISA.

simond-imperas, 2019-11-14:
Tao - I have worked out how to pass options in from command line of cov.py which will work for both the coverage and vectors - thus no change to settings.sv - which will allow me to configure the coverage the way I need it - and to develop the vectors without churning pull requests - so I will make the changes to my branch and ask you to review before I submit a pull request - Simon

taoliug, 2019-12-12:
I think this issue can be closed as the relevant code has been checked in.

closed 2019-12-12 by taoliug