chipsalliance/riscv-dv #295: OVPsim log parsing errors

udinator, 2019-11-11:
Hi @simond-imperas - I'm running the functional coverage flow on the OVPsim trace logs, and I'm getting these errors:
`UVM_ERROR /usr/local/google/home/udij/opentitan/ibex/vendor/google_riscv-dv/test/riscv_instr_cov_test.sv(156) @ 0: uvm_test_top [uvm_test_top] Unrecoganized rd: [mie] (csrrw,mie,00000a0a,a5,bf85df4f,,,,"csrw    mie,a5",000000008000eeec,30479073,,3,,,,,,,,,,,,,,`.
It appears that for these `csrrw` instructions, the `RD` register is not getting written to the CSV files.
Best,
Udi

taoliug, 2019-11-12:
BTW, I fixed one issue related to jalr pseudo instruction handling
https://github.com/google/riscv-dv/pulls?q=is%3Apr+is%3Aclosed

simond-imperas, 2019-11-14:
Hi @udinator can you post a snippet of the .S that exhibits this.
thanks

udinator, 2019-11-14:
@simond-imperas @taoliug  - the original issue relating to `csrrw` instructions is no longer showing up, instead the error is now with `c.jalr` instruction: `Unrecoganized rs2: [] (c.jalr,ra,8000adb0,a4,8000b34c,,,0,jalr    a4,000000008000adae,9702,,3,,,,,,,,,,,,,[],"[['ra', '8000adb0']]`

Here is a snippet of the assembly code:
[image]
In particular, it is the compressed jalr instruction at `8000b75e`.

PR #309 (2019-11-17): Fix ovpsim log parsing issue
Fix #295
closed 2019-11-17 by taoliug