chipsalliance/riscv-dv #300: Define a new format for instruction trace

taoliug, 2019-11-12:
As we are adding more architectural states to the instruction trace (like floating point registers, vector registers, privileged CSRs), the current CSV format doesn't scale well as most of the columns are empty. We need to define a new format for instruction trace. Possible candidates: YAML, Hjson, Protobuf.

taoliug, 2019-12-09:
Thought about this a bit more. The current coverage flow is to use a python script to generate a standard trace log (CSV format now), and use SV to read the trace log to collect functional coverage. SV doesn't provide good support for parsing structured log (YAML, protobuf etc). We may still need to stick with CSV format, but make it more compact.

The current CSV format lists all possible architectural state in the header:
instr,rd,rd_val,rs1,rs1_val,rs2,rs2_val,rs3,rs3_val,imm,str,addr,binary,csr,mode,vd,vd_val,vs1,vs1_val,vs2,vs2_val,vs3,vs3_val,vtype_vsew,vtype_vmul,vtype_vediv,vm,updated_csr,updated_gpr,fd,fd_val,fs1,fs1_val,fs2,fs2_val

It should be changed to a more generic format
instr, result, operand, str,addr,binary,csr,mode

result could be any architectural state, like general purpose integer registers, floating point registers, vector registers, privileged register.  

Original format:
lbu,t4,000000000000002b,s11,0000000080018f05,,,,,23,"lbu     t4, 35(s11)",000000008000046e,023dce83,,3,,,,,,,,,,,,,,,,,,,,

New format:
lbu,t4:000000000000002b,s11:0000000080018f05;imm:23,"lbu     t4, 35(s11)",000000008000046e,023dce83,3

taoliug, 2019-12-09:
@danghai Please let me know if you have any thought or question on this.

danghai, 2019-12-10:
Hi @taoliug . It seems you want to group them together? Ex: `rd,rd_val` becomes `rd:rd_val`.
`rs1,rs1_val` become `rs1:rs1_val`, ...

taoliug, 2019-12-10:
yes, basically we will have two fields, one for operands, one for results. There can be multiple entries separated by ";" in each field. Each entry is a dictionary type <name>:<value>. The name can be any integer, floating point, vector, privileged register.

taoliug, 2019-12-25:
@danghai I am starting to implement this now. It's a bit complicated as it requires the understanding of the end-to-end flow and RISC-V spec. You can take a look at the PRs I send out later and see if anything needs to be enhanced.

danghai, 2019-12-25:
It sounds good. I am reading the RISC-V spec. I think I understand more after your commits.

taoliug, 2019-12-25:
Submitted the 1st PR, basic RV32imc regression passed
python3 run.py --iss spike,ovpsim --exp

taoliug, 2019-12-29:
This feature is mostly done, the remaining work is adding some document to explain the new trace CSV format.

taoliug, 2020-01-21:
Document is updated:
https://github.com/google/riscv-dv/blob/master/docs/source/appendix.rst

closed 2020-01-21 by taoliug