chipsalliance/riscv-dv #314: Issues generating tests without RV32 support

dpetrisko, 2019-11-18:
I have an RV64I core with MSU privileged mode support that I'm trying to verify using riscv-dv. I copied the testlist.yaml and riscv_core_setting.sv from the example RV64GC target and modified it to fit my core.  However, when I try to generate tests involving privileged mode support, such as riscv_machine_mode_rand_test, the generator fails because of inconsistent constraints

Error:
```
Error-[CNST-CIF] Constraints inconsistency failure
/mnt/users/ssd0/no_backup/petrisko/pre-alpha-release/bp_common/test/src/riscv-dv/src/riscv_directed_instr_lib.sv, 323
  Constraints are inconsistent and cannot be solved.
  Please check the inconsistent constraints being printed above and rewrite
  them.
```

I've noticed that the test will successfully generate if I claim to have RV32I support in addition to RV64I in riscv_core_setting.sv i.e.
```
// This works
riscv_instr_group_t supported_isa[$] = {RV64I, RV32I};

// This doesn't
riscv_instr_group_t supported_isa[$] = {RV64I};
```

I'm very unfamiliar with UVM, so debugging this failure is difficult for me.  Could you point me in the right direction?  Below are my full testlist.yaml and riscv_core_setting.sv if it helps.

testlist.yaml:
```
- test: riscv_machine_mode_rand_test
  description: >
    Machine mode random instruction test
  iterations: 2
  gen_test: riscv_instr_base_test
  gen_opts: >
    +instr_cnt=10000
    +num_of_sub_program=5
    +boot_mode=m
  rtl_test: core_base_test
```
riscv_core_setting.sv:
```
//-----------------------------------------------------------------------------
// Processor feature configuration
//-----------------------------------------------------------------------------
// XLEN
parameter int XLEN = 64;

// Parameter for SATP mode, set to BARE if address translation is not supported
parameter satp_mode_t SATP_MODE = SV39;

// Supported Privileged mode
privileged_mode_t supported_privileged_mode[] = {USER_MODE, SUPERVISOR_MODE, MACHINE_MODE};

// Unsupported instructions
riscv_instr_name_t unsupported_instr[];

// ISA supported by the processor
//riscv_instr_group_t supported_isa[$] = {RV64I};
riscv_instr_group_t supported_isa[$] = {RV64I, RV64M, RV64C, RV64A,
                                        RV64F, RV64D};
//riscv_instr_group_t supported_isa[$] = {RV32I, RV32M, RV64I, RV64M, RV32C, RV64C, RV32A, RV64A,
//                                        RV32F, RV64F, RV32D, RV64D};

// Interrupt mode support
mtvec_mode_t supported_interrupt_mode[$] = {DIRECT};

// The number of interrupt vectors to be generated, only used if VECTORED interrupt mode is
// supported
int max_interrupt_vector_num = 16;

// Debug mode support
bit support_debug_mode = 0;

// Support delegate trap to user mode
bit support_umode_trap = 0;

// Support sfence.vma instruction
bit support_sfence = 1;

// ----------------------------------------------------------------------------
// Previleged CSR implementation
// ----------------------------------------------------------------------------

// Implemented previlieged CSR list
`ifdef DSIM
privileged_reg_t implemented_csr[] = {
`else
parameter privileged_reg_t implemented_csr[] = {
`endif
    // User mode CSR
    USTATUS,    // User status
    UIE,        // User interrupt-enable register
    UTVEC,      // User trap-handler base address
    USCRATCH,   // Scratch register for user trap handlers
    UEPC,       // User exception program counter
    UCAUSE,     // User trap cause
    UTVAL,      // User bad address or instruction
    // Supervisor mode CSR
    SSTATUS,    // Supervisor status
    SIE,        // Supervisor interrupt-enable register
    STVEC,      // Supervisor trap-handler base address
    SCOUNTEREN, // Supervisor counter enable
    SSCRATCH,   // Scratch register for supervisor trap handlers
    SEPC,       // Supervisor exception program counter
    SCAUSE,     // Supervisor trap cause
    STVAL,      // Supervisor bad address or instruction
    SIP,        // Supervisor interrupt pending
    SATP,       // Supervisor address translation and protection
    // Machine mode mode CSR
    MVENDORID,  // Vendor ID
    MARCHID,    // Architecture ID
    MIMPID,     // Implementation ID
    MHARTID,    // Hardware thread ID
    MSTATUS,    // Machine status
    MISA,       // ISA and extensions
    MEDELEG,    // Machine exception delegation register
    MIDELEG,    // Machine interrupt delegation register
    MIE,        // Machine interrupt-enable register
    MTVEC,      // Machine trap-handler base address
    MCOUNTEREN, // Machine counter enable
    MSCRATCH,   // Scratch register for machine trap handlers
    MEPC,       // Machine exception program counter
    MCAUSE,     // Machine trap cause
    MTVAL,      // Machine bad address or instruction
    MIP         // Machine interrupt pending
};
// ----------------------------------------------------------------------------
// Supported interrupt/exception setting, used for functional coverage
// ----------------------------------------------------------------------------

`ifdef DSIM
interrupt_cause_t implemented_interrupt[] = {
`else
parameter interrupt_cause_t implemented_interrupt[] = {
`endif
    U_SOFTWARE_INTR,
    S_SOFTWARE_INTR,
    M_SOFTWARE_INTR,
    U_TIMER_INTR,
    S_TIMER_INTR,
    M_TIMER_INTR,
    U_EXTERNAL_INTR,
    S_EXTERNAL_INTR,
    M_EXTERNAL_INTR
};

`ifdef DSIM
exception_cause_t implemented_exception[] = {
`else
parameter exception_cause_t implemented_exception[] = {
`endif
    INSTRUCTION_ADDRESS_MISALIGNED,
    INSTRUCTION_ACCESS_FAULT,
    ILLEGAL_INSTRUCTION,
    BREAKPOINT,
    LOAD_ADDRESS_MISALIGNED,
    LOAD_ACCESS_FAULT,
    STORE_AMO_ADDRESS_MISALIGNED,
    STORE_AMO_ACCESS_FAULT,
    ECALL_UMODE,
    ECALL_SMODE,
    ECALL_MMODE,
    INSTRUCTION_PAGE_FAULT,
    LOAD_PAGE_FAULT,
    STORE_AMO_PAGE_FAULT
};
```
Compile command:
```
python3 run.py --steps gen,gcc_compile --simulator vcs --iterations 1 -o out/ \
        -ct user_extension -cs user_extension --isa rv64i --mabi lp64
```

taoliug, 2019-11-18:
This seems correct to me.
`riscv_instr_group_t supported_isa[$] = {RV64I, RV32I};
`
Do you want to support other ISA extensions?

dpetrisko, 2019-11-18:
RV64I and RV32I are two discrete ISAs.  For example, in RV64I, ADD is a 64-bit add and ADDW is a 32-bit add.  In RV32I, ADD is a 32-bit add and ADDW doesn't exist.  In order to run RV32I instructions on an RV64I core, one must change MXLEN in the misa CSR.

From the unprivileged spec (https://riscv.org/specifications/):
> The four base ISAs in RISC-V are treated as distinct base ISAs. A common question is why
> is there not a single ISA, and in particular, why is RV32I not a strict subset of RV64I? Some
> earlier ISA designs (SPARC, MIPS) adopted a strict superset policy when increasing address
> space size to support running existing 32-bit binaries on new 64-bit hardware.
> The main advantage of explicitly separating base ISAs is that each base ISA can be optimized for its needs without requiring to support all the operations needed for other base ISAs.
> For example, RV64I can omit instructions and CSRs that are only needed to cope with the narrower registers in RV32I. The RV32I variants can use encoding space otherwise reserved for
> instructions only required by wider address-space variants.
> The main disadvantage of not treating the design as a

closed 2019-11-25 by dpetrisko