chipsalliance/riscv-dv #318: adding constraint of group selecting in random instruction

asher-ben-shooshan, 2019-11-21:
https://github.com/google/riscv-dv/blob/e9517397a1eeac8f7a8dfce3e8d8c6b5d28c3dec/src/riscv_instr_base.sv#L63
file riscv_instr_base.sv line 63 in the repository asher-ben-shooshan/mriscv

taoliug, 2019-11-23:
Can you elaborate what's the issue?

asher-ben-shooshan, 2019-11-24:
בס"ד
Hi
When randomizing a new instruction it should be from the supported group (ISA group- RV32I and so on)
So I have added a constraint on the instruction to be from the supported group.

Thanks
Asher

taoliug, 2019-11-24:
There's one constraint here
https://github.com/google/riscv-dv/blob/master/src/riscv_rand_instr.sv#L43

I guess you mean moving this to riscv_instr_base.sv

taoliug, 2019-12-09:
the existing instruction class will be deprecated soon, closing this issue for now.

closed 2019-12-09 by taoliug

asher-ben-shooshan, 2019-12-26:
בס"ד
Hi
I just wanted to know if you have a version of the extension B (bit manipulation) of the google assembly riscv generator ?

Thanks
Asher

https://github.com/google/riscv-dv/blob/master/src/riscv_rand_instr.sv#L43<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fgoogle%2Friscv-dv%2Fblob%2Fmaster%2Fsrc%2Friscv_rand_instr.sv%23L43&data=02%7C01%7Casher%40mellanox.com%7Cc0587da6adbd42575d7708d770f59fe5%7Ca652971c7d2e4d9ba6a4d149256f461b%7C0%7C0%7C637102072572888633&sdata=RVQwQcHfStD%2F3cVuXwRxqQAWCbZLKOkHSqJ%2FSnAgxPM%3D&reserved=0>

taoliug, 2019-12-26:
Not yet, expect to be ready in a couple of months.