chipsalliance/riscv-dv #320: where to add persistent state for mode coverage

simond-imperas, 2019-11-22:
Hi Tao
My impression is that no state is held from instruction to instruction in the coverage processing

For some coverage I need to to cross between instructions and their operands and the vector engine's state - as vector instructions can be polymorphic based on previous instructions configuring the vector engine options.

So - what I need is the best place/class/file to store the state/values of some CSRs when they are set by an instruction such that I can access this state in a later instruction in the coverage setup...

There are the 5 vector CSRs that I am thnking about.

Pls just suggest a file/line of code to focus me.

thanks
Simon

taoliug, 2019-11-22:
The previous instruction is saved here.  If you want to save more instructions, you can add a queue for this.
https://github.com/google/riscv-dv/blob/master/src/riscv_instr_cover_group.sv#L1120

closed 2019-11-22 by taoliug

simond-imperas, 2019-11-22:
Thanks. It is not the instruction I need to save. Some intructions update several CSRs and the values of these affect the operation and results of the following instructions - there may be millions of instructions the follow that are affected by the one that updated the CSRs - so I cant keep a queue of instructions - I will look into the mechanism you point at and see what might work. - thanks.

taoliug, 2019-11-22:
It's better to save the CSR state in some separate variable rather than creating a deep queue. For example, this is how I save branch history to sample different branch hit/miss scenario.
https://github.com/google/riscv-dv/blob/master/src/riscv_instr_cover_group.sv#L1085

simond-imperas, 2019-11-22:
OK thanks - will go with that.