chipsalliance/riscv-dv #329: Running riscv_hint_instr_test on custom target fails

abdullahyildiz, 2019-11-28:
I work on commit id bce290e.

I want to implement HINT instruction tests for RV64G.

I used the **custom_target** option while running the **run.py** script. Before that, I made the following changes in the repository:

- Copied the contents of rv64gc directory into a new one named **rv64g** under **target/**
- Modified riscv_core_setting.sv to remove variables related to C standard extension from **supported_isa** array.

Then I ran this command:

`python3 run.py --custom_target=target/rv64g --isa=rv64g --mabi=lp64 --test=riscv_hint_instr_test --iss=ovpsim,spike -v
`

This command fails while comparing outputs of spike and riscv-ovpsim (see **run.log** and another issue https://github.com/google/riscv-dv/issues/283#issue-517770838 in this repository). In addition to this, although there are no compressed instructions in generated assembly programs, csv files that are generated from riscv-ovpsim log files include compressed instructions:

```
...
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.srai,a0,000000000000000b,a0,000000000000000b,,,0,"srai    a0,a0,0x0",00000000800084f0,8501,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.slli,zero,0,zero,0,,,38,"slli    zero,zero,0x38",0000000080008b3a,1062,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.srai,a4,0000000000000000,a4,0000000000000000,,,0,"srai    a4,a4,0x0",0000000080009076,8701,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.add,zero,0,zero,0,zero,0,,"add     zero,zero,a7",0000000080009274,9046,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.li,zero,0,zero,0,,,fffffff0,"li      zero,-16",0000000080009e06,5041,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.lui,zero,0,zero,0,,,9,"lui     zero,0x9",000000008000af9c,6025,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.lui,zero,0,zero,0,,,f,"lui     zero,0xf",000000008000afa6,603d,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.lui,zero,0,zero,0,,,18,"lui     zero,0x18",000000008000b2f0,6061,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.srli,a5,0000000000000000,a5,0000000000000000,,,0,"srli    a5,a5,0x0",000000008000b636,8381,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.slli,zero,0,zero,0,,,9,"slli    zero,zero,0x9",000000008000b890,0026,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.slli,zero,0,zero,0,,,18,"slli    zero,zero,0x18",000000008000bf02,0062,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.srai,a0,ffffffff80000000,a0,ffffffff80000000,,,0,"srai    a0,a0,0x0",000000008000c70a,8501,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.mv,zero,0,zero,0,a2,0000000000000000,,"mv      zero,a2",000000008000c998,8032,,3,,,,,,,,,,,,,[],[],,,,,,
out_2019-11-28/ovpsim_sim/riscv_hint_instr_test.4.csv:c.slli,s8,0000000000000010,s8,0000000000000010,,,0,"slli    s8,s8,0x0",00000000800013fa,0c02,,3,,,,,,,,,,,,,[],[],,,,,,
...
```

[run.log](https://github.com/google/riscv-dv/files/3901439/run.log)
[riscv_core_setting.txt](https://github.com/google/riscv-dv/files/3901441/riscv_core_setting.txt)

simond-imperas, 2019-11-28:
Hi @abdullah most of the GCC for RV64 we have seen always have been GC i.e. they do compressed so even if your .S is not compressed the compiler makes it into one - and if you look at your .csv you see the binary column shows only 4 hex chars - ie compressed - first line the binary column is 8501 - which is compressed - if you look at the objdump it should show this - and so riscvOVPsim is only listing what is in the log file... regards, Simon

taoliug, 2019-11-28:
You should remove hint instruction test if you want to disable compressed instructions. HINT instructions use compressed instruction encoding.

abdullahyildiz, 2019-11-29:
@simond-imperas @taoliug OK. Thanks for your explanation.

closed 2019-11-29 by abdullahyildiz

abdullahyildiz, 2019-12-02:
@taoliug,

Could you please take a look at the following link:

https://github.com/riscv/riscv-gnu-toolchain/issues/552#issuecomment-560312242

Thanks.

taoliug, 2019-12-02:
Yes, that's exactly how HINT/illegal instruction is injected, as compiler will throw errors if you pass HINT instruction assembly directly. Like I mentioned above, if your processor doesn't support compressed instruction, please remove the hint instruction test from the testlist.

abdullahyildiz, 2019-12-02:
@taoliug,
Is this due to how you define HINT instruction tests in riscv-dv project?
Because I couldn't see any description on how HINT instructions should be encoded in RISC-V instruction set manual.

taoliug, 2019-12-02:
Please refer to user mode spec 16.7 HINT instruction

abdullahyildiz, 2019-12-02:
Yes, I understand your point. -Put it another way- Should all HINT instructions be encoded as compressed instructions?

taoliug, 2019-12-02:
Each HINT has clean encoding definition in the spec. All HINT instruction use RVC encoding. 

> A portion of the RVC encoding space is reserved for microarchitectural HINTs

abdullahyildiz, 2019-12-02:
Shouldn't this definition be valid only for C extension?
Then how a RV64G processor could run HINT instructions?

taoliug, 2019-12-02:
> RV32I reserves a large encoding space for HINT instructions, which are usually used to commu-nicate performance hints to the microarchitecture.  HINTs are encoded as integer computationalinstructions withrd=x0. 

I just checked the spec. Actually there are also HINT instructions in RV32I encoding. These instruction are tested when rd is randomized to 0. The hint instruction test in the regression is for HINT instruction using RVC encoding.

abdullahyildiz, 2019-12-02:
Yes, that was my question.

abdullahyildiz, 2019-12-03:
Do you have a plan to support RV32I HINT instructions in the future?

taoliug, 2019-12-03:
They are supported already, rd could be randomized to x0 for computation instructions. It's pretty much tested in every test.

abdullahyildiz, 2019-12-04:
> They are supported already, rd could be randomized to x0 for computation instructions. It's pretty much tested in every test.

OK. Thanks.