chipsalliance/riscv-dv #342: Cannot generate the csv file

danghai, 2019-12-09:
Hi, I am not sure why I cannot get the `csv` file. I run the command:
`python3 run.py --test riscv_jump_stress_test --iss spike --simulator qrun -o output -v -s all`

The tail of log file:
```
# ** Note: $finish    : /home/danghai/questasim/linux_x86_64/../verilog_src/uvm-1.2/src/base/uvm_root.svh(517)
#    Time: 0 ns  Iteration: 269  Instance: /riscv_instr_gen_tb_top
# End time: 12:29:13 on Dec 09,2019, Elapsed time: 0:00:08
# Errors: 0, Warnings: 0
# *** Summary *********************************************
#     qrun: Errors:   0, Warnings:   0
#     vsim: Errors:   0, Warnings:   0
#   Totals: Errors:   0, Warnings:   0

Mon, 09 Dec 2019 12:29:13 run.py:347   INFO     Compiling output/asm_tests/riscv_jump_stress_test_0.S
Mon, 09 Dec 2019 12:29:13 run.py:348   DEBUG    /opt/riscv/bin/riscv64-unknown-elf-gcc -static -mcmodel=medany              -fvisibility=hidden -nostdlib              -nostartfiles output/asm_tests/riscv_jump_stress_test_0.S              -I/home/danghai/hhoangda/riscv-dv/user_extension              -T/home/danghai/hhoangda/riscv-dv/scripts/link.ld  -o output/asm_tests/riscv_jump_stress_test_0.o  -march=rv32imc -mabi=ilp32
Mon, 09 Dec 2019 12:29:13 run.py:350   DEBUG    b''
Mon, 09 Dec 2019 12:29:13 run.py:352   INFO     Converting to output/asm_tests/riscv_jump_stress_test_0.bin
Mon, 09 Dec 2019 12:29:13 run.py:355   DEBUG    b''
Mon, 09 Dec 2019 12:29:13 run.py:347   INFO     Compiling output/asm_tests/riscv_jump_stress_test_1.S
Mon, 09 Dec 2019 12:29:13 run.py:348   DEBUG    /opt/riscv/bin/riscv64-unknown-elf-gcc -static -mcmodel=medany              -fvisibility=hidden -nostdlib              -nostartfiles output/asm_tests/riscv_jump_stress_test_1.S              -I/home/danghai/hhoangda/riscv-dv/user_extension              -T/home/danghai/hhoangda/riscv-dv/scripts/link.ld  -o output/asm_tests/riscv_jump_stress_test_1.o  -march=rv32imc -mabi=ilp32
Mon, 09 Dec 2019 12:29:13 run.py:350   DEBUG    b''
Mon, 09 Dec 2019 12:29:13 run.py:352   INFO     Converting to output/asm_tests/riscv_jump_stress_test_1.bin
Mon, 09 Dec 2019 12:29:13 run.py:355   DEBUG    b''
Mon, 09 Dec 2019 12:29:13 run.py:87    INFO     Processing ISS setup file : /home/danghai/hhoangda/riscv-dv/yaml/iss.yaml
Mon, 09 Dec 2019 12:29:13 run.py:93    INFO     Found matching ISS: spike
Mon, 09 Dec 2019 12:29:13 run.py:412   INFO     spike sim log dir: output/spike_sim
Mon, 09 Dec 2019 12:29:13 run.py:423   INFO     Running spike sim: output/asm_tests/riscv_jump_stress_test_0.o
Mon, 09 Dec 2019 12:29:13 lib.py:105   DEBUG    /opt/riscv/bin/spike --log-commits --isa=rv32imc -l output/asm_tests/riscv_jump_stress_test_0.o &> output/spike_sim/riscv_jump_stress_test.0.log
Mon, 09 Dec 2019 12:29:13 lib.py:128   DEBUG    
Mon, 09 Dec 2019 12:29:13 run.py:425   DEBUG    /opt/riscv/bin/spike --log-commits --isa=rv32imc -l output/asm_tests/riscv_jump_stress_test_0.o &> output/spike_sim/riscv_jump_stress_test.0.log
Mon, 09 Dec 2019 12:29:13 run.py:423   INFO     Running spike sim: output/asm_tests/riscv_jump_stress_test_1.o
Mon, 09 Dec 2019 12:29:13 lib.py:105   DEBUG    /opt/riscv/bin/spike --log-commits --isa=rv32imc -l output/asm_tests/riscv_jump_stress_test_1.o &> output/spike_sim/riscv_jump_stress_test.1.log
Mon, 09 Dec 2019 12:29:13 lib.py:128   DEBUG    
Mon, 09 Dec 2019 12:29:13 run.py:425   DEBUG    /opt/riscv/bin/spike --log-commits --isa=rv32imc -l output/asm_tests/riscv_jump_stress_test_1.o &> output/spike_sim/riscv_jump_stress_test.1.log

```

It only generates the logfile in `output/spike_sim`. Do I miss something? Thank you

taoliug, 2019-12-09:
trace.csv won't be generated if you run with only one ISS. Two options:
1) Try run with -iss spike,ovpsim
2) Run coverage flow: python3 cov.py --dir output/spike_sim

I'd recommend you to try option 2) as it generates full trace CSV file.

danghai, 2019-12-09:
Thank you. It works for me now
```
danghai@ubuntu:~/hhoangda/riscv-dv$ python3 cov.py --dir output/spike_sim --simulator qrun
Mon, 09 Dec 2019 12:42:34 INFO     Processing trace log under output/spike_sim
Mon, 09 Dec 2019 12:42:34 INFO     Process spike log[1/2] : output/spike_sim/riscv_jump_stress_test.0.log
Mon, 09 Dec 2019 12:42:34 INFO     Processing spike log : output/spike_sim/riscv_jump_stress_test.0.log
Mon, 09 Dec 2019 12:42:34 INFO     Processed instruction count : 1374
Mon, 09 Dec 2019 12:42:34 INFO     CSV saved to : output/spike_sim/riscv_jump_stress_test.0.csv
Mon, 09 Dec 2019 12:42:34 INFO     Process spike log[2/2] : output/spike_sim/riscv_jump_stress_test.1.log
Mon, 09 Dec 2019 12:42:34 INFO     Processing spike log : output/spike_sim/riscv_jump_stress_test.1.log
Mon, 09 Dec 2019 12:42:35 INFO     Processed instruction count : 5135
Mon, 09 Dec 2019 12:42:35 INFO     CSV saved to : output/spike_sim/riscv_jump_stress_test.1.csv
Mon, 09 Dec 2019 12:42:35 INFO     Building the coverage collection framework
Mon, 09 Dec 2019 12:42:38 INFO     Collecting functional coverage from 2 trace CSV
Mon, 09 Dec 2019 12:42:38 INFO     Processing batch 1/1
Mon, 09 Dec 2019 12:42:40 INFO     Collecting functional coverage from 2 trace CSV...done
Mon, 09 Dec 2019 12:42:40 INFO     Coverage results are saved to out_2019-12-09
```

closed 2019-12-09 by taoliug